Koenkk/zigbee2mqtt #14136: [Feature request]:

PheelTi, 2022-09-22:
### Is your feature request related to a problem? Please describe

It would be great to make the frontend exposed through HTTPS.

### Describe the solution you'd like

in configuration.yaml, add :
```yaml
frontend:
  secure:
    ssl_cert: [path to cert file]
    ssl_key: [path to key file]
```

then adapt code in the `start()` function in [https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/frontend.ts](lib/extension/frontend.ts) 

### Describe alternatives you've considered

It can be handled with a proxy like Traefik... I do use one for my websites exposed on the internet.
but I do not want to use such proxy for LAN access.

### Additional context

I may try to code this by myself as it is not really complicated (I suppose), and submit a pull request.

lnawrat, 2022-10-23:
I believe a need for this enhancement is still very much alive. Please do not close the ticket :-)

PheelTi, 2022-10-25:
I have a working fix at home. Will try to submit a pull request in the upcoming weeks.

Omirax, 2022-10-30:
> I have a working fix at home. Will submit a pull request in the upcoming weeks.

We need this!!!

PheelTi, 2022-12-03:
Pull request submitted : #15329

PR #15329 (2022-12-04): Expose frontend in HTTPS
PR for HTTPS exposure, see feature request #14136 
patch to expose frontend in HTTPS.
the chose implementation for simplicity is the add of 2 configuration params under 'frontend' : 
- 'ssl_cert' path to SSL certificate
- 'ssl_key' path to SSL key

100% test coverage.

Closes #14136
closed 2022-12-04 by Koenkk