Koenkk/zigbee2mqtt #14151: [New device support]: HEIMAN - TempDimmerSw-EM-3.0 - HS2WDSC-E

olivr2s, 2022-09-23:
### Link

https://www.amazon.fr/gp/product/B096M2BDSR/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x000d6f001619265f","nwkAddr":4733,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"TempDimmerSw-EM-3.0","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[0,1,3,4,5,2821,4096,64644],"outClusterList":[3,6,8,25,768,4096],"clusters":{"genBasic":{"attributes":{}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffef17dcc","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffef17dcc","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffef17dcc","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffef17dcc","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2019.9.27","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1663926834003,"defaultSendRequestWhen":"immediate"}

### Comments

I have this HEIMAN remote which is not supported but is very similar to the HS2WDSR-E model.
I found the configuration file with other similar devices and managed to pair and receive information about button presses.

On the other hand I have errors of conversion, it must be necessary to change converter. (log of remote control presses attached)
`Received Zigbee message from '0x000d6f001619265f', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":245,"transtime":10}' from endpoint 1 with groupID 0
No converter available for 'HS2WDSC-E' with cluster 'lightingColorCtrl' and type 'commandMoveToColorTemp' and data '{"colortemp":245,"transtime":10}'`

[log télécommande.txt](https://github.com/Koenkk/zigbee2mqtt/files/9632891/log.telecommande.txt)

I hope to provide all the necessary elements for its addition.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TempDimmerSw-EM-3.0'],
    model: 'HS2WDSC-E',
    vendor: 'HEIMAN',
    description: 'Remote dimmer and temp control',
    fromZigbee: [fz.battery, fz.command_on, fz.command_off, fz.command_move, fz.command_stop, fz.command_move_to_color],
    exposes: [e.battery(), e.action(['on', 'off', 'move', 'stop', 'color_move'])],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff', 'genLevelCtrl', 'lightingColorCtrl']);
        await reporting.batteryPercentageRemaining(endpoint, {min: constants.repInterval.MINUTES_5, max: constants.repInterval.HOUR});
    },
};

module.exports = definition;
```

### Supported color modes

color temperature CCT

### Color temperature range

{"colorTempPhysicalMin":153, "colorTempPhyiscalMax":370}

Koenkk, 2022-09-23:
> On the other hand I have errors of conversion, it must be necessary to change converter. (log of remote control presses attached)

Try adding `fz.command_move_to_color_temp` to `fromZigbee`.

olivr2s, 2022-09-23:
I add fz.command_move_to_color_temp to fromZigbee

log of remote control presses attached

[log télécommande 2.txt](https://github.com/Koenkk/zigbee2mqtt/files/9636116/log.telecommande.2.txt)

olivr2s, 2022-09-23:
> > On the other hand I have errors of conversion, it must be necessary to change converter. (log of remote control presses attached)
> 
> Try adding `fz.command_move_to_color_temp` to `fromZigbee`.

[log télécommande 2.txt](https://github.com/Koenkk/zigbee2mqtt/files/9636120/log.telecommande.2.txt)

Koenkk, 2022-09-23:
Great looks that it works. I've integrated the device.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

olivr2s, 2022-09-24:
> Great looks that it works. I've integrated the device.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Thanks for adding the remote in z2m.
I did the inclusion with no problem. Thanks.

On the other hand, I only have an "action" info for the 5 buttons and when I press the keys, I have no numerical value that goes back...
Center button -> on - off (it's ok)
Left and right buttons -> color_temperature_move
Up and down buttons -> brightness_stop - brightness_move_up - brightness_move_down

how to have the numerical values which are in the logs ?  ("action_rate":50 or "action_color_temperature":285)

here are the first recognition logs:
[log télécommande 2.txt](https://github.com/Koenkk/zigbee2mqtt/files/9639064/log.telecommande.2.txt)

here are the logs now:
[log télécommande 3.txt](https://github.com/Koenkk/zigbee2mqtt/files/9639068/log.telecommande.3.txt)

Koenkk, 2022-09-25:
> how to have the numerical values which are in the logs ? ("action_rate":50 or "action_color_temperature":285)

I guess you are talking to HA? You can use this in your automations by using: https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt

olivr2s, 2022-09-25:
> > how to have the numerical values which are in the logs ? ("action_rate":50 or "action_color_temperature":285)
> 
> I guess you are talking to HA? You can use this in your automations by using: https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt

It's not HA that I use, but Jeedom.
Thank you for this information, I will see if I can adapt to Jeedom.

olivr2s, 2022-09-25:
> It's not HA that I use, but Jeedom. Thank you for this information, I will see if I can adapt to Jeedom.

it's ok for me, everything works.
I manage to retrieve the numerical values of the remote control using the jmqtt plugin in Jeedom.
Thanks.

Koenkk, 2022-09-28:
Great, assuming this can be closed now.

closed 2022-09-28 by Koenkk

mtgpcs, 2023-10-09:
Sorry to comment on an old thread but how did you get the remote to pair with Z2M? I have tried pushing the button by the battery and it flashes for a few seconds then nothing, it never shows up in Z2M.