Koenkk/zigbee2mqtt #14154: Pls add MOES ZS-EUD_1gang and ZS-EUD_3gang and change model ZS-EUD into ZS-EUD_2gang in moes.js

louismulder, 2022-09-23:
### Link

https://nl.aliexpress.com/item/1005004198981729.html?spm=a2g0o.productlist.0.0.1a481457wBuqWh&algo_pvid=b5856b03-a152-4d77-afdf-5a9cb8f1f2a2&algo_exp_id=b5856b03-a152-4d77-afdf-5a9cb8f1f2a2-1&pdp_ext_f=%7B%22sku_id%22%3A%2212000028353511244%22%7D&pdp_npi=2%40dis%21EUR%21%2118.06%2118.06%21%21%21%21%400b0a050116559201304511937eda98%2112000028353511244%21sea

### Database entry

3gang: "modelId":"TS0601","manufacturerName":"_TZE200_vm1gyrso"

### Comments

The ZS-EUD_1gang is not working with the current TuYa
//            {modelID: 'TS0601', manufacturerName: '_TZE200_ip2akl4w'},
So pls remove the fingerprint in tuya.js

1. Pls extend in the moes.js ZSEUS for 3 gang support with:
                 case 15:
                    result.state_l3 = value ? "ON" : "OFF";
                    break;
                case 16:
                    result.brightness_l3 = utils.mapNumberRange(value, 0, 1000, 0, 254);
                    break;

See:
const fzLocal = {
    ZSEUD: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: async (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case 1:
                    result.state_l1 = value ? 'ON' : 'OFF';
                    break;
                case 2:
                    result.brightness_l1 = utils.mapNumberRange(value, 0, 1000, 0, 254);
                    break;
                case 7:
                    result.state_l2 = value ? 'ON' : 'OFF';
                    break;
                case 8:
                    result.brightness_l2 = utils.mapNumberRange(value, 0, 1000, 0, 254);
                    break;
                 case 15:
                    result.state_l3 = value ? "ON" : "OFF";
                    break;
                case 16:
                    result.brightness_l3 = utils.mapNumberRange(value, 0, 1000, 0, 254);
                    break;
               default:
                    meta.logger.warn(`zigbee-herdsman-converters:ZSEUD: NOT RECOGNIZED DP #${dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    },
};

2. Rename current model : 'ZS-EUD' into 'ZS-EUD_2gang' if needed

3. Add  model: "ZS-EUD_1gang'

    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ip2akl4w'}],
        model: 'ZS-EUD_1gang',
        vendor: 'Moes',
        description: '1 gang light dimmer switch',
        fromZigbee: [fzLocal.ZSEUD, fz.ignore_basic_report],
        toZigbee: [tzLocal.ZSEUD_brightness, tzLocal.ZSEUD_state],
        meta: {turnsOffAtBrightness1: true, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        },
        exposes: [e.light_brightness().withEndpoint('l1').setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET)],
        endpoint: (device) => {
            return {'l1': 1};
        },
    },

4. Add model 'ZS-EUD_3gang'

    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_vm1gyrso'}],
        model: 'ZS-EUD_3gang',
        vendor: 'Moes',
        description: '3 gang light dimmer switch',
        fromZigbee: [fzLocal.ZSEUD, fz.ignore_basic_report],
        toZigbee: [tzLocal.ZSEUD_brightness, tzLocal.ZSEUD_state],
        meta: {turnsOffAtBrightness1: true, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
            if (device.getEndpoint(2)) await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(3)) await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
        },
        exposes: [e.light_brightness().withEndpoint('l1').setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET),
            e.light_brightness().withEndpoint('l2').setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET),
            e.light_brightness().withEndpoint('l3').setAccess('state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET)],
        endpoint: (device) => {
            return {'l1': 1, 'l2': 1, 'l3': 1};
        },
    },

Koenkk, 2022-09-23:
Could you check if all 3 devices work correctly with the following external converter: https://gist.github.com/Koenkk/f56a3bfef6a4dd87aab01f672261f86a

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

louismulder, 2022-09-24:
Hi Koen,

There is something wrong in the ext_converter.js.
When I start z2m it gives the following error:

UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "SyntaxError: Duplicate parameter name not allowed in this context".

Koenkk, 2022-09-25:
I updated https://gist.github.com/Koenkk/f56a3bfef6a4dd87aab01f672261f86a, can you try again?

louismulder, 2022-09-26:
Hi Koen,

Same error with the updated converter

Regards Louis

Koenkk, 2022-09-28:
Sry, updated https://gist.github.com/Koenkk/f56a3bfef6a4dd87aab01f672261f86a again.

louismulder, 2022-09-29:
Hi Koen,

I have good and bad news.
The good news is that the ext_converter works without start errors.
And the 2gand and 3gang dimers are working
The bad news is that the 1gang is not working at all. It gives Error: No datapoints map defined

Btw I saw also a typo? in  the 3gang return statement :          return {'l1': 1, 'l2': 1, 'l3': 3};
Shouldn't the return parameter l3 -> 1?

I have added the test in a datafile 
[ext_converter_test.txt](https://github.com/Koenkk/zigbee2mqtt/files/9677915/ext_converter_test.txt)

Koenkk, 2022-09-29:
I updated https://gist.github.com/Koenkk/f56a3bfef6a4dd87aab01f672261f86a , can you provide the same logging again?

louismulder, 2022-09-30:
Hi Koen,

I have tested the 1gang and 2gang with the updated converter.
The 1gang is not working, but the error is gone.

see the test in the attached file.
regards Louis
[ext_converter_test2.txt](https://github.com/Koenkk/zigbee2mqtt/files/9683723/ext_converter_test2.txt)

Koenkk, 2022-10-01:
I updated https://gist.github.com/Koenkk/f56a3bfef6a4dd87aab01f672261f86a , please try again. Note that this converter requires zigbee2mqtt **1.28.0**

louismulder, 2022-10-01:
Hi Koen,

I have installed 1.28.0 and tested again.
The 1gang does not work:
1. Pairing OK
2. set brightness to 28,
debug 2022-10-01 16:54:49: Received MQTT message on 'zigbee2mqtt/0xa4c1382b7d0b60a2/set' with data '{"brightness":28}'
debug 2022-10-01 16:54:49: Publishing 'set' 'brightness' to '0xa4c1382b7d0b60a2'
error 2022-10-01 16:54:49: Publish 'set' 'brightness' to '0xa4c1382b7d0b60a2' failed: 'Error: No datapoints map defined'
debug 2022-10-01 16:54:49: Error: No datapoints map defined
    at Object.convertSet (D:\Domotica\Java\Output\zigbee2mqtt\node_modules\zigbee-herdsman-converters\lib\tuya.js:1209:74)
    at Publish.onMQTTMessage (D:\Domotica\Java\Output\zigbee2mqtt\lib\extension\publish.ts:246:52)
    at EventEmitter.emit (node:events:402:35)
    at EventBus.emitMQTTMessage (D:\Domotica\Java\Output\zigbee2mqtt\lib\eventBus.ts:109:22)
    at MQTT.onMessage (D:\Domotica\Java\Output\zigbee2mqtt\lib\mqtt.ts:140:27)
    at MqttClient.emit (node:events:390:28)
    at MqttClient._handlePublish (D:\Domotica\Java\Output\zigbee2mqtt\node_modules\mqtt\lib\client.js:1547:12)
    at MqttClient._handlePacket (D:\Domotica\Java\Output\zigbee2mqtt\node_modules\mqtt\lib\client.js:535:12)
    at work (D:\Domotica\Java\Output\zigbee2mqtt\node_modules\mqtt\lib\client.js:438:12)
    at Writable.writable._write (D:\Domotica\Java\Output\zigbee2mqtt\node_modules\mqtt\lib\client.js:452:5)

Regards Louis

Koenkk, 2022-10-03:
Updated https://gist.github.com/Koenkk/f56a3bfef6a4dd87aab01f672261f86a , please try again.

louismulder, 2022-10-04:
Ho Koen,

I have tested all 3 dimmers with the updated converter.

See the attached file for the results.
[ext_converter_test3.txt](https://github.com/Koenkk/zigbee2mqtt/files/9706418/ext_converter_test3.txt)

regards Louis

Koenkk, 2022-10-04:
Updated https://gist.github.com/Koenkk/f56a3bfef6a4dd87aab01f672261f86a

Note that brightness will indeed not turn on the switch, this is expected TuYa behaviour.

louismulder, 2022-10-04:
Hi Koen,

All dimmers are working without errors.
But pls change the behavior of turning ON the Dimmer if not yet ON when the brightness request is given.
Otherwise I have to give 2 request for dimming (1 for status ON and the other for the brightness) and I have to request the state in the sender of my application.  Giving the status and brightness in one payload does not work. This is totally different from other dimmers I use.

And BTW I see more states and brightness values state_l1_l1, brightness_l1_l1

debug 2022-10-04 19:09:46: Received Zigbee message from '0xa4c138c2eb17b616', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,209],"type":"Buffer"},"datatype":2,"dp":2}],"seq":23552}' from endpoint 1 with groupID 0
info  2022-10-04 19:09:46: MQTT publish: topic 'zigbee2mqtt/0xa4c138c2eb17b616', payload '{"brightness_l1":53,"brightness_l1_l1":53,"brightness_l2":53,"brightness_l2_l2":53,"brightness_l3":71,"brightness_l3_l3":71,"linkquality":171,"state_l1":"ON","state_l1_l1":"ON","state_l2":"OFF","state_l2_l2":"OFF","state_l3":"OFF","state_l3_l3":"OFF"}'

regards Louis

Koenkk, 2022-10-04:
@louismulder you can use a payload like this: `{"state":"ON","brightness":130}`

Regarding the duplicate properties, please stop z2m, use the latest [external converter](https://gist.github.com/Koenkk/f56a3bfef6a4dd87aab01f672261f86a) and clear `data/state.json`. Start z2m control the device and check if the duplicates are gone.

Koenkk, 2022-10-05:
Fixed in https://gist.github.com/Koenkk/f56a3bfef6a4dd87aab01f672261f86a

Koenkk, 2022-10-06:
Please check with: https://gist.github.com/Koenkk/f56a3bfef6a4dd87aab01f672261f86a

louismulder, 2022-10-07:
Hi Koen,

I have tested all 3 gang dimmers, with the status and brightness in one payload.
1gang is working OK
2gang and 3gang not possible to change the brightness.

Koenkk, 2022-10-08:
@louismulder does it work when the device is already on and you just change the brightness?

louismulder, 2022-10-08:
Hi Koen,

Yes, the brightness is changed, when I use 2 requests,  First the status and then the brightness.

regards Louis

Koenkk, 2022-10-08:
When the device is off, and you publish `{"state":"ON", "brightness":123}` does it work? (I guess you use `{"brightness":123, "state":"ON"}` now.

louismulder, 2022-10-08:
Hi Koen,

I always use: {"state":"ON", "brightness":123}, but that doesn't work.

Koenkk, 2022-10-09:
And if you publish ` {"state":"ON"}` first and then ` {"brightness":123}`?

louismulder, 2022-10-09:
Hi Koen,

If I publish each time {"state":"ON"} first and then {"brightness":123} for my DIM command, then it does not work. So I have to check in my sender if the state is ON, then I can publish the the  {"brightness":123} only. But that is not a solution.
The original request I did in this issue (update the moes.js and tuya.js) is working correctly.

regards Louis

Koenkk, 2022-10-09:
But if first a `{"state":"ON"}` and a few secs after that `{"brightness":123}` doesn't work, how does: 

> So I have to check in my sender if the state is ON

Fix it?

louismulder, 2022-10-09:
Hi Koen,

I have fixed it in my sender, but it is not a nice solution.
    case DIMMER:
      switch ( message.Cmd )
      {
      case DIMON:
      case ON:
        body = "\"state" + label + "\":\"ON\"";
        break;
      case DIM:
        if ( device.LastCommand == CommandTypesEnum.DIMOFF || device.LastCommand == CommandTypesEnum.OFF)
        {
          body = "\"state" + label + "\":\"ON\"";
          if ( ZigbeeDiscoveryService.publishMessage( unit.Address + "/set", "{" + body + "}", 0 ) )
          {
            Utils.sleep( 500 );
            MessageQueue.addMessage( message );
          }
        }
        else
        {
          body = "\"brightness" + label + "\":" + Integer.toString( (int) Math.round( (message.Brightness * (254 / 100.0)) ) );
          brightness = Integer.toString( message.Brightness );
        }
        break;

      default:
      body = "\"state" + label + "\":\"OFF\"";
      }
      break;

Koenkk, 2022-10-10:
I'm not sure what it does exactly; but can you check if https://github.com/Koenkk/zigbee2mqtt/issues/14154#issuecomment-1272597100 works?

Koenkk, 2022-10-11:
And a `{"state":"ON"}` immediately followed by a `{"brightness":123}` works?

louismulder, 2022-10-11:
Hi Koen,

already tried. Doesn't work

louismulder, 2022-10-11:
Hi Koen,

here is the complete logging. BTW I see that devoices which are not connect have the state onlien?

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9757491/log.txt)

Koenkk, 2022-10-11:
Thanks for the logging, can you try if the device correctly responds when you publish the following messages with at least 1 seconds in between?

- `{"state_l1":"ON"}`
- `{"state_l1":"ON"}`
- `{"brightness_l1":91}`

louismulder, 2022-10-11:
Hi Koen,

I have tested your requested sequence. See log file
[ext_converter_test4.txt](https://github.com/Koenkk/zigbee2mqtt/files/9759175/ext_converter_test4.txt)
But the brightness will not be changed.

Regarding the devices which are not connected.
My test is always with the coordinator and one device active (power on).
So I see that the other devices gives the "state":"online", while the power is off on those devices.
debug 2022-10-11 17:17:11: Received MQTT message on 'zigbee2mqtt/0xa4c1382b7d0b60a2/availability' with data '{"state":"online"}'
debug 2022-10-11 17:17:11: Received MQTT message on 'zigbee2mqtt/0xa4c138c8ced1cd44/availability' with data '{"state":"online"}'
debug 2022-10-11 17:17:11: Received MQTT message on 'zigbee2mqtt/0xa4c138c2eb17b616/availability' with data '{"state":"online"}'

Koenkk, 2022-10-12:
When off, does the following sequence change the brightness?

- `{"state_l1":"ON"}`
- `{"state_l1":"ON"}`
- `{"brightness_l1":150}`
- `{"brightness_l1":100}`
- `{"brightness_l1":50}`

> So I see that the other devices gives the "state":"online", while the power is off on those devices.

This depends on whether z2m already pinged those devices. See https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#device-availability on how the available state is determined.

louismulder, 2022-10-12:
Hi Koen,

Brightness does not change. Here is the logging
debug 2022-10-12 19:52:06: Received MQTT message on 'zigbee2mqtt/0xa4c13856f0447f4b/set' with data '{"state_l1":"ON"}'
debug 2022-10-12 19:52:06: Publishing 'set' 'state' to '0xa4c13856f0447f4b'
info  2022-10-12 19:52:06: MQTT publish: topic 'zigbee2mqtt/0xa4c13856f0447f4b', payload '{"brightness_l1":28,"brightness_l2":3,"linkquality":131,"state_l1":"ON","state_l2":"OFF"}'
debug 2022-10-12 19:52:06: Received Zigbee message from '0xa4c13856f0447f4b', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":64768}' from endpoint 1 with groupID 0
info  2022-10-12 19:52:06: MQTT publish: topic 'zigbee2mqtt/0xa4c13856f0447f4b', payload '{"brightness_l1":28,"brightness_l2":3,"linkquality":131,"state_l1":"ON","state_l2":"OFF"}'
debug 2022-10-12 19:52:07: Received MQTT message on 'zigbee2mqtt/0xa4c13856f0447f4b/set' with data '{"state_l1":"ON"}'
debug 2022-10-12 19:52:07: Publishing 'set' 'state' to '0xa4c13856f0447f4b'
info  2022-10-12 19:52:07: MQTT publish: topic 'zigbee2mqtt/0xa4c13856f0447f4b', payload '{"brightness_l1":28,"brightness_l2":3,"linkquality":131,"state_l1":"ON","state_l2":"OFF"}'
debug 2022-10-12 19:52:07: Received Zigbee message from '0xa4c13856f0447f4b', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":65024}' from endpoint 1 with groupID 0
info  2022-10-12 19:52:07: MQTT publish: topic 'zigbee2mqtt/0xa4c13856f0447f4b', payload '{"brightness_l1":28,"brightness_l2":3,"linkquality":127,"state_l1":"ON","state_l2":"OFF"}'
debug 2022-10-12 19:52:08: Received MQTT message on 'zigbee2mqtt/0xa4c13856f0447f4b/set' with data '{"brightness_l1":150}'
debug 2022-10-12 19:52:08: Publishing 'set' 'brightness' to '0xa4c13856f0447f4b'
info  2022-10-12 19:52:08: MQTT publish: topic 'zigbee2mqtt/0xa4c13856f0447f4b', payload '{"brightness_l1":150,"brightness_l2":3,"linkquality":127,"state_l1":"ON","state_l2":"OFF"}'
debug 2022-10-12 19:52:08: Received Zigbee message from '0xa4c13856f0447f4b', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,2,79],"type":"Buffer"},"datatype":2,"dp":2}],"seq":65280}' from endpoint 1 with groupID 0
info  2022-10-12 19:52:08: MQTT publish: topic 'zigbee2mqtt/0xa4c13856f0447f4b', payload '{"brightness_l1":150,"brightness_l2":3,"linkquality":131,"state_l1":"ON","state_l2":"OFF"}'
debug 2022-10-12 19:52:09: Received MQTT message on 'zigbee2mqtt/0xa4c13856f0447f4b/set' with data '{"brightness_l1":100}'
debug 2022-10-12 19:52:09: Publishing 'set' 'brightness' to '0xa4c13856f0447f4b'
info  2022-10-12 19:52:09: MQTT publish: topic 'zigbee2mqtt/0xa4c13856f0447f4b', payload '{"brightness_l1":100,"brightness_l2":3,"linkquality":131,"state_l1":"ON","state_l2":"OFF"}'
debug 2022-10-12 19:52:09: Received Zigbee message from '0xa4c13856f0447f4b', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,138],"type":"Buffer"},"datatype":2,"dp":2}],"seq":0}' from endpoint 1 with groupID 0
info  2022-10-12 19:52:09: MQTT publish: topic 'zigbee2mqtt/0xa4c13856f0447f4b', payload '{"brightness_l1":100,"brightness_l2":3,"linkquality":131,"state_l1":"ON","state_l2":"OFF"}'
debug 2022-10-12 19:52:10: Received MQTT message on 'zigbee2mqtt/0xa4c13856f0447f4b/set' with data '{"brightness_l1":74}'
debug 2022-10-12 19:52:10: Publishing 'set' 'brightness' to '0xa4c13856f0447f4b'
info  2022-10-12 19:52:10: MQTT publish: topic 'zigbee2mqtt/0xa4c13856f0447f4b', payload '{"brightness_l1":74,"brightness_l2":3,"linkquality":131,"state_l1":"ON","state_l2":"OFF"}'
debug 2022-10-12 19:52:11: Received Zigbee message from '0xa4c13856f0447f4b', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,35],"type":"Buffer"},"datatype":2,"dp":2}],"seq":256}' from endpoint 1 with groupID 0
info  2022-10-12 19:52:11: MQTT publish: topic 'zigbee2mqtt/0xa4c13856f0447f4b', payload '{"brightness_l1":74,"brightness_l2":3,"linkquality":131,"state_l1":"ON","state_l2":"OFF"}'

Koenkk, 2022-10-12:
and with: 

```
- `{"state_l1":"ON"}`
- `{"state_l1":"ON"}`
- `{"brightness_l1":254}`
- `{"brightness_l1":91}`
- `{"brightness_l1":53}`
- `{"brightness_l1":97}`
```

This is exactly the same as what worked in https://github.com/Koenkk/zigbee2mqtt/issues/14154#issuecomment-1275177679

louismulder, 2022-10-12:
Hi Koen,

could you pls implement the 1gang in the moes.js as tested with the latest ext_converter.js and implement the 3gang in the moes.js as I requested at the top of this issue? (extended with the new endpoints in ZSEUD_state and ZSEUD_brigthness)
I think you have already spent a lot of time in this issue. 

regards Louis

Koenkk, 2022-10-18:
This is in the old format and I don't want to add any new devices based on the old TuYa format.

Do I understand correctly that with the converter you provided in the OP the 3 gang works correctly when sending `{"state": "ON", "brightness_l1":91}`? If yes, then please provide me the herdsman debug logging when doing both first with your converter and than with mine.

louismulder, 2022-10-18:
Hi Koen,

Ok, I understand.
Sending {"state": "ON", "brightness_l1":91}, does not work in any of our tests.
The only thing that works is first sending {"state": "ON"} (only when the state=OFF), wait 500ms and then sending  {"brightness_l1":91}.
I'm using your latest ext_converter.js in my production environment, with a patch in my sender.

regards Louis

closed 2022-10-19 by Koenkk