Koenkk/zigbee2mqtt #14157: Rapid Battery Drain of Moes_ZS-SR4-2169 / TS0044

sembeeuk, 2022-09-23:
### What happened?

I have a couple of the Moes_ZS-SR4-2169 / TS0044 remotes. 
https://zigbee.blakadder.com/Moes_ZS-SR4-2169.html

They appear to be mis-identified by Z2M, which doesn't stop them working, but might be related to the issue. 

A battery in these remotes will last a matter of weeks, sometimes as little as ten days with no significant use. A remote sat less than a metre away from the coordinator with no use last just over three weeks! 
While there is a battery indication, it appears to be identified by Z2M as a mains powered switch, going by the picture. Therefore has something happened with regards to the configuration which means it is treated as mains powered?

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

EvertDekker, 2022-09-24:
My experience with battery operated switches is that the battery is drained very fast with z2m.
Disabling the Powercfg report did solve the drain problem for all my switches.
In the z2m frontend goto your device and disable the report by pressing the disable button, at the same time press a button on your switch to keep the device awake to receive the changes.

[image]

TeHashX, 2022-09-28:
> My experience with battery operated switches is that the battery is drained very fast with z2m. Disabling the Powercfg report did solve the drain problem for all my switches. In the z2m frontend [...]

This will disable battery level complete or just will not wake device to query level?

EvertDekker, 2022-09-28:
It will not wake up the device to query every x time.
If you press a button the battery level will still be send.
Give it a try, if it doesn't solve your problem you can always enable it again.

robertalexa, 2022-10-04:
I suspect this issue to be a duplicate, or at least an effect of the issue i have described here and is being investigated https://github.com/Koenkk/zigbee2mqtt/issues/8072

closed 2022-11-12 by github-actions[bot]

Awebian, 2022-11-29:
WOW!  I just turned it off on two of my units/switches.  One I changed the batter on just a few days ago and its already reporting only 86% power!  

Great to hear we will still get power updates with push of a button.

Now I just need to understand the Xiaomi WSDCGQ11LM temperature and humidity sensors.