Koenkk/zigbee2mqtt #14163: Tuya TV02 Thermostat: unable to set window_open (_TZE200_hue3yfsn)

station7, 2022-09-23:
### What happened?

When I switch "open_window" to ON at my Tuya TV02 Thermostat via Webfrontend or mqtt, it changes back to OFF after some seconds.

https://user-images.githubusercontent.com/12595922/192005174-b91eb55f-c61f-49f0-8b09-57d4eedc611e.mov

### What did you expect to happen?

I expected that open_window would stay at "ON" until changed actively.

### How to reproduce it (minimal and precise)

Switch "open_window" via Webfrontend

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

Slaesh's CC2652RB Stick

station7, 2022-09-23:
Okay, removing the batteries as a restart of the thermostat seems to have fixed this issue. Sorry for the confusion.

station7, 2022-10-04:
Sorry to reopen this issue, but after some hours the error returned and I am unable to set window_open...

closed 2022-11-12 by github-actions[bot]

Wikibear, 2022-11-25:
I can confirm this issue. But it must be a software related thing with z2m. After restart of z2m it's fine for hours or days. And then it didn't work. Switch to ON results to OFF.

eknoes, 2022-12-22:
I have the same issue, but can't confirm that a restart of z2m solves the issue. For me, only restarting the device works be removing the batteries works.

Yoda2003, 2023-01-03:
Can confirm the problem since for at least a few weeks. Reinstallation of the complete home assistant doen't changed anything. Problem occurs after a few hours and is always reproducible. For the moment it isn't possible to use the TV02 with zigbee2mqtt.

Wikibear, 2023-01-03:
https://github.com/Koenkk/zigbee2mqtt/issues/15164 check this.