Koenkk/zigbee2mqtt #14166: Tuya TV02 Thermostat (TS0601 _TZE200_hue3yfsn): Check for firmware update fails

station7, 2022-09-23:
### What happened?

The check for a firmware update of my TV02 (TS0601 _TZE200_hue3yfsn) fails ("No image available for imageType '5634"):

`Info 2022-09-23 18:42:14MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"gaestezimmer/trv/eins"},"error":"Failed to check if update available for 'gaestezimmer/trv/eins' (No image available for imageType '5634')","status":"error","transaction":"1xw8k-2"}'
Error 2022-09-23 18:42:14Failed to check if update available for 'gaestezimmer/trv/eins' (No image available for imageType '5634')`

A date or version of the **current firmware** is not displayed either.

[image]

### What did you expect to happen?

Update firmware or report that no firmware update is available or do not show device on OTA page, if firmware updates are not supported

### How to reproduce it (minimal and precise)

Try to update TV02 firmware via webfrontend

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

Slaesh's CC2652RB Stick

auanasgheps, 2022-10-04:
I have a similar issue but for TS0601_thermostat, different model than yours.

auanasgheps, 2022-10-04:
I realized what's going on.
Tuya platform does not have a public URL for ZigBee OTAs. They have to be manually sourced by the community using the original gateway and a proxy to intercept the OTA URL.

If you have the original gateway you can get the URL if you have the skills [to do so](https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html).
I have the skills but not the gateway... I'll try to get hold of one.

station7, 2022-10-04:
Hi, many thanks for following up. If I find the time I can try to get the OTA URL, but I am willing to sponsor a gateway for you, because I have another "[open_window-issue](https://github.com/Koenkk/zigbee2mqtt/issues/14163)" with my Tuya-thermostat and would appreciate if someone could look into this as well...

staastas, 2022-10-04:
https://github.com/Koenkk/zigbee-OTA/blob/master/index.json

```

    {
        "fileVersion": 87,
        "fileSize": 234174,
        "manufacturerCode": 4098,
        "manufacturerName": [
            "_TZE200_ckud7u2l",
            "_TZE200_cwnjrr72",
            "_TZE200_ywdxldoj"
        ],
        "imageType": 5634,
        "sha512": "1113486a67403d922fc4a462b7477e0a0986fa8862a89b74ab8c5eac7b4ecb31c30f1da9ae1bbc1d6d50ae1fed66d38c37a1834ab71fd384c0555002783aba13",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Tuya/1615190575-si32_zg_uart_connect_sleep_ZS5_ty_OTA_1.1.7.bin",
        "path": "images/Tuya/1615190575-si32_zg_uart_connect_sleep_ZS5_ty_OTA_1.1.7.bin"
    },
```

staastas, 2022-10-04:
maybe this information will help you. let me know if you manage to update, I also have the same problem(

https://github.com/Koenkk/zigbee-OTA/blob/master/index.json

```

    {
        "fileVersion": 87,
        "fileSize": 234174,
        "manufacturerCode": 4098,
        "manufacturerName": [
            "_TZE200_ckud7u2l",
            "_TZE200_cwnjrr72",
            "_TZE200_ywdxldoj"
        ],
        "imageType": 5634,
        "sha512": "1113486a67403d922fc4a462b7477e0a0986fa8862a89b74ab8c5eac7b4ecb31c30f1da9ae1bbc1d6d50ae1fed66d38c37a1834ab71fd384c0555002783aba13",
        "url": "https://github.com/Koenkk/zigbee-OTA/raw/master/images/Tuya/1615190575-si32_zg_uart_connect_sleep_ZS5_ty_OTA_1.1.7.bin",
        "path": "images/Tuya/1615190575-si32_zg_uart_connect_sleep_ZS5_ty_OTA_1.1.7.bin"
    },
```

auanasgheps, 2022-10-04:
Mine is _TZE200_cpmgn2cf which in theory is not in that list, but I appreciate the effort!

To be honest I'm not going to force that OTA.

I'm also not sure we should actually go down this route: I am setting up Better Thermostat and in [this old discussion](https://github.com/KartoffelToby/better_thermostat/issues/41#issuecomment-978094996) they do not advice to upgrade this device's firmware. 
I still have to test if it's ok (winter season is not yet here) but if the valve works I'm not going to pursue this route, and hopefully i'll just live with these TRVs "as is".

auanasgheps, 2022-10-04:
@station7 it's not clear to me why your valve is identified by Z2M as TV02, which is another model, but the code is still TS0601 as mine.
Wondering if there's something wrong there.

mkormout, 2022-10-16:
Although zigbee2mqtt is better than ZHA so in the case of hue3yfsn the implementation is slightly better in ZHA. I think you could get some inspiration there. I also have a couple of these heads.

For example, ZHA allows you to monitor if the internal thermostat is on or not. The device in HA then contains nonsense modes that TRV doesn't support (cooling, drying, fan only).

I would also like it if it could control or at least monitor the valve opening range.

If I can help in any way to fix hue3yfsn, please let me know, I'll be happy to provide logs or whatever needs to be debugged.

vTomTom, 2022-11-06:
Anything new about this ? Also Version _TZE200_hue3yfsn ?
Something we can do to help to find a solution? Currently I have no Tuya Gateway.
I‘m looking for a new firmware because of some delayed actions of the thermostat itself.

Lipown, 2022-12-04:
Any news on this please? :/ The valves are acting really weird :/

staastas, 2023-01-04:
Not solved

Wikibear, 2023-03-10:
Stale block. It isn't implemented yet.

Koenkk, 2023-03-11:
I cannot fix this, someone needs to grab the OTA: https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html

IceCupe123, 2023-03-17:
need to wait on update currently there is no new update , so i cant grab the url

Wikibear, 2023-04-17:
Push

Wikibear, 2023-05-19:
Push

Schnittcher, 2023-05-30:
@Koenkk, I want to help, but I can`t download the apk file (https://aapks.com/download.php?dl=48267761).
I get 502 Bad Gateway.
Can you send me the right version?

auanasgheps, 2023-06-12:
Hey, have you actually completed the work?

My Valves `_TZE200_cpmgn2cf` are reporting that there's an update, they never did, so someone has grabbed the URL for us. 

[image]

Wikibear, 2023-06-13:
No reporting for OTA with _TZE200_hue3yfsn

Wikibear, 2023-07-15:
1.32.1-dev commit: [af0521b5](https://github.com/Koenkk/zigbee2mqtt/commit/af0521b5)
No reporting for OTA with _TZE200_hue3yfsn

Wikibear, 2023-08-16:
1.32.2-dev commit: [3736b590](https://github.com/Koenkk/zigbee2mqtt/commit/3736b590)
Update not possible

Wikibear, 2023-09-17:
Push no update available

bolzerrr, 2023-09-29:
Also stumbled into this issue, i get such report:
Failed to check if update available for 'OG_BAD_H_RECHTS' (Device didn't respond to OTA request)

Device:
Zigbee Model
    TS0601
Zigbee Manufacturer
    _TZE200_hue3yfsn

I found a bunch of other threads, but cant see any solution. If i am wrong maybe someone can point me towards the right direction?

Marcwa19197, 2023-10-06:
> Also stumbled into this issue, i get such report: Failed to check if update available for 'OG_BAD_H_RECHTS' (Device didn't respond to OTA request)
>
> Device: Zigbee Model TS0601 Zigbee Manufacturer [...]

Im also looking for a solution to update the Firmware.

Using:
TS0601
von _TZE200_hue3yfsn

via standard ZHA integration (Using SkyConnect Stick in HA). Thanks.

Wikibear, 2023-11-06:
Push

Lipown, 2023-11-07:
Any way how to update manually pls?

oneofthemany, 2024-02-04:
> > Also stumbled into this issue, i get such report: Failed to check if update available for 'OG_BAD_H_RECHTS' (Device didn't respond to OTA request)
> > 
> > Device: Zigbee Model TS0601 Zigbee Manufacturer _TZE200_hue3yfsn
> > 
> > I found a bunch of other threads, but cant see any solution. If i am wrong maybe someone can point me towards the right direction?
> 
> Im also looking for a solution to update the Firmware.
> 
> Using:
> TS0601
> _TZE200_hue3yfsn
> 
> via standard ZHA integration (Using SkyConnect Stick in HA). Thanks.
> 
> My main problem is, that if the Climates are "off" the valve is still opened, so the heater gets hot.
> Also see: https://community.home-assistant.io/t/ts0601-unusual-switch-and-functions/535914/7

Have you seen this thread:

https://community.home-assistant.io/t/tuya-tv02-trv-valve-not-closing-completely/406707

Has worked for me

Wikibear, 2024-02-05:
Nice to know but the OTA issue stay.

oneofthemany, 2024-02-05:
> Nice to know but the OTA issue stay. 

https://github.com/Koenkk/zigbee2mqtt/issues/21084

I've also got this ticket open

Wikibear, 2024-08-04:
Push it isn't solved yet.

oneofthemany, 2025-01-07:
I've just had all my trv02 fail but 1. I've now moved to Aqara which do exactly what I need. Id stay away from these as lack of auto calibration can lead to motor failure

Wikibear, 2025-03-09:
Push again

Wikibear, 2025-05-09:
Push

Wikibear, 2025-07-09:
Push

Wikibear, 2025-09-08:
Push

closed 2025-11-16 by github-actions[bot]