Koenkk/zigbee2mqtt #14172: New Device Support: Smart Plug 16A (NAS-WR01B) model 'TS011F' and manufacturer name '_TZ3000_gjnozsaz'

tabletka-ao, 2022-09-23:
### Link

https://www.rtx24.pl/gniazdko-wtyczka-16a-inteligentna-rtx-zigbee-tuya-p-588.html

### Database entry

{ 	"id": 2, 	"type": "Router", 	"ieeeAddr": "0xa4c138375c26bb8a", 	"nwkAddr": 28064, 	"manufId": 4417, 	"manufName": "_TZ3000_gjnozsaz", 	"powerSource": "Mains (single phase)", 	"modelId": "TS011F", 	"epList": [ 		1, 		242 	], 	"endpoints": { 		"1": { 			"profId": 260, 			"epId": 1, 			"devId": 266, 			"inClusterList": [ 				3, 				4, 				5, 				6, 				1794, 				2820, 				57345, 				57344, 				0 			], 			"outClusterList": [ 				25, 				10 			], 			"clusters": { 				"genBasic": { 					"attributes": { 						"modelId": "TS011F", 						"manufacturerName": "_TZ3000_gjnozsaz", 						"powerSource": 1, 						"zclVersion": 3, 						"appVersion": 74, 						"stackVersion": 0, 						"hwVersion": 1, 						"dateCode": "" 					} 				}, 				"genOnOff": { 					"attributes": { 						"20480": 1, 						"32768": 0, 						"onOff": 0, 						"onTime": 0, 						"offWaitTime": 0, 						"tuyaBacklightMode": 1, 						"moesStartUpOnOff": 2 					} 				} 			}, 			"binds": [ 				{ 					"cluster": 25, 					"type": "endpoint", 					"deviceIeeeAddress": "0x00124b001cd47e50", 					"endpointID": 1 				} 			], 			"configuredReportings": [], 			"meta": {} 		}, 		"242": { 			"profId": 41440, 			"epId": 242, 			"devId": 97, 			"inClusterList": [], 			"outClusterList": [ 				33 			], 			"clusters": {}, 			"binds": [], 			"configuredReportings": [], 			"meta": {} 		} 	}, 	"appVersion": 74, 	"stackVersion": 0, 	"hwVersion": 1, 	"dateCode": "", 	"zclVersion": 3, 	"interviewCompleted": true, 	"meta": {}, 	"lastSeen": 1663961659711 }

### Comments

I try to connect plug via Zigbee2mqtt but logged that unsupported device model 'TS011F' and manufacturer name '_TZ3000_gjnozsaz'

Koenkk, 2022-09-25:
`_TZ3000_gjnozsaz` should be supported in z2m 1.27.2

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

tabletka-ao, 2022-09-25:
Hello,
Thanks for info. I missed change the version of z2m. Now plug works correctly.

closed 2022-09-25 by tabletka-ao