Koenkk/zigbee2mqtt #14173: After each HA restart, Z2M cannot connect until I restart MQTT manually

dexonter, 2022-09-23:
### What happened?

After each restart Z2M fails to start (502 bad gateway)
```
Zigbee2MQTT:info  2022-09-24 01:42:26: Connecting to MQTT server at mqtt://core-mosquitto:1883
Zigbee2MQTT:error 2022-09-24 01:42:26: MQTT failed to connect: Connection refused: Not authorized
```
Once I restart MQTT manually, everything works.

This makes restarting HA a pain....

### What did you expect to happen?

I expect Zigbee2MQTT to work after each HA restart

### How to reproduce it (minimal and precise)

No clue...

### Zigbee2MQTT version

1.27.2-1

### Adapter firmware version

20210708

### Adapter

Sonoff 3.0 Dongle Plus (the P model)

### Debug log

s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[01:42:18] INFO: Socat not enabled
[01:42:19] INFO: Handing over control to Zigbee2MQTT Core ...
Zigbee2MQTT:info  2022-09-24 01:42:25: Logging to console and directory: '/config/zigbee2mqtt/log/2022-09-24.01-42-25' filename: log.txt
Zigbee2MQTT:info  2022-09-24 01:42:25: Starting Zigbee2MQTT version 1.27.2 (commit #unknown)
Zigbee2MQTT:info  2022-09-24 01:42:25: Starting zigbee-herdsman (0.14.53)
Zigbee2MQTT:info  2022-09-24 01:42:26: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-09-24 01:42:26: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2022-09-24 01:42:26: Currently 0 devices are joined:
Zigbee2MQTT:info  2022-09-24 01:42:26: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-09-24 01:42:26: Connecting to MQTT server at mqtt://core-mosquitto:1883
Zigbee2MQTT:error 2022-09-24 01:42:26: MQTT failed to connect: Connection refused: Not authorized
Zigbee2MQTT:error 2022-09-24 01:42:26: Exiting...
Zigbee2MQTT:info  2022-09-24 01:42:26: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-09-24 01:42:36: Stopped zigbee-herdsman
[01:42:37] INFO: Handing over control to Zigbee2MQTT Core ...
Zigbee2MQTT:info  2022-09-24 01:42:40: Logging to console and directory: '/config/zigbee2mqtt/log/2022-09-24.01-42-40' filename: log.txt
Zigbee2MQTT:info  2022-09-24 01:42:40: Starting Zigbee2MQTT version 1.27.2 (commit #unknown)
Zigbee2MQTT:info  2022-09-24 01:42:40: Starting zigbee-herdsman (0.14.53)
Zigbee2MQTT:info  2022-09-24 01:42:41: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-09-24 01:42:41: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2022-09-24 01:42:41: Currently 0 devices are joined:
Zigbee2MQTT:info  2022-09-24 01:42:41: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-09-24 01:42:41: Connecting to MQTT server at mqtt://core-mosquitto:1883
Zigbee2MQTT:error 2022-09-24 01:42:41: MQTT failed to connect: Connection refused: Not authorized
Zigbee2MQTT:error 2022-09-24 01:42:41: Exiting...
Zigbee2MQTT:info  2022-09-24 01:42:41: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-09-24 01:42:51: Stopped zigbee-herdsman

ciB89, 2022-09-24:
I have experienced the same issue.

AlexMKX, 2022-09-25:
It is a HA problem. Other MQTT appliances (openmqttgw) are unable to connect Mosquitto untill broker restart.
It started 2 weeks ago after mqtt update I suppose

Geoffrey54, 2022-09-27:
> It is a HA problem. Other MQTT appliances (openmqttgw) are unable to connect Mosquitto untill broker restart. It started 2 weeks ago after mqtt update I suppose

What can we do with that ? Mosquitto broker is unable to start at all since this week for me in HA 2022.9.6. What are the solutions available (except waiting a fix in HA) ?
Thanks !

ashleysommer, 2022-09-28:
I'm new to HA, installed latest hassos and using HA v2022.9.7. I'm trying to set up zigbee2mqtt to talk to Mosquito for the first time, and running into this problem.

extra username and password are defined in HA, and configuration for z2m is correct.

z2m logs giving me:
```
MQTT failed to connect: Connection refused: Not authorized
```

mosquito logs giving me:
```
2022-09-28 19:34:10: New connection from 172.30.32.1:49388 on port 1883.
error: received null username or password for unpwd check
2022-09-28 19:34:09: Client <unknown> disconnected, not authorised.
```

I honestly don't know if this is a problem with z2m, with Mosquito, or with HA.

Koenkk, 2022-09-28:
> It is a HA problem. Other MQTT appliances (openmqttgw) are unable to connect Mosquitto untill broker restart. It started 2 weeks ago after mqtt update I suppose

@AlexMKX is there already a HA issue open? If not please create it.

closed 2022-11-05 by github-actions[bot]