Koenkk/zigbee2mqtt #14176: Develco SPLZB-132 - incorrect power (wattage) reporting

f432, 2022-09-24:
### What happened?

Power reporting is incorrect for this device

[image]

So 234.13 x 0.12 = 28.095W

So a /1000 error ?

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20210708

### Adapter

Slaesh CC2652rb stick

Koenkk, 2022-09-25:
@NameLessJedi does this device still work correctly for you?

NameLessJedi, 2022-10-03:
Actually I did not check this that carefully till now.
One thing is that reporting of each value might be taken at different point of time so the value might not match.
@f432 please check the "Reporting" tab of device and see how (and if) rep intervals and min rep change are set.
Put some constant load on this device and check again.
I'll try to check this too

f432, 2022-10-11:
Sorry about delay, been away for work. Using a tumble dryer i get a constant reading but not 100% correct. Reporting is default

[image]

Below is the reporting for all devices incl where I am getting a 0.1 W reading. No Min rep change setting for "seMetering" if that makes a difference 

[image]

closed 2022-11-19 by github-actions[bot]