Koenkk/zigbee2mqtt #14179: Moes BRT-100-TRV local temperature calibration

willhaggan, 2022-09-24:
### What happened?

the local temperature calibration has no affect on the loical temperature reading

### What did you expect to happen?

an offset of the local temperature reading

### How to reproduce it (minimal and precise)

add a value between -9 and  +9 to the local temperature calibraion value which has no affect non the local temperature reading

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

    0x26580700

### Adapter

ConBee2/RaspBee2

callacomp, 2022-10-11:
I've given up on this. I've tried all of the blueprints, better thermostat, I've written my own, with numerous calculations on changed time, etc and I've just made sure the temp sensor is pointed away from the radiator, put a temp sensor near it and set manual calibrations per TRV and used schedy, not 100% but seems to be working

Koenkk, 2022-11-10:
Should be fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/4917

closed 2022-11-10 by Koenkk