Koenkk/zigbee2mqtt #14183: Zigbee2MQTT shows hundreds of unknown devices

and7ey, 2022-09-25:
### What happened?

The main page of Zigbee2MQTT HA addon shows hundreds of unknown devices:
[image]
I would like to understand why they appeared and how to delete them in bulk (clicking 'delete', then 'force remove' one by one will take a lot of time).

### Zigbee2MQTT version

1.27.2 commit: unknown

### Adapter firmware version

6.7.10.0 build 423

### Adapter

EZSP v8 (xiaomi gateway 3)

sergeymatsay, 2022-09-26:
After updating to 1.27.2, all tuya ts0011 relays stopped working.  when trying to paring gives the same error.
[image]

marwonn, 2022-09-26:
> ### What happened?
> The main page of Zigbee2MQTT HA addon shows hundreds of unknown devices: [image] I would like to understand why they appeared and how to delete them in bulk (clicking 'delete', [...]

Same here. Dozens of unknown devices since a few days.

Koenkk, 2022-09-28:
@kirovilya could you check this?

kirovilya, 2022-09-29:
@marwonn @sergeymatsay @and7ey Please check on the latest/edge dev version of z2m - changes have been made to improve device pairing.
But! You need to create a new network! To do this, you should change the channel or panId - in this case, the list of devices will be cleared and you will have to pair the devices again.

and7ey, 2022-09-29:
Thanks. I've just deleted unknown devices manually from .db file. Works well now.

closed 2022-11-07 by github-actions[bot]