Koenkk/zigbee2mqtt #14184: Lidl HG08164 turns on two xiaomi smart plug on its own

mr2c12, 2022-09-25:
### What happened?

When I single click the Lidl HG08164 (https://www.zigbee2mqtt.io/devices/HG08164.html#lidl-hg08164), two other devices (both are Xiaomi ZNCZ02LM) turn on automagically!

### What did you expect to happen?

No automatic action.

### How to reproduce it (minimal and precise)

Pair and click :)

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20210708

### Adapter

Texas Instruments LAUNCHXL-CC26X2R1 #

### Debug log

> Debug 2022-09-25 10:00:19Received Zigbee message from 'lidl_button_fridge', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
> Info 2022-09-25 10:00:19MQTT publish: topic 'zigbee2mqtt/lidl_button_fridge', payload '{"action":"on","battery":null,"last_seen":"2022-09-25T08:00:19.401Z","linkquality":105}'
> Info 2022-09-25 10:00:19MQTT publish: topic 'zigbee2mqtt/lidl_button_fridge', payload '{"action":"","battery":null,"last_seen":"2022-09-25T08:00:19.401Z","linkquality":105}'
> Info 2022-09-25 10:00:19MQTT publish: topic 'zigbee2mqtt/lidl_button_fridge/action', payload 'on'
> Debug 2022-09-25 10:00:19Received Zigbee message from 'mijia_plug_salamusica_ng', type 'attributeReport', cluster 'genOnOff', data '{"61440":127738405,"onOff":1}' from endpoint 1 with groupID 0
> Debug 2022-09-25 10:00:19Received Zigbee message from 'mijia_plug_camera', type 'attributeReport', cluster 'genOnOff', data '{"61440":127738405,"onOff":1}' from endpoint 1 with groupID 0
> Debug 2022-09-25 10:00:19Received Zigbee message from 'mijia_plug_cameretta', type 'attributeReport', cluster 'genOnOff', data '{"61440":127738405,"onOff":1}' from endpoint 1 with groupID 0
> Info 2022-09-25 10:00:19MQTT publish: topic 'zigbee2mqtt/mijia_plug_salamusica_ng', payload '{"consumption":49.33,"device_temperature":31,"energy":49.33,"last_seen":"2022-09-25T08:00:19.474Z","linkquality":78,"power":0,"power_outage_count":38,"power_outage_memory":false,"state":"ON","temperature":34,"update":{"state":null},"update_available":null}'
> Info 2022-09-25 10:00:19MQTT publish: topic 'zigbee2mqtt/mijia_plug_camera', payload '{"consumption":1.8,"device_temperature":31,"energy":1.8,"last_seen":"2022-09-25T08:00:19.482Z","linkquality":105,"power":0,"power_outage_count":19,"power_outage_memory":true,"state":"ON","temperature":31,"update":{"state":"idle"},"update_available":false}'
> Info 2022-09-25 10:00:19MQTT publish: topic 'zigbee2mqtt/mijia_plug_cameretta', payload '{"consumption":18.72,"device_temperature":33,"energy":18.72,"last_seen":"2022-09-25T08:00:19.492Z","linkquality":66,"power":0,"power_outage_count":61,"state":"ON","temperature":35,"update":{"state":"idle"},"update_available":false,"voltage":null}'
> Debug 2022-09-25 10:00:20Received Zigbee message from 'mijia_plug_cameretta', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":691.2000122070312}' from endpoint 2 with groupID 0
> Info 2022-09-25 10:00:20MQTT publish: topic 'zigbee2mqtt/mijia_plug_cameretta', payload '{"consumption":18.72,"device_temperature":33,"energy":18.72,"last_seen":"2022-09-25T08:00:20.509Z","linkquality":66,"power":691.2,"power_outage_count":61,"state":"ON","temperature":35,"update":{"state":"idle"},"update_available":false,"voltage":null}'
> Debug 2022-09-25 10:00:21Received Zigbee message from 'mijia_plug_cameretta', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":933.1199951171875}' from endpoint 2 with groupID 0
> Info 2022-09-25 10:00:21MQTT publish: topic 'zigbee2mqtt/mijia_plug_cameretta', payload '{"consumption":18.72,"device_temperature":33,"energy":18.72,"last_seen":"2022-09-25T08:00:21.514Z","linkquality":63,"power":933.12,"power_outage_count":61,"state":"ON","temperature":35,"update":{"state":"idle"},"update_available":false,"voltage":null}'

mr2c12, 2022-10-26:
Unfortunately this is still present.

The button turns on (with single click) or off (with double click) every ZNCZ02LM in my network.

Even with latest version (28).

MattWestb, 2022-10-26:
Light controller is sending commands to lights and if not have configure them they is normally using group 0x0000 for doing it.
Eases is removing the group 0 from your light so they is not listening to commands from group 0 and all shall working OK.

mr2c12, 2022-10-26:
@MattWestb maybe a very stupid question, but how do you remove a light from a group? i don't see such an option in the web gui!

MattWestb, 2022-10-26:
I dont knowing how to do it in Z2M.
In the Zgbee layer is command `remove all groups` or `remove group` `0x00` shall doing it.
In ZHA you can doing in the `manager Zigbee device` from the device card.

thomashogema, 2022-11-21:
Hi,

I'm experiencing the same behavior. The Lidl HG08164 (Silvercrest) button turns on my Xiami ZNCZ02LM without any configuration. The behavior is similar to that reported above: a single click turns it on, while a double click turns it off. This corresponds to the commands sent by the Lidl button (`on` on single and `off` on double), but not to any setup I did in HA.

I'm not familiar with Zigbee groups, but the groups page on the Z2M frontend appears to indicate there are no groups configured:
[image]

**Zigbee 2 MQTT version**
1.28.2-1

**Adapter**
ConBee II

P4tje, 2022-11-30:
Had the same issue with this switch and a Xiaomi curtain motor (ZNCLDJ11LM). Fixed it by sending "zigbee2mqtt/bridge/request/group/members/remove_all" with payload {"device":"affected device name"}.

https://www.zigbee2mqtt.io/guide/usage/groups.html#configuration

mr2c12, 2022-12-11:
I can confirm @P4tje solution do work. Thank you man!

closed 2022-12-11 by mr2c12

mr2c12, 2022-12-17:
Unfortunately this solution is not permanent. The command should be sent again after the affected devices are powered off and on again.

@P4tje have you been able to find a "fix" to make this permanent?