Koenkk/zigbee2mqtt #14188: Legrand 067724 always starting new pairing

Tonio16, 2022-09-25:
### What happened?

Hello

I updated a Legrand 067724 wireless double switch to the last firmware with legrand gateway, new firmware is 31. I wanted to add it back to zigbee2mqtt but I have a strange behavior. Each time I am using it, it's starting a new pairing, though the switch isn't in pairing mode and z2m isn't in pairing mode too.

Since it's pairing each time I am using it, I have no update of the status, for example toggle_left, _right.
I tried to reset it several times, without success.

My z2m:
Version de Zigbee2MQTT     [1.27.2] commit: [7dc48fb6]

Type de coordinateur     zStack3x0

Révision du coordinateur    20220219

### What did you expect to happen?

Pairing once and when both z2m and device are in pairing mode.

### How to reproduce it (minimal and precise)

I supposed with the device and firmware.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

Sonoff CC2652P

lightelfua, 2022-10-05:
Use this:
https://zigbee.blakadder.com/Legrand_067774.html

jiquem, 2022-11-02:
hello @Tonio16, did you find a solution ? I am facing the same issue and reintegrating the device doesn't fix it

jiquem, 2022-11-02:
Thanks @Tonio16 : what is 73 ?

Tonio16, 2022-11-02:
067773
The single switch.

jiquem, 2022-11-02:
unfortunately not

closed 2022-12-10 by github-actions[bot]