Koenkk/zigbee2mqtt #14190: [New device support]: Tuya Motion Sensor Pro with Scene Switch Linkoze LKMSZ001

Blacksli, 2022-09-25:
### Link

https://expo.tuya.com/product/1078025

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c138aef8dc64b0","nwkAddr":6222,"manufId":4417,"manufName":"_TZ3210_cwamkvua","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[1,1280,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x84b4dbfffeefd0c8","zoneState":1}},"genPowerCfg":{"attributes":{"batteryVoltage":26,"batteryPercentageRemaining":20}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffeefd0c8","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffeefd0c8","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1443710779},"lastSeen":1664127447413,"defaultSendRequestWhen":"immediate"}

### Comments

[image]

The device has a light sensor that shows either bright or dark (no numbers). I don't know what to call this sensor. Any ideas?

dark:
`Received Zigbee message from '0xa4c138aef8dc64b0', type 'commandActiveStatusReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":102}],"seq":11776}' from endpoint 1 with groupID 0`

bright:
`Received Zigbee message from '0xa4c138aef8dc64b0', type 'commandActiveStatusReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":102}],"seq":12032}' from endpoint 1 with groupID 0`

and can you optimize the converter?

[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const fzLocal = {
    MOTIONPRO: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
			return {
                occupancy: (zoneStatus & 1) > 0,
            };
        },
    },
	 
	MOTIONPRO1: {
        cluster: 'manuSpecificTuya',
        type: 'commandActiveStatusReport',
        convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'MOTIONPRO1');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
            meta.logger.debug(`from MOTIONPRO1, dp=[${dp}], datatype=[${dpValue.datatype}], value=[${value}]`);
			switch (dp) {
            case 102:
				return {light_state: {false: 'dark', true: 'bright'}[value]};
             case 101: 
                 return {action: {2: 'hold', 1: 'double', 0: 'single'}[value]};
            default: 
                meta.logger.warn('fz.MOTIONPRO1: Unhandled DP #${dp}: ${JSON.stringify(dpValue)}');
            };
		},
    },
};

const definition = {
	
[... 16 lines cut ...]
		fzLocal.MOTIONPRO1,
    ],
    toZigbee: [
       //tz.tuya_data_point_test, // Another debug converter
    ],
    //onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
        //await reporting.bind(endpoint, coordinatorEndpoint, ['ssIasZone']);
		await endpoint.read('genPowerCfg', ['batteryVoltage', 'batteryPercentageRemaining']);
        await reporting.batteryPercentageRemaining(endpoint);
		await reporting.batteryVoltage(endpoint);
    },
    exposes: [
		e.battery(), 
		e.battery_voltage(),
		e.occupancy(),
		e.action(['single', 'double', 'hold']),
		exposes.enum('light_state', ea.STATE,['dark', 'bright']),
    ],
};

module.exports = definition;
```

Koenkk, 2022-09-28:
The converter looks good, what is still missing? Otherwise I can integrate it

Blacksli, 2022-10-13:
I don't know if fz.ignore_basic_report is needed or not

Koenkk, 2022-10-18:
Can you update to the latest z2m dev (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html), force remove the device through z2m frontend, restart z2m, use [this](https://gist.github.com/Koenkk/3cc04fe750fd146ce8c5b42c79ec6ddb) external converter, start z2m & repair device and see if everything still works?

brunopiras, 2022-10-30:
Thank you man, works very well!!

Koenkk, 2022-10-30:
Given this is a motion sensor, shouldn't it also be able to detect occupancy?

brunopiras, 2022-10-30:
Yes, it does!

Occupancy (motion it works well)
Light (works)
Action (it works on all three mode)

Koenkk, 2022-10-30:
I see, added it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-30 by Koenkk

nielsnieland2, 2022-11-08:
It doesnt work for me, only the PIR motion detection works in home assistant, button doesnt work, any idea Koen? @Koenkk

Koenkk, 2022-11-09:
@nielsnieland2 
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0202', ['_TZ3210_cwamkvua']),
    model: 'TS0202_2',
    vendor: 'TuYa',
    description: 'Motion sensor with scene switch',
    fromZigbee: [tuya.fzDataPoints, fz.ias_occupancy_alarm_1, fz.battery],
    toZigbee: [tuya.tzDataPoints],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.batteryVoltage(endpoint);
    },
    whiteLabel: [{vendor: 'Linkoze', model: 'LKMSZ001'}],
    exposes: [e.battery(), e.battery_voltage(), e.occupancy(), e.action(['single', 'double', 'hold']),
        exposes.enum('light', ea.STATE, ['dark', 'bright'])],
    meta: {
        tuyaDatapoints: [
            [102, 'light', tuya.valueConverterBasic.lookup({'dark': true, 'bright': false})],
            [101, 'action', tuya.valueConverterBasic.lookup({'single': 0, 'hold': 1, 'double': 2})],
        ],
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

AliakseiT, 2022-11-27:
@Koenkk, I have the same problem as @nielsnieland2 and tried your recommendation. It didn't help, but it based on logs I suspect its mapping issue. The logs with action (go dark, go bright, single tap, double tap) are below.
I've also noticed that my device does not expose `hold`. Instead `1` in `{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":102}]` corresponds to double tap. Also dark maps to `0`, and bright to `1`. But here I'm guessing, I don't understand how exactly the `valueConverterBasic` and `tuyaDatapoints` perform the mapping.
```
Debug 2022-11-27 02:58:58Received Zigbee message from 'Tuya occupancy sensor and switch', type 'commandActiveStatusReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":102}],"seq":40192}' from endpoint 1 with groupID 0
Debug 2022-11-27 02:58:58No converter available for 'TS0202_2' with cluster 'manuSpecificTuya' and type 'commandActiveStatusReport' and data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":102}],"seq":40192}'
Debug 2022-11-27 02:58:59Received Zigbee message from 'Tuya occupancy sensor and switch', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":5}' from endpoint 1 with groupID 0
Info 2022-11-27 02:58:59MQTT publish: topic 'zigbee2mqtt/Tuya occupancy sensor and switch', payload '{"action":null,"battery":10,"battery_low":false,"light":null,"linkquality":68,"occupancy":true,"tamper":true,"voltage":2600}'
Debug 2022-11-27 02:59:00Received Zigbee message from 'Tuya occupancy sensor and switch', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":5}' from endpoint 1 with groupID 0
Info 2022-11-27 02:59:00MQTT publish: topic 'zigbee2mqtt/Tuya occupancy sensor and switch', payload '{"action":null,"battery":10,"battery_low":false,"light":null,"linkquality":100,"occupancy":true,"tamper":true,"voltage":2600}'
Debug 2022-11-27 02:59:00Received Zigbee message from 'Tuya occupancy sensor and switch', type 'commandActiveStatusReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":102}],"seq":40448}' from endpoint 1 with groupID 0
Debug 2022-11-27 02:59:00No converter available for 'TS0202_2' with cluster 'manuSpecificTuya' and type 'commandActiveStatusReport' and data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":102}],"seq":40448}'
Debug 2022-11-27 02:59:00Received Zigbee message from 'Tuya occupancy sensor and switch', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":5}' from endpoint 1 with groupID 0
Info 2022-11-27 02:59:00MQTT publish: topic 'zigbee2mqtt/Tuya occupancy sensor and switch', payload '{"action":null,"battery":10,"battery_low":false,"light":null,"linkquality":184,"occupancy":true,"tamper":true,"voltage":2600}'
Debug 2022-11-27 02:59:03Received Zigbee message from 'Tuya occupancy sensor and switch', type 'commandActiveStatusReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":101}],"seq":40704}' from endpoint 1 with groupID 0
Debug 2022-11-27 02:59:03No converter available for 'TS0202_2' with cluster 'manuSpecificTuya' and type 'commandActiveStatusReport' and data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":101}],"seq":40704}'
Debug 2022-11-27 02:59:07Received Zigbee message from 'Tuya occupancy sensor and switch', type 'commandActiveStatusReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":101}],"seq":40960}' from endpoint 1 with groupID 0
Debug 2022-11-27 02:59:07No converter available for 'TS0202_2' with cluster 'manuSpecificTuya' and type 'commandActiveStatusReport' and data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":101}],"seq":40960}
```

Koenkk, 2022-11-27:
@AliakseiT I've pushed a fix, can you try with the latest-dev? With this the `No converter available for` should not appear anymore.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

AliakseiT, 2022-11-27:
Thanks a lot, @Koenkk! It works much better now, only two minor issues to correct:
1) flipped dark and bright states
2) single/double taps to be corrected for UI exposure

[image]

Logs of for interactions (go dark, go bright, single tap, double tap)
```
2022-11-27 16:30:37Received Zigbee message from 'Tuya occupancy sensor and switch', type 'commandActiveStatusReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":102}],"seq":21761}' from endpoint 1 with groupID 0
Info 2022-11-27 16:30:37MQTT publish: topic 'zigbee2mqtt/Tuya occupancy sensor and switch', payload '{"action":null,"battery":6,"battery_low":false,"light":"bright","linkquality":92,"occupancy":true,"tamper":true,"voltage":2500}'
Debug 2022-11-27 16:30:37Received Zigbee message from 'Tuya occupancy sensor and switch', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":5}' from endpoint 1 with groupID 0
Info 2022-11-27 16:30:37MQTT publish: topic 'zigbee2mqtt/Tuya occupancy sensor and switch', payload '{"action":null,"battery":6,"battery_low":false,"light":"bright","linkquality":88,"occupancy":true,"tamper":true,"voltage":2500}'
Debug 2022-11-27 16:30:39Received Zigbee message from 'Tuya occupancy sensor and switch', type 'commandActiveStatusReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":102}],"seq":22017}' from endpoint 1 with groupID 0
Info 2022-11-27 16:30:39MQTT publish: topic 'zigbee2mqtt/Tuya occupancy sensor and switch', payload '{"action":null,"battery":6,"battery_low":false,"light":"dark","linkquality":172,"occupancy":true,"tamper":true,"voltage":2500}'
Debug 2022-11-27 16:30:39Received Zigbee message from 'Tuya occupancy sensor and switch', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":5}' from endpoint 1 with groupID 0
Info 2022-11-27 16:30:39MQTT publish: topic 'zigbee2mqtt/Tuya occupancy sensor and switch', payload '{"action":null,"battery":6,"battery_low":false,"light":"dark","linkquality":168,"occupancy":true,"tamper":true,"voltage":2500}'
Debug 2022-11-27 16:30:41Received Zigbee message from 'Tuya occupancy sensor and switch', type 'commandActiveStatusReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":101}],"seq":22273}' from endpoint 1 with groupID 0
Info 2022-11-27 16:30:41MQTT publish: topic 'zigbee2mqtt/Tuya occupancy sensor and switch', payload '{"action":"single","battery":6,"battery_low":false,"light":"dark","linkquality":188,"occupancy":true,"tamper":true,"voltage":2500}'
Info 2022-11-27 16:30:41MQTT publish: topic 'zigbee2mqtt/Tuya occupancy sensor and switch', payload '{"action":"","battery":6,"battery_low":false,"light":"dark","linkquality":188,"occupancy":true,"tamper":true,"voltage":2500}'
Info 2022-11-27 16:30:41MQTT publish: topic 'zigbee2mqtt/Tuya occupancy sensor and switch/action', payload 'single'
Debug 2022-11-27 16:30:43Received Zigbee message from 'Tuya occupancy sensor and switch', type 'commandActiveStatusReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":101}],"seq":22529}' from endpoint 1 with groupID 0
Info 2022-11-27 16:30:43MQTT publish: topic 'zigbee2mqtt/Tuya occupancy sensor and switch', payload '{"action":"double","battery":6,"battery_low":false,"light":"dark","linkquality":164,"occupancy":true,"tamper":true,"voltage":2500}'
Info 2022-11-27 16:30:43MQTT publish: topic 'zigbee2mqtt/Tuya occupancy sensor and switch', payload '{"action":"","battery":6,"battery_low":false,"light":"dark","linkquality":164,"occupancy":true,"tamper":true,"voltage":2500}'
Info 2022-11-27 16:30:43MQTT publish: topic 'zigbee2mqtt/Tuya occupancy sensor and switch/action', payload 'double'
```

Koenkk, 2022-11-27:
1. updated https://github.com/Koenkk/zigbee2mqtt/issues/14190#issuecomment-1309044926
2. if you have the home assistant integration enabled this is expected: https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty

AliakseiT, 2022-11-27:
Works 👍 
While testing I've confirmed that release after holding for >2 seconds does issue hold. But the payload is `2` (`1` is `double`). So the `tuyaDatapoint` 101 shall be:
`[101, 'action', tuya.valueConverterBasic.lookup({'single': 0, 'double': 1, 'hold': 2})],`

Koenkk, 2022-11-28:
Thanks, pushed the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

nielsnieland2, 2022-11-28:
It will be pushed on the normal Z2Mqtt on the first of December?

Koenkk, 2022-11-28:
Yes! Will be included in the 1 December release.

katlimero, 2023-10-21:
Hello, I am a beginner and don't know who to turn to for help. I have Zigbee2mqtt which I use with node-red. I just installed motion sensors TuYa TS0202_2 and I have a lot of error messages in the zigbee2mqtt logs.
This tends to slow down the general operation of my installation. can you help me?
[image]