Koenkk/zigbee2mqtt #14191: [New device support]: Philipps Hue White GU10 9290019535

GeckoSplinter, 2022-09-25:
### Link

https://www.philips-hue.com/fr-fr/p/hue-white-pack-de-1-gu10/8719514340060#specifications

### Database entry

{"id":22,"type":"Unknown","ieeeAddr":"0x0017880109dc32a1","nwkAddr":64036,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1664134380660,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,

I tried to pair it but I got unsupported device I cannot get the the ZigbeeModel information.
These are just simple spot GU10 white no color mode and bluetooth support.

So I guess the same  than this one : https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/philips.js#L893
Alias of https://www.zigbee2mqtt.io/devices/LWG004.html

### Supported color modes

none

### Color temperature range

none

danieledwardgeorgehitchcock, 2022-09-28:
Hi,

It looks like the binding / pairing / interview process for this device didn't work. Please could you retry pairing the device so that you can get a database entry that has an `"interviewCompleted":true` value, and post that up?

GeckoSplinter, 2022-09-29:
Hello thanks for answering.

I was trying to connect 3 GU10 at the same time I think the devices were conflicting with each others.
After your message I reset them and try to pair them again but one by one and it worked
They are seen as LWG004 that's perfect thanks

Sorry for disturbing you

closed 2022-09-29 by GeckoSplinter