Koenkk/zigbee2mqtt #14193: Moes MS-108ZR inverted states

aqwserf, 2022-09-26:
### What happened?

I just realized that states Open/Close are inverted on my 4 Moes MS-108ZR and therefore also in HA (while it was not the case when I initially configured it at the beginning of the year IIRC).
I tried to play with the parameter `invert_cover` without success with a cover fully opened:

- `inverted_cover` off:
`{
    "calibration_time": 28,
    "linkquality": 252,
    "motor_reversal": "OFF",
    "moving": "STOP",
    "position": 0,
    "state": "CLOSE"
}`
- `inverted_cover` on:
`{
    "calibration_time": 28,
    "linkquality": 252,
    "motor_reversal": "OFF",
    "moving": "STOP",
    "position": 100,
    "state": "CLOSE"
}`

I've also tried to turn off the power while manipulating the parameter and then to turn it on again without success either.
Posting an MQTT message to `Open` works until I manipulate the cover again.

The only way I found to make it work properly is to create a cover template in HA like this:

`value_template: "{% if is_state('cover.rollershutter_1', 'open') %} closed {% else %} open {% endif %}"`.

The space being very limited to access the module, I cannot try to reset it.

### What did you expect to happen?

The message to be `OPEN` when the shutter is opened and `CLOSE` when it's closed

### How to reproduce it (minimal and precise)

Add a Moes MS-108ZR in Zigbee2MQTT, open or close it and check states in the `State` panel

### Zigbee2MQTT version

1.27.2 

### Adapter firmware version

26720700

### Adapter

ConBee2

### Debug log

info 2022-09-26 11:13:39MQTT publish: topic 'zigbee2mqtt/Volet Bureau', payload '{"calibration_time":28,"linkquality":252,"motor_reversal":"OFF","moving":"STOP","position":0,"state":"CLOSE"}'
debug 2022-09-26 11:14:13Received MQTT message on 'zigbee2mqtt/Volet Bureau/set' with data 'CLOSE'
debug 2022-09-26 11:14:13Publishing 'set' 'state' to 'Volet Bureau'
debug 2022-09-26 11:14:13Received Zigbee message from 'Volet Bureau', type 'attributeReport', cluster 'closuresWindowCovering', data '{"tuyaMovingState":2}' from endpoint 1 with groupID null
info 2022-09-26 11:14:13MQTT publish: topic 'zigbee2mqtt/Volet Bureau', payload '{"calibration_time":28,"linkquality":255,"motor_reversal":"OFF","moving":"DOWN","position":0,"state":"CLOSE"}'
debug 2022-09-26 11:14:13Received Zigbee message from 'Volet Bureau', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100}' from endpoint 1 with groupID null
info 2022-09-26 11:14:13MQTT publish: topic 'zigbee2mqtt/Volet Bureau', payload '{"calibration_time":28,"linkquality":255,"motor_reversal":"OFF","moving":"DOWN","position":0,"state":"CLOSE"}'
debug 2022-09-26 11:14:14Received MQTT message on 'zigbee2mqtt/Volet Bureau/set' with data 'STOP'
debug 2022-09-26 11:14:14Publishing 'set' 'state' to 'Volet Bureau'
debug 2022-09-26 11:14:14Received Zigbee message from 'Volet Bureau', type 'attributeReport', cluster 'closuresWindowCovering', data '{"tuyaMovingState":1}' from endpoint 1 with groupID null
info 2022-09-26 11:14:14MQTT publish: topic 'zigbee2mqtt/Volet Bureau', payload '{"calibration_time":28,"linkquality":255,"motor_reversal":"OFF","moving":"STOP","position":0,"state":"CLOSE"}'
debug 2022-09-26 11:14:14Received Zigbee message from 'Volet Bureau', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":95}' from endpoint 1 with groupID null
info 2022-09-26 11:14:14MQTT publish: topic 'zigbee2mqtt/Volet Bureau', payload '{"calibration_time":28,"linkquality":255,"motor_reversal":"OFF","moving":"STOP","position":5,"state":"OPEN"}'
debug 2022-09-26 11:14:15Received MQTT message on 'zigbee2mqtt/Volet Bureau/set' with data 'OPEN'
debug 2022-09-26 11:14:15Publishing 'set' 'state' to 'Volet Bureau'
debug 2022-09-26 11:14:15Received Zigbee message from 'Volet Bureau', type 'attributeReport', cluster 'closuresWindowCovering', data '{"tuyaMovingState":0}' from endpoint 1 with groupID null
info 2022-09-26 11:14:15MQTT publish: topic 'zigbee2mqtt/Volet Bureau', payload '{"calibration_time":28,"linkquality":255,"motor_reversal":"OFF","moving":"UP","position":5,"state":"OPEN"}'
debug 2022-09-26 11:14:16Received Zigbee message from 'Volet Bureau', type 'attributeReport', cluster 'closuresWindowCovering', data '{"tuyaMovingState":1}' from endpoint 1 with groupID null
info 2022-09-26 11:14:16MQTT publish: topic 'zigbee2mqtt/Volet Bureau', payload '{"calibration_time":28,"linkquality":255,"motor_reversal":"OFF","moving":"STOP","position":5,"state":"OPEN"}'
debug 2022-09-26 11:14:16Received Zigbee message from 'Volet Bureau', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100}' from endpoint 1 with groupID null
info 2022-09-26 11:14:16MQTT publish: topic 'zigbee2mqtt/Volet Bureau', payload '{"calibration_time":28,"linkquality":255,"motor_reversal":"OFF","moving":"STOP","position":0,"state":"CLOSE"}'

Koenkk, 2022-09-28:
Integrated the changes. The HA webui only allows to configure the mqtt and serial, rest has to be configured via the z2m webui.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-28 by Koenkk

TTR6214, 2022-09-30:
I have also this device and I confirm that your proposed external converter fixed the issue.

friebi, 2022-10-02:
I have the same device twice and I was confused that after the update yesterday my shutters behave weird 😆 
I'm using my own configuration with node red.
Fair enough, the states are now more consistent. Maybe I would've preferred to remap the OPEN and CLOSED state 😕 which might have had less impact (at least on my end).
Nonetheless I have adapted my code in Nodered and it does now what it should do again.

dabohda, 2023-01-08:
I have the same device and can also confirm that this fixed the issue. Thx!