Koenkk/zigbee2mqtt #14198: Enocean Zigbee Switches PTM216Z stop working when permit join = false

t-o-o-m, 2022-09-26:
### What happened?

If I disable joining new devices my Enocean Zigbee Switches PTM216Z stop working after a while.

They immediately come back to life as soon as I permit joining again

This behaviour might be by design/necessity; with energy harvesting devices being weird entities and all?

### What did you expect to happen?

The Enocean Zigbee Switches PTM216Z should continue to work with disabled joining of new devices.

### How to reproduce it (minimal and precise)

Have Enocean Zigbee Switches PTM216Z in your Zigbee-network.

1. Disable joining
2. Wait a few hours
3. Enocean Zigbee Switches PTM216Z will stop working

1. Enable Joining
2. Enocean Zigbee Switches PTM216Z immediately start working again

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20200417

### Adapter

TI LAUNCHXL-CC26X2R1

t-o-o-m, 2022-10-27:
Unstale

t-o-o-m, 2022-11-28:
Unstale

t-o-o-m, 2022-11-28:
Unstale

t-o-o-m, 2022-12-29:
Unstale

t-o-o-m, 2023-02-05:
Unstale

t-o-o-m, 2023-03-08:
Still relevant

t-o-o-m, 2023-04-09:
Unstale

t-o-o-m, 2023-05-10:
Unstale

closed 2023-06-18 by github-actions[bot]

TjardE, 2023-07-30:
This issue happens to me too. I have at the moment 8 of those switches in use. They all work when I enable joining (either via the tickbox in settings or on the user interface). When I disable joining 2 of them stop working. The other 6 work without issue. I tried repairing and such but could not get them to work without enabling joining.

KlimekDE, 2023-08-06:
I'am also affected from the same problem. After some hours after disabling joining, the switches randomly stops working. After enabling the joining again, all switches start to work again.

TjardE, 2024-02-19:
I got it working this weekend. It seems that I had to join the switch via a specific device instead of Join All. Now everything works with the permit joing OFF in settings. It has to do with unicast or broadcast for Zigbee Green Power devices.
see https://github.com/Koenkk/zigbee2mqtt/issues/18464

t-o-o-m, 2025-01-05:
The newest update of Zigbee2MQTT includes the following change (see https://github.com/Koenkk/zigbee2mqtt/discussions/24198): 

> The option to permit joining forever has been removed; it is now limited to a maximum of 254 seconds.

So this issue might pop up for more people with Enocean Zigbee switches now.

Some of my switches stopped working today because of this.

I was able to resolve this quickly be rejoining my switches to the network - no need to switch to unicast on my side though (for details see the duplicated issue linked above, #18464 + the official docs: https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices)