Koenkk/zigbee2mqtt #14202: [New device support]: Philips Hue Tuar 1740447P0

Hurr1c4n3, 2022-09-26:
### Link

https://www.philips-hue.com/nl-be/p/hue-white-tuar-buitenwandlamp/1740447P0#specifications

### Database entry

{"id":58,"type":"Router","ieeeAddr":"0x0017880108c1cfcd","nwkAddr":39582,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"1740447P0","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64515,64514],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"1740447P0","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20211210","swBuildId":"1.93.7"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20211210","swBuildId":"1.93.7","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1664216771136,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the guide to add the new device myself but got stuck. I got the following:

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['1740447P0'],
    model: '1740447P0',
    vendor: 'Philips',
    description: 'Philips Wandlamp Hue Tuar', 
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
	extend: extend.light_onoff_brightness(),
};

module.exports = definition;
```

### Supported color modes

none

### Color temperature range

none

danieledwardgeorgehitchcock, 2022-09-28:
Hi,

If you can get the color temperature range for the device by following [this guide](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-1-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature) I can assist you with getting it added into the Z2M codebase

Hurr1c4n3, 2022-09-28:
It doesn't support a color temperature, I get the following response:

Hurr1c4n3, 2022-09-28:
`Error 2022-09-28 17:39:20Publish 'set' 'read' to 'Lamp achterdeur' failed: 'Error: Read 0x0017880108c1cfcd/11 lightingColorCtrl(["colorTempPhysicalMin","colorTempPhysicalMax"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

`

danieledwardgeorgehitchcock, 2022-09-28:
Sorry, my bad! Shall I create a PR on your behalf to get this merged?

Hurr1c4n3, 2022-09-28:
If you could do that, it would be great. Can I change anything to support it right away?

danieledwardgeorgehitchcock, 2022-09-28:
Done! Once it is merged, you could switch to the Dev branch of Z2M.

In the meantime, your external converter should work

Hurr1c4n3, 2022-09-28:
Thnx for the help. But with the converter I posted in my firtst post, I can't do anything. It keeps telling me, No converter available for ON/OFF/brightness

danieledwardgeorgehitchcock, 2022-09-28:
Try this:

```javascript
const extendDontUse = require('zigbee-herdsman-converters/lib/extend');

const hueExtend = {
    light_onoff_brightness: (options={}) => ({
        ...extendDontUse.light_onoff_brightness(options),
        ota: ota.zigbeeOTA,
        meta: {turnsOffAtBrightness1: true},
        toZigbee: extendDontUse.light_onoff_brightness(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
};

const definition = {
    zigbeeModel: ['1740447P0'],
    model: '8718696166079',
    vendor: 'Philips',
    description: 'Hue Tuar Outdoor Wall Light',
    extend: hueExtend.light_onoff_brightness(),
};

module.exports = definition;
```

Hurr1c4n3, 2022-09-28:
z2m can't boot up:
```
[18:19:29] INFO: Handing over control to Zigbee2MQTT Core ...
/app/dist/util/externally-loaded.js:6
        ota: ota.zigbeeOTA,
             ^
ReferenceError: ota is not defined
    at Object.light_onoff_brightness (/app/dist/util/externally-loaded.js:6:14)
    at /app/dist/util/externally-loaded.js:17:23
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
```

danieledwardgeorgehitchcock, 2022-09-28:
sorry, I updated my comment above once I had a scan through what I posted up originally. Can you check you are using my latest update?

Hurr1c4n3, 2022-09-28:
Glad you helped. Everything seems to work now. Thank you for your time.

closed 2022-09-28 by Hurr1c4n3