Koenkk/zigbee2mqtt #14203: TS0601 (_TZE200_zah67ekd) does not report closed

mdallaire, 2022-09-26:
### What happened?

When closing the cover the state never reaches 0, it always stop at 1 so it never show as closed in zigbee2mqtt or in Home-Assistant.

Also:
https://github.com/Koenkk/zigbee2mqtt/issues/7961
https://github.com/Koenkk/zigbee2mqtt/issues/7969

### What did you expect to happen?

I expect that when the cover is closed the status shows as closed correctly.

A remapping of 1 as the value for closed would probably fix this.

### How to reproduce it (minimal and precise)

- Purchase a [Moes House Chain Motorized Roller shutter](https://www.aliexpress.com/item/1005001698400295.html)
- Set it up and calibrate it.
- Add it to zigbee2mqtt
- Send a close command or close it using the physical button
- Check the position in zigbee2mqtt: It shows 1 with an "Open" state

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

 20211217

### Adapter

Electrolama

### Debug log

Zigbee Model: TS0601
Zigbee Manufacturer: _TZE200_zah67ekd

There is probably a change required here : https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js#L1848 for this specific Model/Manufacturer but I am unsure of how to define such value.

leenemo, 2022-09-29:
I have the same issue.Did you fix it? Can u tell the method? Thanks.

mdallaire, 2022-09-29:
Unfortunately I don't have it fixed and am unsure how to do it.

It looks like several cover share the same code. I am no developer but I spent a good amount of time trying to figure out where things are and to me it looks like some kind of override to this [default for covers](https://github.com/Koenkk/zigbee-herdsman-converters/blob/96bd6aaddd32c8d84cd6bdcd47df9720f0e05919/lib/exposes.js#L335):

```
class Cover extends Base {
(...)
    withPosition() {
        assert(!this.endpoint, 'Cannot add feature after adding endpoint');
        this.features.push(new Numeric('position', access.ALL).withValueMin(0).withValueMax(100).withDescription('Position of this cover'));
        return this;
    }
(...)
    }
}
```

need to be done in [this tuya cover section](https://github.com/Koenkk/zigbee-herdsman-converters/blob/96bd6aaddd32c8d84cd6bdcd47df9720f0e05919/devices/tuya.js#L1865):

```
        exposes: [
            e.cover_position().setAccess('position', ea.STATE_SET),
            exposes.composite('options', 'options')
                .withFeature(exposes.numeric('motor_speed', ea.STATE_SET)
                    .withValueMin(0)
                    .withValueMax(255)
                    .withDescription('Motor speed'))],
    },
```

 A new section of the second file will probably need to be created to address this specific device without breaking the values for the others.

mdallaire, 2022-10-31:
This is still an issue.

closed 2022-12-09 by github-actions[bot]