Koenkk/zigbee2mqtt #14204: Implementing repeaters/Aqara battery device issues

keness521, 2022-09-27:
### What happened?

I have several Aqara battery devices at the far end of my network which "give up" and go offline. I've read quite a number of posts about this issue and gather it is a well known characteristic of Aqara battery devices if signal is weak. (My other devices in the same area are always responsive, so I don't think signal is TOO weak, but it must be for what the battery device wants.)

By and large, the solution given is always to have more routers. But that's where I start to become confused... I have a 47 device network, and over 30 of those devices ARE routers. At least 6 of them in the same room as the problematic Aqara devices.

Most of my routers are eWeLink SA-003-Zigbee wall switches (or outlets, same identification) and they always perform flawlessly, but maybe they are poor routers? They seem to have very low lqi compared to my Aqara powered devices, but always work fine. I've also forced the battery devices onto a specific Aqara wall switch router with "Permit Join (Specific Router)", but still they drop off, even though the wall switch itself never does, and is Aqara as well.

Is there something I'm missing? Almost all of the routers seem to connect (solid line) with the coordinator (CC2652 - Sonoff Dongle Plus) directly, with really low lqi for the more distant ones (in the 1 to 10 range, often) but have tons of dashed lines amongst themselves as well.

Thank you for any advice you may have!

### What did you expect to happen?

My space is long and skinny, about 80 feet long. But there are wall switch routers about every 10 feet all the way down the hallway. I figured that would easily create a robust mesh.

Most of my routers are the eWeLink SA-003-Zigbee, so perhaps it isn't a good router, but I do also have a number of Aqara wall switches which are routers as well.

Would creating a daisy chain of a couple known good Aqara-friendly routers/repeaters down the length of the hallway, and then using "Permit Join (Specific Router)" to force the problematic Aqara battery devices to join the last (nearest) repeater be a worthwhile attempt?

### How to reproduce it (minimal and precise)

At the farthest room, just wait. An Aqara battery device will drop off sooner or later, usually sooner. (A few times a week.) Despite there being tons of routers on the network and nearby.

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20220219

### Adapter

Sonoff USB 3.0 Dongle Plus, CC2652 version, on an extension

Shred99, 2022-09-27:
The Aqara devices are "fussy" about the routers that they are happy with.  At the start of this massive thread (which you have probably seen) there is a list of "known to work" and "known to not work" routers:

[https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623](url)

I've found any router with the zigbee2mqtt or the PTVO firmware (based on the Zigbee2mqtt firmware I believe) works reasonably well with them.  Recent coordinator firmware releases seem to have improved my Aqara device stability significantly.  The addition of a Sonoff USB Dongle (the CC2652P based one - beware that there is another with a bleeding edge chipset that doesn't yet play well) has helped significantly.  The Sonoff dongles are fairly cheap, but seem to be good quality.

keness521, 2022-09-27:
Thank you for the information!

I have seen it mentioned that Aqara/Xiaomi end devices don't (ever? rarely?) switch routers once they attach to a router.

Is it known if this is definitively true, or just anecdotal? If it is true, I would think that using the "Permit Join (Specific Router)" would be a good way to force it onto a known-good router. The eWeLink SA-003-Zigbee that I have so many of isn't on either list, good or bad, so I'm unsure if they are reliable.

I'm thinking of getting 1 or 2 more CC2562-based Sonoff dongles and flashing with router firmware as it seems they should be good.

Even if I force the first CC2562 router to join to the coordinator, and the second to join to the first, though, I am assuming the CC2562 devices might switch their route, like a normal Zigbee device, if they decide they should?

Thanks again.

Shred99, 2022-09-28:
Yes, the CC2562P devices are able to dynamically find the best path through the network, since (at least when running Zigbee2MQTT) they comply with the Zigbee protocol standards.

Take a look at the steps to crank the transmit power up on the coordinator and the routers too.  It's not that straightforward on the CC2562P routers.  I wanted to experiment with different Tx power levels, so I just bought licensed copies of the PTVO firmware for my routers.  The PTVO firmware lets you change output power "on the fly" via MQTT commands.  If all your CC2562P devices are running +5 or +10 dBm (can go up to 20dBm or 100x the Tx power of a standard device), they will form a really solid backbone to your network.  As I said earlier though - higher output power from your router won't help the router hear a feeble signal from a distant battery powered device.

I haven't seen anyone research the how and why of Aqara devices deciding which router to communicate through.  The devices do occasionally change router, but there doesn't seem to be any rhyme or reason to it.  I've seen Aqara devices refuse to change routers, even when there are two that are much closer and have much stronger signals.  I've seen the devices leave a good, solid nearby router and join one that is 40m away, for no apparent reason (and then they've refused to change back).  It's weird and random and I'm not convinced that the newer Aqara E1 series sensors that are supposedly Zigbee 3.0 compliant are much better.

keness521, 2022-09-29:
I think the Sonoff CC2562P dongles flashed with router software may be the way to go. It's a shame they've jumped up in price, they were so cheap before! hah

I'll have to look into boosting the Tx power... I know there are some plusses and minuses. Especially if I do several of the routers. That's a lot of shouting going on...!

I've read some of the older threads about boosting the Tx power, but that was in the early days, and I'm not quite sure where it stands now as to what the actual procedure is to achieve it. I'll have to see if I can't find a more recent thread with current instructions.

ashokaiyar, 2022-10-01:
@keness521 

I have an 85 node zigbee2mqtt mesh with ~20 routers and ~65 end-devices. 55 of the end-devices are Aqara. This mesh has been up and running very stably for 2.5 years. 

[image]

I have Sonoff CC2652P dongles flashed as routers. But the routers that work best remain the [TuYa USB zigbee routers](https://www.aliexpress.us/item/3256804277232493.html). They are tiny, inexpensive, and easily support 6+ end-devices per router. Here's an example:

[image]

keness521, 2022-10-01:
> I have an 85 node zigbee2mqtt mesh with ~20 routers and ~65 end-devices. 55 of the end-devices are Aqara. This mesh has been up and running very stably for 2.5 years.
>
> I have Sonoff CC2652P dongles [...]

Ok, that's actually great to hear, because I bought 4 of the Sonoff Dongle-P adapters to flash as routers, but I also bought 4 of those exact TuYa repeaters.

Did you force your Aqara devices to join the TuYa repeater using "Permit Join (the TuYa repeater)" or did you just let them do whatever they wanted, and that's where they went? (In your second picture, for example.) I gather Aqara devices are non-standard in that they don't (ever? often?) switch routers once they've joined one.

ashokaiyar, 2022-10-01:
> Did you force your Aqara devices to join the TuYa repeater using "Permit Join (the TuYa repeater)" or did you just let them do whatever they wanted, and that's where they went? (In your second picture, for example.)

I put the mesh into panic mode for 20-30 minutes. You can do this by shutting down z2m and unpowering the coordinator. When you bring it back up, your Xiaomi/Aqara devices will have globbed onto repeaters that they like. And will stay there.

> Do you have any recommendation on how to "disperse" the new routers (Sonoff and TuYa) around a long, skinny space, so that devices which most need range extension actually take advantage of it?

I use the Sonoff routers in outdoor space. And I've the TuYa all over my house. Not sure how to place them in a long skinny space.