Koenkk/zigbee2mqtt #14205: [New device support]: Skydance S1-B (WZ)

jmbinette, 2022-09-27:
### Link

http://www.iskydance.com/index.php?c=product_show&a=index&id=1616

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0xa4c138dbd140c3f6","nwkAddr":64416,"manufId":4417,"manufName":"_TZB210_rkgngb5o","powerSource":"Mains (single phase)","modelId":"TS0502B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0502B","manufacturerName":"_TZB210_rkgngb5o","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":49}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1664243346258,"defaultSendRequestWhen":"immediate"}

### Comments

This is a Tuya device similar to other Skydance variations

For some reason, if I use : 
extend: extend.light_onoff_brightness({disableEffect: true}),
Brightness setting will now work.

I have to use : 
extend: extend.light_onoff_brightness_color({disableEffect: true}),
And everything works except for sure there is no color control

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  fingerprint: [{modelID: 'TS0502B', manufacturerName: '_TZB210_rkgngb5o'}],
  model: 'TS0502B',
  vendor: 'TuYa',
  description: 'Zigbee Dim light',
  extend: extend.light_onoff_brightness_color({disableEffect: true}),
};

module.exports = definition;
```

### Supported color modes

None, dim only

Koenkk, 2022-09-28:
Added!

closed 2022-09-28 by Koenkk

jmbinette, 2022-09-28:
Thanks !
Any idea why I have to add _color for the brightness to work correctly ?

Koenkk, 2022-09-28:
Can you provide me the herdsman debug log when setting the brightness for both cases? (with and without `_color`)

jmbinette, 2022-09-28:
For some reason, I removed _color from the external converter again and seems to work fine now !

Maybe I forgot to restart after the change.

Thanks again !