Koenkk/zigbee2mqtt #14222: ZlinkTIC Support firmware

bilbon0bud, 2022-09-28:
### What happened?

The ZlinkTIC has two firmware, one with child routing and another without.
The power of zlinktic is not much powered, so routing child item make it to stop working correctly.

### What did you expect to happen?

Could it be possible to offer the choice of the firmware we want to use (with routing or not) or just push the safest firmware with no routing function ?

here is the firmware

https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v7.0/ZLinky_router_v7.0_no_route.ota

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20210708

### Adapter

sonoff

starox, 2022-10-05:
I encounter the same issue, the router firmware is not advised.
Since v7 IIRC I can't have the ZLinky work reliably more than one day

vingerha, 2022-10-07:
Same interest here, it would be nice if the OTA could select the file from the source (if there is something to select)

vingerha, 2022-11-07:
@Koenkk 
The only option to load this to Zlinky is via cable which is not very practical as multiple updates occur during the year.
If you however believe that this non-route option has little value to add into OTA, then please reject the case