Koenkk/zigbee2mqtt #14225: [New device support]: Dresden Elektronik FLS-A2

l-mb, 2022-09-28:
### Link

https://www.dresden-elektronik.com/product/fls-a-lp.html

### Database entry

{"id":48,"type":"Router","ieeeAddr":"0x00212effff00f2bb","nwkAddr":30330,"manufId":4405,"manufName":"dresden elektronik\u0000","powerSource":"Mains (single phase)","modelId":"FLS-A2 (1-10V)\u0000","epList":[11,12,13,14],"endpoints":{"11":{"profId":49246,"epId":11,"devId":256,"inClusterList":[0,4096,4,3,5,6,8,64512],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"FLS-A2 (1-10V)\u0000","manufacturerName":"dresden elektronik\u0000","powerSource":1,"zclVersion":1,"appVersion":2,"stackVersion":3,"hwVersion":1,"dateCode":"20161013\u0000","swBuildId":"020E.201000CE\u0000"}}},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":256,"inClusterList":[0,4096,4,3,5,6,8,64512],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":256,"inClusterList":[0,4096,4,3,5,6,8,64512],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"14":{"profId":49246,"epId":14,"devId":256,"inClusterList":[0,4096,4,3,5,6,8,64512],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":3,"hwVersion":1,"dateCode":"20161013\u0000","swBuildId":"020E.201000CE\u0000","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1664358858454,"defaultSendRequestWhen":"immediate"}

### Comments

This device exposes four different channels in this configuration that should be possible to individually control. There was a stale request that I could not reopen, and I got much further here.

I looked at a pre-existing device by Dresden Elektronik and Osram which expose more than one endpoint.

I can, in fact, control the four channels, and the commands send do take effect, but only after a delay and there's still an error/timeout message in the logs:

```
Sep 28 13:14:04 spof zigbee2mqtt[20047]: Zigbee2MQTT:error 2022-09-28 13:14:04: Publish 'set' 'state' to '0x00212effff00f2bb' failed: 'Error: Command 0x00212effff00f2bb/13 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 30330 - 13 - 151 - 6 - 11 after 10000ms)'
Sep 28 13:14:27 spof zigbee2mqtt[20047]: Zigbee2MQTT:error 2022-09-28 13:14:27: Failed to read state of '0x00212effff00f2bb' after reconnect (Read 0x00212effff00f2bb/11 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 30330 - 11 - 152 - 8 - 1 after 10000ms))
Sep 28 13:14:50 spof zigbee2mqtt[20047]: Zigbee2MQTT:error 2022-09-28 13:14:50: Publish 'set' 'brightness' to '0x00212effff00f2bb' failed: 'Error: Command 0x00212effff00f2bb/12 genLevelCtrl.moveToLevelWithOnOff({"level":133,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 30330 - 12 - 159 - 8 - 11 after 10000ms)'
Sep 28 13:15:13 spof zigbee2mqtt[20047]: Zigbee2MQTT:error 2022-09-28 13:15:13: Failed to read state of '0x00212effff00f2bb' after reconnect (Read 0x00212effff00f2bb/11 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 30330 - 11 - 169 - 8 - 1 after 10000ms))
```

I'm not sure how to proceed here.

I tried removing the `configure` section and corresponding `noConfigure: true` setting as well, but this did not help.

Reporting of the device state also seems to not be working which is probably related?

Any hint appreciated!

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['FLS-A2 (1-10V)'],
    model: 'BN-600078',
    vendor: 'Dresden Elektronik',
    description: 'Zigbee controller for 1-10V/PWM',
    extend: extend.light_onoff_brightness({noConfigure:true}),
    exposes: [ e.light_brightness().withEndpoint('l1'),
               e.light_brightness().withEndpoint('l2'),
               e.light_brightness().withEndpoint('l3'),
               e.light_brightness().withEndpoint('l4') ],
    endpoint: (device) => {
       return {'l1': 11, 'l2': 12, 'l3': 13, 'l4': 14};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
       await reporting.bind(device.getEndpoint(11), coordinatorEndpoint, ['genLevelCtrl', 'genOnOff']);
       await reporting.bind(device.getEndpoint(12), coordinatorEndpoint, ['genLevelCtrl', 'genOnOff']);
       await reporting.bind(device.getEndpoint(13), coordinatorEndpoint, ['genLevelCtrl', 'genOnOff']);
       await reporting.bind(device.getEndpoint(14), coordinatorEndpoint, ['genLevelCtrl', 'genOnOff']);
       await reporting.onOff(device.getEndpoint(11));
       await reporting.brightness(device.getEndpoint(11));
       await reporting.onOff(device.getEndpoint(12));
       await reporting.brightness(device.getEndpoint(12));
       await reporting.onOff(device.getEndpoint(13));
       await reporting.brightness(device.getEndpoint(13));
       await reporting.onOff(device.getEndpoint(14));
       await reporting.brightness(device.getEndpoint(14));
    },
};

module.exports = definition;
```
```

### Supported color modes

Brightness

l-mb, 2022-09-28:
Yes, the dip switch is set to 4-channel operation. I'm migrating the devices away from the deconz raspberry setup, so I know they work right - the other one is still paired there and operational via that Home Assistant integration.

I'm getting the impression that something more fundamental is going wrong - even a simple read of the genBasic manufacturerName, a near instantaneous request on the other devices, times out on me. Yet the interrogation at pairing time obviously worked.

I've also tried unpairing. When I repair them with the deconz gateway, they work there. When I _then_ un- and repair them with the coordinator, they work very briefly for a few seconds/commands, until they hit the timeouts again.

Koenkk, 2022-09-28:
Could you sniff the traffic of this? Starting from the moment you pair until controlling it fails.

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

l-mb, 2022-09-28:
The docs are for the CC2531, and while I see a firmware for the SONOFF3 with packet-sniffer-2 (`sniffer_fw/bin/cc2652rblp/sniffer_fw.hex`), the doc referenced explicitly warn against that version. Would this be a problem?

If not, this is going to be a dumb question, but will the device still work as a coordinator with no harm to the rest of the setup? I'd like to avoid repairing 48 devices :) If so, I'll give this a try. Otherwise, I might try to order a CC2531 for sniffing purposes.

Koenkk, 2022-09-29:
Lets not try that. Can you change the `meta` in the converter to `meta: {disableDefaultResponse: true, multiEndpoint: true},`

l-mb, 2022-10-06:
> Lets not try that. Can you change the `meta` in the converter to `meta: {disableDefaultResponse: true, multiEndpoint: true},`

Indeed, that seems to have done the trick! I've since managed to create a time-out once, but I can't reproduce it. Not sure I see any ill effects of this flag so far.

Thanks a lot, I'll hopefully be able to assemble the rest of the device support PR soon.

l-mb, 2022-10-06:
I've got to take that back. :-/ They've now decided to drop of the network with "No network route (205)". Looks like I need to debug a bit more.

l-mb, 2022-11-04:
With the various recent updates to z2m, the coordinator, possibly the device's own firmware, it now appears to work mostly stable (and is definitely much better than not supporting the device at all). I've had the device drop once in a couple of days.

The manufacturer pointed me at http://deconz.dresden-elektronik.de/otau/1135-0004-201000E9-FLS-A2_MD.zigbee for the OTA update - given the problems I had with the z2m integration up to that point, I ran that via the vendor's bridge, so I have no way to add/test support for this aspect.

l-mb, 2022-12-05:
Let's close. It seems to be working well enough, even if apparently every 2-3 weeks I have to physically power cycle the device to make it reconnect, but that could also just be a bad signal strength ... Thanks all.

closed 2022-12-05 by l-mb