Koenkk/zigbee2mqtt #14226: Xiaomi SJCGQ11LM water_leak state is not decoded from periodic report

dgwynne, 2022-09-28:
### What happened?

I'm trying z2m out as an alternative to a Tasmota zbbridge and hit an unexpected regression. I have a pair of Xiaomi SJCGQ11LM water leak sensors that periodically report their state, but those periodic reports do not include the `water_leak` state being output to MQTT.

I have a manually configured MQTT sensor in HA like this:

```yaml
binary_sensor kitchen_leak:
  - platform: mqtt
    name: "Kitchen Leak"
    device_class: moisture
    state_topic: "zigbee2mqtt/kitchen_leak/water_leak"
    payload_on: "true"
    payload_off: "false"
    force_update: true
    expire_after: 7200
```

I rely on the periodic reports to keep sensor "alive".

### What did you expect to happen?

I expect the `water_leak` attribute to be part of the MQTT messages generated from the periodic report thing from the device.

It should be decoded like tasmota does at https://github.com/arendst/Tasmota/blob/91f5b302121a4a814b6ca87d6f6a67824dee4cb5/tasmota/tasmota_xdrv_driver/xdrv_23_zigbee_5_2_converters.ino#L1236-L1239

An equivalent fz handler could look something like this for zigbee-herdsman-converters/converters/fromZigbee.js wired into the lumi.sensor_wleak.aq1 device in xiaomi.js:

```js
    xiaomi_sensor_wleak_aq1: {
        cluster: 'genBasic',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            if (msg.data['65281']) {
                const data = msg.data['65281'];
                return {
                    water_leak: data['100'] === 1,
                };
            }
        },
    },
```

### How to reproduce it (minimal and precise)

Use config like this:

```yaml
  output: attribute_and_json
  cache_state: false
  cache_state_persistent: false
  cache_state_send_on_startup: false
```

Pair a water leak sensor, and press the button on it to force a report. Note the lack of "water_leak" state in the MQTT messages.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20210708

### Adapter

Zigbee 3.0 USB Dongle Plus-P

### Debug log

```
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,87,68,129,0,0,0,0,51,89,1,1,0,87,0,167,132,55,0,0,67,28,95,17,19,10,5,0,66,21,108,117,109,105,46,115,101,110,115,111,114,95,119,108,101,97,107,46,97,113,49,1,255,66,34,1,33,209,11,3,40,26,4,33,168,19,5,33,27,0,6,36,0,0,0,0,0,8,33,6,2,10,33,0,0,100,16,0,51,89,29,51] +15s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,87,68,129,0,0,0,0,51,89,1,1,0,87,0,167,132,55,0,0,67,28,95,17,19,10,5,0,66,21,108,117,109,105,46,115,101,110,115,111,114,95,119,108,101,97,107,46,97,113,49,1,255,66,34,1,33,209,11,3,40,26,4,33,168,19,5,33,27,0,6,36,0,0,0,0,0,8,33,6,2,10,33,0,0,100,16,0,51,89,29,51] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 87 - 2 - 4 - 129 - [0,0,0,0,51,89,1,1,0,87,0,167,132,55,0,0,67,28,95,17,19,10,5,0,66,21,108,117,109,105,46,115,101,110,115,111,114,95,119,108,101,97,107,46,97,113,49,1,255,66,34,1,33,209,11,3,40,26,4,33,168,19,5,33,27,0,6,36,0,0,0,0,0,8,33,6,2,10,33,0,0,100,16,0,51,89,29] - 51 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":22835,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":87,"securityuse":0,"timestamp":3638439,"transseqnumber":0,"len":67,"data":{"type":"Buffer","data":[28,95,17,19,10,5,0,66,21,108,117,109,105,46,115,101,110,115,111,114,95,119,108,101,97,107,46,97,113,49,1,255,66,34,1,33,209,11,3,40,26,4,33,168,19,5,33,27,0,6,36,0,0,0,0,0,8,33,6,2,10,33,0,0,100,16,0]}} +15s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":19,"manufacturerCode":4447,"commandIdentifier":10},"Payload":[{"attrId":5,"dataType":66,"attrData":"lumi.sensor_wleak.aq1"},{"attrId":65281,"dataType":66,"attrData":{"1":3025,"3":26,"4":5032,"5":27,"6":[0,0],"8":518,"10":0,"100":0}}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":22835,"endpoint":1,"linkquality":87,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}' +33s
Zigbee2MQTT:debug 2022-09-28 21:13:58: Received Zigbee message from 'kitchen_leak', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"10":0,"100":0,"3":26,"4":5032,"5":27,"6":[0,0],"8":518},"modelId":"lumi.sensor_wleak.aq1"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-09-28 21:13:58: lumi.sensor_wleak.aq1: unknown key 6 with value 0,0
Zigbee2MQTT:debug 2022-09-28 21:13:58: lumi.sensor_wleak.aq1: Processed data into payload {"voltage":3025,"battery":100,"device_temperature":26,"power_outage_count":26}
Zigbee2MQTT:debug 2022-09-28 21:13:58: lumi.sensor_wleak.aq1: Processed data into payload {"voltage":3025,"battery":100,"device_temperature":26,"power_outage_count":26}
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +68ms
Zigbee2MQTT:info  2022-09-28 21:13:58: MQTT publish: topic 'zigbee2mqtt/kitchen_leak', payload '{"battery":100,"device_temperature":26,"linkquality":87,"power_outage_count":26,"voltage":3025}'
Zigbee2MQTT:info  2022-09-28 21:13:58: MQTT publish: topic 'zigbee2mqtt/kitchen_leak/voltage', payload '3025'
Zigbee2MQTT:info  2022-09-28 21:13:58: MQTT publish: topic 'zigbee2mqtt/kitchen_leak/battery', payload '100'
Zigbee2MQTT:info  2022-09-28 21:13:58: MQTT publish: topic 'zigbee2mqtt/kitchen_leak/device_temperature', payload '26'
Zigbee2MQTT:info  2022-09-28 21:13:58: MQTT publish: topic 'zigbee2mqtt/kitchen_leak/mode_switch', payload ''
Zigbee2MQTT:info  2022-09-28 21:13:58: MQTT publish: topic 'zigbee2mqtt/kitchen_leak/power_outage_count', payload '26'
Zigbee2MQTT:info  2022-09-28 21:13:58: MQTT publish: topic 'zigbee2mqtt/kitchen_leak/switch_type', payload ''
Zigbee2MQTT:info  2022-09-28 21:13:58: MQTT publish: topic 'zigbee2mqtt/kitchen_leak/linkquality', payload '87'
```

Koenkk, 2022-09-28:
This is intentionally ignored since it is not reliable
- https://github.com/Koenkk/zigbee-herdsman-converters/blob/771e1fff62645ff17eda9361eddcb9cfa9abb264/lib/xiaomi.js#L241
- https://github.com/Koenkk/zigbee2mqtt/issues/12018

dgwynne, 2022-09-28:
Except it has been reliable and useful in my experience. Could there be a device specific option provided to enable use of it, or perhaps expose the value from the xiaomi cluster as a separate state?

Koenkk, 2022-09-29:
If more people are interested in this we can add an option for this.

closed 2022-11-06 by github-actions[bot]

Paul-C-S, 2023-01-08:
Hi, I would like this option made available if possible please.