Koenkk/zigbee2mqtt #14233: Relay TS011F _TZ3000_7dndcnnb

amgstone, 2022-09-28:
### What happened?

after each update for a minute, current and power are reset to 0 and then immediately restore the value. i have 2 relay, one of them works like this.
[image]

### Zigbee2MQTT version

edge

### Adapter firmware version

edge

### Adapter

CC2538

Malaysk, 2022-09-29:
_TZ3000_b28wrpvx
also this behavior. Sets to 0 every 2-3 minutes
[image]

amgstone, 2022-09-29:
energy count, but curwnt and powers are not updated

closed 2022-11-06 by github-actions[bot]

zhvial, 2022-11-29:
I have the same problem.
Updated to the latest firmware and that didn't solve the problem.
[image]
Voltage drops occure the same time power drops to 0.