Koenkk/zigbee2mqtt #14239: [New device support]:

konvex1337, 2022-09-28:
### Link

https://www.globus-baumarkt.de/essentials-smart-home-heizkoerperthermostat-zigbee?utm_source=awin&utm_medium=176013_RedBrain&utm_campaign=awo&utm_content=0&utm_term=26068275667&awc=11830_1664396331_181d5f856dc4fb5bb39cc41791b288c7

### Database entry

debug 2022-09-28 22:20:33Received Zigbee message from '0x84ba20fffe3246a1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,153],"type":"Buffer"},"datatype":2,"dp":34}],"seq":8704}' from endpoint 1 with groupID 0 info 2022-09-28 22:20:33zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint #34 from 0x84ba20fffe3246a1 with raw data '{"dp":34,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,153]}}': type='commandDataReport', datatype='value', value='153', known DP# usage: [] debug 2022-09-28 22:21:03Received Zigbee message from '0x84ba20fffe3246a1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,45],"type":"Buffer"},"datatype":2,"dp":16}],"seq":4096}' from endpoint 1 with groupID 0 info 2022-09-28 22:21:03zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint #16 from 0x84ba20fffe3246a1 with raw data '{"dp":16,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,45]}}': type='commandDataReport', datatype='value', value='45', known DP# usage: ["moesHeatingSetpoint","tvHeatingSetpoint","haozeeMaxTemp","AM02Border","x5hSetTemp","connecteHeatingSetpoint","alectoSilence"] debug 2022-09-28 22:22:12Received Zigbee message from '0x84ba20fffe3246a1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,34],"type":"Buffer"},"datatype":2,"dp":16}],"seq":4096}' from endpoint 1 with groupID 0 info 2022-09-28 22:22:12zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint #16 from 0x84ba20fffe3246a1 with raw data '{"dp":16,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,34]}}': type='commandDataReport', datatype='value', value='34', known DP# usage: ["moesHeatingSetpoint","tvHeatingSetpoint","haozeeMaxTemp","AM02Border","x5hSetTemp","connecteHeatingSetpoint","alectoSilence"] debug 2022-09-28 22:22:59Received Zigbee message from '0x84ba20fffe3246a1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":2}],"seq":512}' from endpoint 1 with groupID 0 info 2022-09-28 22:22:59zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint #2 from 0x84ba20fffe3246a1 with raw data '{"dp":2,"datatype":4,"data":{"type":"Buffer","data":[2]}}': type='commandDataReport', datatype='enum', value='2', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tthHumidity","tIlluminanceLux","evanellMode","AM02PercentControl","x5hMode","connecteMode","tshpscSensitivity","alectoSmokeValue"] debug 2022-09-28 22:23:48Received Zigbee message from '0x84ba20fffe3246a1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":2}],"seq":512}' from endpoint 1 with groupID 0 info 2022-09-28 22:23:48zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint #2 from 0x84ba20fffe3246a1 with raw data '{"dp":2,"datatype":4,"data":{"type":"Buffer","data":[1]}}': type='commandDataReport', datatype='enum', value='1', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tthHumidity","tIlluminanceLux","evanellMode","AM02PercentControl","x5hMode","connecteMode","tshpscSensitivity","alectoSmokeValue"] debug 2022-09-28 22:24:50Received Zigbee message from '0x84ba20fffe3246a1', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,43],"type":"Buffer"},"datatype":2,"dp":16}],"seq":4096}' from endpoint 1 with groupID 0 info 2022-09-28 22:24:50zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint #16 from 0x84ba20fffe3246a1 with raw data '{"dp":16,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,43]}}': type='commandDataReport', datatype='value', value='43', known DP# usage: ["moesHeatingSetpoint","tvHeatingSetpoint","haozeeMaxTemp","AM02Border","x5hSetTemp","connecteHeatingSetpoint","alectoSilence"]

### Comments

Need help to setup the external converter.
The device looks like the silvercrest thermostat from the vendor lidl.
I tested some changes in the line "fromzigbee" and "tozigbee" but nothing worked.
I copyed the lines of the lidl device to my external converter file but zigbee did not restart "Error const" not find
Changed the const lines to the lidl once but the same happens.

Did i have to creat a fromzigbee.js and tozigbee.js file in any folder ?

Thank you

### Supported color modes

no

### Color temperature range

no

konvex1337, 2022-09-29:
i think it is the same issue as #13962

konvex1337, 2022-09-29:
get suck with the payload test

topic :
zigbee2mqtt/0x84ba20fffe3246a1/set/tuya_data_point_test
payload :
16,2,[0,0,0,45]
is it the correct payload ?

Longlife4ever, 2022-10-01:
I have the same Valve, need help, too.

What has to be done to get all the information for a full support of the device?

TAKeanice, 2022-10-25:
Someone was successful (at least partially) by adding the device identifier _TZE200_i48qyn9s to this line https://github.com/Koenkk/zigbee-herdsman-converters/blob/b776f639bc37339b328abd612a6fb367a5cba76b/devices/lidl.js#L842 in the lidl.js device adapter, in this issue: https://github.com/ioBroker/ioBroker.zigbee/issues/1400. Of course, this is kind of hacky.

Allegedly this converter https://gist.github.com/serrj-sv/af142b25de2d7ac54c3a2eb2623d9a6d is what solved the issue in the forum entry https://forum.iobroker.net/topic/58272/essentials-zigbee-thermostat-external-converter-einbinden (at least on the Zigbee2Mqtt side) and led to the already mentioned issue https://github.com/ioBroker/ioBroker.zigbee/issues/1527

TAKeanice, 2022-10-29:
I can report that https://gist.github.com/serrj-sv/af142b25de2d7ac54c3a2eb2623d9a6d really seems to cover and properly parse all the data points from the essentials thermostat. I will work a little on the exposes part of the definition and then publish my solution as a basis for integrating it into the main converter repo.

TAKeanice, 2022-11-03:
So far, I have this:
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const tuyaLocal = {
  dataPoints: {
  sh4Mode: 2,
  sh4HeatingSetpoint: 16,
  sh4LocalTemp: 24,
  sh4ChildLock: 30,
  sh4Battery: 34,
  sh4FaultCode: 45,
  sh4ComfortTemp: 101,
  sh4EcoTemp: 102,
  sh4VacationPeriod: 103,
  sh4TempCalibration: 104,
  sh4ScheduleTempOverride: 105,
  sh4RapidHeating: 106,
  sh4WindowStatus: 107,
  sh4Hibernate: 108,
  sh4ScheduleMon: 109,
  sh4ScheduleTue: 110,
  sh4ScheduleWed: 111,
  sh4ScheduleThu: 112,
  sh4ScheduleFri: 113,
  sh4ScheduleSat: 114,
  sh4ScheduleSun: 115,
  sh4OpenWindowTemp: 116,
  sh4OpenWindowTime: 117,
  sh4RapidHeatCntdownTimer: 118,
  sh4TempControl: 119,
  sh4RequestUpdate: 120,
  },
};
const fzLocal = {
  sh4_thermostat: {
    cluster: 'manuSpecificTuya',
    type: ['commandDataResponse', 'commandDataReport'],
    convert: (model, msg, publish, options, meta) => {
      //const dp = msg.data.dp;
      //const value = tuya.getDataValue(msg.data.datatype, msg.data.data);
      const dpValue = tuya.firstDpValue(msg, meta, 'sh4_thermostat');
      const dp = dpValue.dp;
      const value = tuya.getDataValue(dpValue);
      function weeklySchedule(day, value) {
        // byte 0 - Day of Week (0~7 = Mon ~ Sun) <- redundant?
        // byte 1 - 1st period Temperature (1~59 = 0.5~29.5°C (0.5 step))
        // byte 2 - 1st period end time (1~96 = 0:15 ~ 24:00 (15 min increment, i.e. 2 = 0:30, 3 = 0:45, ...))
        // byte 3 - 2nd period Temperature
        // byte 4 - 2nd period end time
        // ...
        // byte 16 - 8th period end time
        // byte 17 - 9th period Temperature
        let daySchedule = {};
        for (let i = 1; i <= 9; i++) {
          const tempId = ((i-1) * 2) +1;
          const timeId = ((i-1) * 2) +2;
          setpoint = {};
          setpoint['temperature'] = (value[tempId] / 2).toFixed(1);
          if (i!=9) {
              setpoint['hour'] = Math.floor(value[timeId] / 4).toString().padStart(2, '0');
              setpoint['minute'] = ((value[timeId] % 4) *15).toString().padStart(2, '0');
          }
[... 301 lines cut ...]
      const expose = exposes.composite(`${day}_schedule`, day);
      [1, 2, 3, 4, 5, 6, 7, 8, 9].forEach((i) => {
        const ex = exposes.composite(`setpoint_${i}`, `${i}`)
          ex.withFeature(exposes.numeric('temperature', ea.STATE_SET).withValueMin(0.5)
              .withValueMax(29.5).withValueStep(0.5).withUnit('°C').withDescription(`Temperature ${i}`));
          if (i < 9) {
            ex.withFeature(exposes.enum('hour', ea.STATE_SET,
                ['00', '01', '02', '03', '04', '05', '06', '07', '08', '09',
                    '10', '11', '12', '13', '14', '15', '16', '17', '18', '19',
                    '20', '21', '22', '23', '24']).withDescription(`Hour TO for temp ${i}`));
            ex.withFeature(exposes.enum('minute', ea.STATE_SET, ['00', '15', '30', '45'])
                .withDescription(`Minute TO for temp ${i}`));
          }
          expose.withFeature(ex);
      });
      return expose;
    }),
  ],
};

module.exports = device;

```

However, as you can see, the "to-zigbee" part of the week schedule is not there. This is because the handling of composites seems broken or I am doing something very wrong, but I cannot get any value to be sent to this endpoint. When I call it from the UI shipping with Zigbee2mqtt, the sliders, dropdowns and radio buttons just don´t do anything. When using the channel zigbee2mqtt/thermostat/set/<weekday>-<number>-<temperature/hour/minute>, a value is sent to Zigbee2mqtt, but the log says there is "no converter" for that channel (although the same channel without /set delivers the numbers correctly). Can anyone help me with that part?

TAKeanice, 2022-11-03:
I assume the same is true for away_schedule, which is also a composite

IerwinDE, 2022-11-14:
I tried the essentials.js from TAKeanice as an external converter with the Zigbeee2Mqtt based Adapter in ioBroker.
The thermostat is detected but the only value I an able to set on the device is the mode (off/auto/heat).
The only values that is sucessfully received seems to be local_temperature and Im not even sure that one updates.

Is anyone else using the converter with ioBroker and can confirm if it works for them?

TAKeanice, 2022-11-14:
I have tested with my essentials smart home thermostat. With the gui of Zigbee2Mqtt the converter works fine. Do you have logs which can tell me a little more about the problems?

The Lidl renaming trick works but there are missing values (i.e., warnings) and some other minor differences.

I will overhaul the converter anyway, as suggested above, and revisit if there are any problems or differences to my local version.

Longlife4ever, 2022-11-14:
> I have tested with my essentials smart home thermostat. With the gui of Zigbee2Mqtt the converter works fine. Do you have logs which can tell me a little more about the problems?
>
> The Lidl renaming [...]

Hi TAKeanice,

is there a way to get this implementet in the Zigbee Adapter of iobroker? Thought it is based on the Zigbee2Mqtt.

TAKeanice, 2022-11-14:
@Longlife4ever I think that is true, but I am using Zigbee2Mqtt in Openhab, so I don´t really know how IOBroker handles things. I just know that the converter should convert zigbee messages correctly to mqtt messages, and that the exposes definition of the adapter makes the UI that zigbee2mqtt shows work correctly. But as I said, I will look into it shortly.

mucke1985, 2022-11-18:
> Hi TAKeanice,
>
> is there a way to get this implementet in the Zigbee Adapter of iobroker? Thought it is based on the Zigbee2Mqtt.

Hello I have the same problem I opened a thread in the IoBroker forum

(German language)

https://forum.iobroker.net/topic/60098/zigbee2mqtt-converter-me16

TAKeanice, 2022-11-18:
@mucke1985 I realized that the converter only works as external converter currently, because I didn't change all paths of the dependencies of the converter when I moved it. Could you try to take the essentials.js out of the devices folder, put it into the data folder of zigbee2mqtt and register it as external converter in the config file?

I will refactor and test the converter this weekend, afterwards you can retest it.

mucke1985, 2022-11-19:
> 

I tested the converter file me167.js in several locations, it always ends up with the same result.

Unfortunately, I can't follow the instructions on the zigbee2mqtt page for inserting the converter because the file structure in connection with the IoBroker adapter is different.

Unfortunately, I don't have any idea where I could look for the problem because my JS knowledge is not sufficient to understand how the three components (zigbee2mqtt/Converter/IoBroker) work together

I have also read in another forum post that some data is only sent every 10 minutes, which should also be the case with the current temperature.

I'm not sure what the problem is, I have three options.

1, The converter works correctly but the data is not transferred correctly to the iobroker adapter

2, The converter works correctly but the incoming data such as the temperature is always overwritten with (null) because these are only sent every 10 minutes.

3, the converter does not work properly

mucke1985, 2022-11-19:
> @Longlife4ever I think that is true, but I am using Zigbee2Mqtt in Openhab, so I don´t really know how IOBroker handles things. I just know that the converter should convert zigbee messages [...]

TAKeanice, 2022-11-19:
@mucke1985 I just realized that the thermostat you have and the converter you use for it is a different one than this issue talks about. Please open an issue for your converter after ruling out that there is some other problem with your setup.

mucke1985, 2022-11-19:
> @mucke1985 I just realized that the thermostat you have and the converter you use for it is a different one than this issue talks about. Please open an issue for your converter after ruling out that there is some other problem with your setup.

oh sorry i completely overlooked that

TAKeanice, 2022-12-20:
Another 30 days... But I am still on it. A bit difficult in Christmas time to find time.

TAKeanice, 2023-01-22:
The integration is essentially ready (no pun intended). @Koenkk requested that I make the schedule a general function to use in other converters, that's why this is still open. Has anyone in this issue already tested the most current version on my fork of the converters project?

pianojockl, 2023-01-28:
Yes, I've been using it for about a week with homeassistant. Everything seems to work - really the only thing is the difference in interface between the essentials thermostat and another one I have (https://www.zigbee2mqtt.io/devices/TS0601_thermostat_1.html).
I'm not using the heating schedule and instead letting homeassistant handle setting the desired temperatures - for the TS0601 that means setting "system_mode" to auto and "preset" to manual. For the essential thermostat however, I'm setting "system_mode" to manual and there's no "preset" option. Also, the TS0601 shows me its "position", i.e. how far its valve is opened.
If anything, it'd be nice to have a consistent behavior throughout different models...
In any case though: thank you so much for your work already!

TAKeanice, 2023-01-28:
Thank you for the feedback! Happy to hear it's working for you. I may take a look at the valve position variable in the definition of your thermostat and if it exists in the essentials thermostat, too. There is just one variable to set the mode of operation on the essentials thermostat, but the definition I used as starting point distributed it to a "preset" and "mode" variable. I would not like to do that, because it creates strange dependencies between two seemingly independent settings.

closed 2024-12-23 by Koenkk