Koenkk/zigbee2mqtt #14240: [Add to device support]: Sinope TH1300ZB

bT0mm, 2022-09-29:
### Link

https://www.sinopetech.com/en/product/smart-floor-heating-thermostat-3600-w-zigbee/

### Database entry

{"id":19,"type":"Router","ieeeAddr":"0x500b914000021bb1","nwkAddr":43501,"manufId":4508,"manufName":"Sinope Technologies\u0000\u0000 \u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":"TH1300ZB\u0000\u0000\u 0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epLis t":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,513,516,1026,1794,2820,2821,65281],"outClusterLi st":[10,65281,25],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"hwVersion":0,"modelId":"TH1300ZB\u0000\u0000\u0000\u0000\u 0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","zclVersion":2}}," haElectricalMeasurement":{"attributes":{"1361":1453400,"acPowerMultiplier":1,"acPowerDivisor":1,"acCurrentMultiplier":1,"acCurrentDi visor":100,"acVoltageMultiplier":1,"acVoltageDivisor":1,"activePower":0,"rmsCurrent":0,"rmsVoltage":240,"apparentPower":2032}},"hvac Thermostat":{"attributes":{"localTemp":2223,"occupiedHeatingSetpoint":2200,"systemMode":4,"pIHeatingDemand":0,"outdoorTemp":-32768," ocupancy":1}},"manuSpecificSinope":{"attributes":{"112":0,"301":2250,"GFCiStatus":0,"timeFormatToDisplay":0,"floorLimitStatus":0,"te mperatureSensor":0,"floorMinHeatSetpointLimit":2200,"floorMaxHeatSetpointLimit":2700,"outdoorTempToDisplay":1142,"floorControlMode": 1,"ambiantMaxHeatSetpointLimit":2350,"outdoorTempToDisplayTimeout":10800}},"msTemperatureMeasurement":{"attributes":{"tolerance":50} },"genDeviceTempCfg":{"attributes":{}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cbf2b3","endpointID ":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cbf2b3","endpointID":1},{"cluster":4,"type":"endpoint","deviceI eeeAddress":"0x00124b0021cbf2b3","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cbf2b3","endpoint ID":1},{"cluster":516,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cbf2b3","endpointID":1},{"cluster":2820,"type":"endpoint"," deviceIeeeAddress":"0x00124b0021cbf2b3","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cbf2b3"," endpointID":1},{"cluster":65281,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cbf2b3","endpointID":1}],"configuredReportings":[ {"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":8,"minRepIntval":0,"maxRepInt val":3600,"repChange":10},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":2820,"attrId":1 291,"minRepIntval":10,"maxRepIntval":305,"repChange":1},{"cluster":2820,"attrId":1288,"minRepIntval":10,"maxRepIntval":306,"repChang e":100},{"cluster":2820,"attrId":1285,"minRepIntval":10,"maxRepIntval":307,"repChange":5},{"cluster":65281,"attrId":277,"minRepIntva l":1,"maxRepIntval":3600,"repChange":1},{"cluster":65281,"attrId":268,"minRepIntval":1,"maxRepIntval":3600,"repChange":1}],"meta":{} }},"appVersion":109,"stackVersion":0,"hwVersion":0,"dateCode":"20191021\u0000\u0000\u0000\u0000\u0000\u0000\u0000","swBuildId":"746\ u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","zclVersion":2,"interviewCompleted":true,"meta":{"configure d":1173388086},"lastSeen":1664421625178}

### Comments

I can request report of two parameters through Dev Console but they don't show in MQTT or hassio.
I tried to configure expose of those paramaters but this is unsupported by device.
They also appear in pairing database entry.

I can request data by MQTT publish to topic "zigbee2mqtt/Cuisine Thermostat/set"
{
  "read": {"cluster": "manuSpecificSinope", "attributes": ["floorMinHeatSetpointLimit","floorMaxHeatSetpointLimit"]}
}
I tried creating an external converter for this but z2m fails to start after this... I'm doing something wrong.

log:
Received MQTT message on 'zigbee2mqtt/Cuisine Thermostat/set' with data '{ "read": {"cluster": "manuSpecificSinope", "attributes": ["floorMinHeatSetpointLimit"]} }'
Debug Publishing 'set' 'read' to 'Cuisine Thermostat'
Debug Received Zigbee message from 'Cuisine Thermostat', type 'readResponse', cluster 'manuSpecificSinope', data '{"floorMinHeatSetpointLimit":2200}' from endpoint 1 with groupID 0
Info Read result of 'manuSpecificSinope': {"floorMinHeatSetpointLimit":2200}

### External converter

```shell
sinope_TH1300ZB_custom: {
    cluster: 'manuSpecificSinope',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        const result = {};
        if (msg.data.hasOwnProperty('floorMinHeatSetpointLimit')) {
            result[postfixWithEndpointName('floorMinHeatSetpointLimit', msg, model, meta)] = precisionRound(msg.data['floorMinHeatSetpointLimit'], 2) / 100;
        }
        if (msg.data.hasOwnProperty('floorMaxHeatSetpointLimit')) {
            result[postfixWithEndpointName('floorMaxHeatSetpointLimit', msg, model, meta)] = precisionRound(msg.data['floorMaxHeatSetpointLimit'], 2) / 100;
        }
        return result;
    }
}
```

danieledwardgeorgehitchcock, 2022-09-29:
Hi,

Couple of things - can you post up your full external converter? The above looks like it is just a fromZigbee entry..

Secondly, can you post the log messages of what happens when you try to start up Z2M with the external converter?

bT0mm, 2022-09-30:
Hi,
here is my full external converter, as this device is already supported and I'm only trying to add functions I thought I didn't need much... see attached file:
[external_conv_customTH1300ZB.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/9679805/external_conv_customTH1300ZB.js.txt)

I can't figure out why the logs are empty when it fails to run
I'm running z2m in a python virtual environment, looking at logs in /opt/zigbee2mqtt/data/log, files are created but empty

Thanks

closed 2022-11-07 by github-actions[bot]