Koenkk/zigbee2mqtt #14243: SONOFF Zigbee 3.0 USB Dongle Plus V2 (EFR32MG21) - Waiting in HASSIO

raducostinas, 2022-09-29:
### What happened?

Zigbee2MQTT logs:

```
09:29:14] INFO: Zigbee Herdsman debug logging enabled
[09:29:14] INFO: Socat not enabled
[09:29:15] INFO: Handing over control to Zigbee2MQTT Core ...
Zigbee2MQTT:debug 2022-09-29 09:29:17: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-09-29 09:29:17: Logging to console and directory: '/config/zigbee2mqtt/log/2022-09-29.09-29-17' filename: log.txt
Zigbee2MQTT:debug 2022-09-29 09:29:17: Removing old log directory '/config/zigbee2mqtt/log/2022-09-29.02-43-48'
Zigbee2MQTT:info  2022-09-29 09:29:17: Starting Zigbee2MQTT version 1.27.2 (commit #unknown)
Zigbee2MQTT:info  2022-09-29 09:29:17: Starting zigbee-herdsman (0.14.53)
Zigbee2MQTT:debug 2022-09-29 09:29:17: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220719095939-if00"}}'
2022-09-29T07:29:17.501Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220719095939-if00","adapter":"ezsp"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-09-29T07:29:17.503Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220719095939-if00 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2022-09-29T07:29:17.509Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2022-09-29T07:29:17.509Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2022-09-29T07:29:17.509Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2022-09-29T07:29:17.510Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
```

Zigbee2MQTT configuration.yaml

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://core-mosquitto:1883
  base_topic: zigbee2mqtt
  user: mqtt_user
  password: secret
serial:
  port: >-
    /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220719095939-if00
  adapter: ezsp
advanced:
  network_key: GENERATE
zigbee_herdsman_debug: true
```

Same behaviour in `dev-branch`
Frontend shows a `502: Bad Gateway` error

### What did you expect to happen?

Normal operation. Launching the frontend

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

ncp-uart-sw_EZNet6.10.3_V1.0.1.gbl

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2 (EFR32MG21)

kirovilya, 2022-09-29:
something is wrong with your port - it is not responding to the request. Do you have your own firmware or native, which was in the stick?
maybe in the firmware the port speed is not 115200?

raducostinas, 2022-09-29:
> 
This is the firmware I've uploaded to the device

https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/NCP

kirovilya, 2022-09-29:
This is the official firmware for Sonoff.
strange that the port is not responding.
check the stick with the utility https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/

raducostinas, 2022-09-29:
> This is the official firmware for Sonoff. strange that the port is not responding. check the stick with the utility https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/

python3 Elelabs_EzspFwUtility.py probe -p /dev/tty.wchusbserial54DD0051021
2022/09/29 09:59:38 Elelabs_EzspFwUtility:   EZSP adapter in bootloader mode detected:
2022/09/29 09:59:38 Elelabs_EzspFwUtility:   Gecko Bootloader v1.12.00

kirovilya, 2022-09-29:
@raducostinas switch to normal mode

raducostinas, 2022-09-29:
> 

python3 Elelabs_EzspFwUtility.py probe -p /dev/tty.wchusbserial202207190959391
2022/09/29 10:01:37 Elelabs_EzspFwUtility:   Couldn't communicate with the adapter in Zigbee (EZSP) mode, Thread (Spinel) mode or bootloader mode

kirovilya, 2022-09-29:
here is the problem. Apparently the stick was badly flashed. try to flash again.
or you can try alternative firmware from here https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP

raducostinas, 2022-09-29:
@kirovilya this is output when I try to flash again.

```
2/09/29 10:08:15 Elelabs_EzspFwUtility:   Gecko Bootloader v1.12.00
2022/09/29 10:08:15 Elelabs_EzspFwUtility:   Allready in bootloader mode. No need to restart
2022/09/29 10:08:16 Elelabs_EzspFwUtility:   Successfully restarted into X-MODEM mode! Starting upload of the new firmware... DO NOT INTERRUPT(!)
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....

2022/09/29 10:08:50 Elelabs_EzspFwUtility:   Firmware upload complete
2022/09/29 10:08:50 Elelabs_EzspFwUtility:   Rebooting NCP...
2022/09/29 10:09:05 Elelabs_EzspFwUtility:   Couldn't communicate with the adapter in Zigbee (EZSP) mode, Thread (Spinel) mode or bootloader mode
```

After the auto reboot in normal mode, still not communicating.

I'm trying another firmware

raducostinas, 2022-09-29:
```
python3 Elelabs_EzspFwUtility.py flash -f ncp-uart-sw_7.1.1.0_115200.gbl -p /dev/tty.wchusbserial54DD0051021
2022/09/29 10:14:44 Elelabs_EzspFwUtility:   EZSP adapter in bootloader mode detected:
2022/09/29 10:14:44 Elelabs_EzspFwUtility:   Gecko Bootloader v1.12.00
2022/09/29 10:14:44 Elelabs_EzspFwUtility:   Allready in bootloader mode. No need to restart
2022/09/29 10:14:45 Elelabs_EzspFwUtility:   Successfully restarted into X-MODEM mode! Starting upload of the new firmware... DO NOT INTERRUPT(!)
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
.....
...
2022/09/29 10:15:23 Elelabs_EzspFwUtility:   Firmware upload complete
2022/09/29 10:15:23 Elelabs_EzspFwUtility:   Rebooting NCP...
2022/09/29 10:15:30 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2022/09/29 10:15:30 Elelabs_EzspFwUtility:   Firmware: 7.1.1-17
2022/09/29 10:15:30 Elelabs_EzspFwUtility:   EZSP v9
```

```
python3 Elelabs_EzspFwUtility.py probe -p /dev/tty.wchusbserial54DD0051021
2022/09/29 10:15:41 Elelabs_EzspFwUtility:   Generic Zigbee EZSP adapter detected:
2022/09/29 10:15:41 Elelabs_EzspFwUtility:   Firmware: 7.1.1-17
2022/09/29 10:15:41 Elelabs_EzspFwUtility:   EZSP v9
```

kirovilya, 2022-09-29:
EZSP v9 working only with dev-version of z2m

raducostinas, 2022-09-29:
@kirovilya finally! 🙏🏻 🙏🏻 

[image]

closed 2022-09-29 by raducostinas