Koenkk/zigbee2mqtt #14244: SEA802-Zigbee not returning "heating" state

MG-5, 2022-09-29:
### What happened?

I have three Saswell SEA802-Zigbee paired with your Zigbee2Mqtt.  When I get a message with all relevant values, the heating state is always `Null`. Its also `Null` when the themostat is showing the 🔥 symbol. 

### What did you expect to happen?

The current heating state. 

### How to reproduce it (minimal and precise)

Simply communicating with the devices. 

### Zigbee2MQTT version

1.27.2 commit: 7dc48fb

### Adapter firmware version

20211217

### Adapter

CC2652P2

### Debug log

```Debug 2022-09-29 11:17:06Received Zigbee message from 'Bad-Thermostat', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,220],"type":"Buffer"},"datatype":2,"dp":103}],"seq":768}' from endpoint 1 with groupID 0```

```Info 2022-09-29 11:17:06MQTT publish: topic 'zigbee2mqtt/Bad-Thermostat', payload '{"anti_scaling":"ON","away_mode":"OFF","battery_low":false,"child_lock":"LOCK","current_heating_setpoint":22,"frost_detection":"ON","heating":null,"linkquality":109,"local_temperature":21.5,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"auto","window_detection":"ON"}'```

MG-5, 2022-09-29:
And this is returning, when the thermostat is showing the flame symbol.

```Debug 2022-09-29 11:27:53Received Zigbee message from 'Bad-Thermostat', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,44],"type":"Buffer"},"datatype":2,"dp":103}],"seq":768}' from endpoint 1 with groupID 0```

```Info 2022-09-29 11:27:53MQTT publish: topic 'zigbee2mqtt/Bad-Thermostat', payload '{"anti_scaling":"ON","away_mode":"OFF","battery_low":false,"child_lock":"LOCK","current_heating_setpoint":30,"frost_detection":"ON","heating":null,"linkquality":98,"local_temperature":21.5,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"auto","window_detection":"ON"}'```

machv, 2022-10-04:
I have the same issue with mine TRVs.

KardinalReusen, 2022-10-05:
I can confirm this for my SEA802-Zigbee device

juantxorena, 2022-10-05:
The same for me with the Hama model. However it works with another model, RTX ZB-RT1 (Zigbee Manufacturer _TZE200_yw7cahqs)

zibous, 2022-10-15:
Same Problem `Hama 00176593 Thermostat` here:

[image]

```ini
Zigbee2MQTT version: 1.28.0-dev commit: e3e4a0a4
Coordinator type:  ConBee2/RaspBee2
Coordinator revision : 0x26660700
Frontend version: 0.6.113
```

```json
{
    "anti_scaling": "ON",
    "away_mode": "OFF",
    "battery_low": false,
    "child_lock": "UNLOCK",
    "current_heating_setpoint": 30,
    "frost_detection": "ON",
    "last_seen": "2022-10-15T10:40:12+02:00",
    "linkquality": 231,
    "local_temperature": 23.3,
    "local_temperature_calibration": 0,
    "preset_mode": "none",
    "system_mode": "heat",
    "window_detection": "OFF",
    "device": {
        "applicationVersion": 85,
        "dateCode": "",
        "friendlyName": "thermostat-wohnzimmer2",
        "hardwareVersion": 1,
        "ieeeAddr": "0x847127fffe0954af",
        "manufacturerID": 4098,
        "manufacturerName": "_TZE200_yw7cahqs",
        "model": "SEA801-Zigbee/SEA802-Zigbee",
        "networkAddress": 57759,
        "powerSource": "Battery",
        "stackVersion": 0,
        "type": "EndDevice",
        "zclVersion": 3
    },
    "elapsed": 493261,
    "heating": null  <---- !!!!!!
}
```

iksrd, 2022-11-05:
I have the same problem

closed 2023-01-14 by github-actions[bot]

zibous, 2024-01-13:
With 3 pieces (I get the status “Heating”), with 10 pieces no status.

The only difference is the applicationVersion:

    - 3x “applicationVersion”: 67 works: “heating”: “ON”
          {
               "anti_scaling": "ON",
               "away_mode": "OFF",
               "battery_low": false,
               "child_lock": "LOCK",
               "current_heating_setpoint": 16,
               "device": {
                   "applicationVersion": 67,
                   "dateCode": "",
                   "friendlyName": "thermostat-fitnessraum2",
                   "hardwareVersion": 1,
                   "ieeeAddr": "0xa4c138dd160d7831",
                   "manufacturerID": 4417,
                   "manufacturerName": "_TZE200_h4cgnbzg",
                   "model": "SEA801-Zigbee/SEA802-Zigbee",
                   "networkAddress": 28931,
                   "powerSource": "Battery",
                   "stackVersion": 0,
                   "type": "EndDevice",
                   "zclVersion": 3
               },
               "frost_detection": "ON",
               "heating": "ON",  <------
               "last_seen": "2024-01-12T13:00:38+01:00",
               "linkquality": 255,
               "local_temperature": 17,
               "local_temperature_calibration": 0,
               "preset_mode": "none",
               "system_mode": "heat",
               "window_detection": "ON",
               "elapsed": 3241
           }
          
    - 10x “applicationVersion”: 85 not working: “heating”: null
        {
        "anti_scaling": "ON",
        "away_mode": "OFF",
        "battery_low": false,
        "child_lock": "UNLOCK",
        "current_heating_setpoint": 18,
        "device": {
            "applicationVersion": 85,
            "dateCode": "",
            "friendlyName": "thermostat-fitnessraum",
            "hardwareVersion": 1,
            "ieeeAddr": "0x5c0272fffea346d5",
            "manufacturerID": 4098,
            "manufacturerName": "_TZE200_yw7cahqs",
            "model": "SEA801-Zigbee/SEA802-Zigbee",
            "networkAddress": 25693,
            "powerSource": "Battery",
            "stackVersion": 0,
            "type": "EndDevice",
            "zclVersion": 3
        },
        "frost_detection": "ON",
	    "heating": null   <------
        "last_seen": "2024-01-12T13:37:29+01:00",
        "linkquality": 255,
        "local_temperature": 19,
        "local_temperature_calibration": 0,
        "preset_mode": "none",
        "system_mode": "heat",
        "window_detection": "ON",
        "elapsed": 390141,
    }

Do I need to update the firmware for the 10 and how can I do that or is there another solution ?

Thanks.