Koenkk/zigbee2mqtt #14245: Firmware data not correct in UI

Scoff123, 2022-09-29:
### What happened?

I am seeing the incorrect firmware data reported for one of my Philips Hue bulbs in the Z2M UI. This is subsequent to using the OTA firmware feature in Z2M for this device. The old firmware version still shows in the UI.

I have tried force removing the device, restarting Z2M and adding the device fresh, but still the old incorrect firmware data shows in the UI once the device is successfully added.

### What did you expect to happen?

I expected the new version and build data data to show in the Z2M UI. 

Conversely, I updated an identical Philips Hue bulb using the Philips Hue app on my smartphone, and the new firmware data shows correctly in Z2M. 

### How to reproduce it (minimal and precise)

Update firmware of Philips Hue Bulb using OTA firmware update process in Z2M.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

6.7.8.0 build 373

### Adapter

Sonoff ZBBridge

Koenkk, 2022-09-29:
Probably the OTA update failed because this device requires OTA updates in 2 steps (first an older FW before the newer one). What Hue model is this?

Scoff123, 2022-09-29:
The device is a Philips Hue white ambiance E27 1100lm with Bluetooth bulb,
LTA009
Zigbee Manufacturer
Signify Netherlands B.V.

There was only one firmware update available for the device, which was updating from 1.93.7 to 1.93.11 according to the Hue app.

Using the Hue app to carry the update out on one of my 2 identical bulbs worked correctly, and the updated firmware shows for that bulb in my Z2M UI.
Using the Z2M OTA upgrade on the second bulb appears to have been successful, as viewing this bulb in the Hue app now shows the newer updated firmware. 
Unfortunately Z2M is still showing the old firmware version, but this is incorrect and not what is actually installed on the device. Removing and rejoining the device to Z2M makes no difference, as if the old firmware version is somehow cached or the database entry for firmware version and build date are not getting overwritten correctly for this device following the OTA update.

Koenkk, 2022-09-29:
Can you remove the device, restart z2m and repair the device?

Scoff123, 2022-09-29:
Yes I've tried doing that but unfortunately it makes no difference and the older firmware version and firmware build date show in the Z2M UI oddly

Koenkk, 2022-09-30:
What shows up when you read the values through the z2m frontend -> device -> dev console?

[image]

Scoff123, 2022-09-30:
Interesting, that shows the correct build date and firmware version 

[image]

Whereas on the main device info page it shows the old data

[image]

Koenkk, 2022-09-30:
It should show the correct info when restarting z2m now.

Scoff123, 2022-09-30:
It does indeed - thank you :) 
[image]

How does reading the attribute cause the update of data shown in the UI, and is there any way this can happen automatically without a manual read in future?

Koenkk, 2022-10-01:
When you update the bulb through z2m, z2m should read these attributes after the fw update. It seems something went wrong there. Any chances you still have the debug log of that update?

Scoff123, 2022-10-01:
No unfortunately not :( I'll just keep an eye the next time there is an update, and open an issue with the logs then, if it does the same. I'll close this one down now - thanks for your help.

closed 2022-10-01 by Scoff123