Koenkk/zigbee2mqtt #14246: Sonoff zigbee 3.0 issues with lxc proxmox

guarddog13, 2022-09-29:
### What happened?

I have HAos on Proxmox as a VM running ZHA with the conbee 2 as a coordinator. I wanted to switch to zigbee2mqtt so I purchased a Sonoff 3.0 hoping to make the switch seamlessly. I flashed the dongle and installed zigbee2mqtt. I was able to get the dongle to be recognized once by zigbee2mqtt and when I exited the shell and restarted the zigbee2mqtt lxc the dongle wouldn't be recognized. It's still in the lxc if I use ls -l /dev/serial/by-id. 

I've several times to reflash the device, restarted a new zigbee2mqtt instance. I tried it on a deconz lxc and still nothing. I'm fairly close to returning it and just resetting everything and using my conbee 2 in zigbee2mqtt as this definitely isn't seamless. 

Any ideas or help would be greatly appreciated.

### What did you expect to happen?

I would be able to connect to my controller with minimal effort.

### How to reproduce it (minimal and precise)

I followed videos online to flash the dongle, updated config and tried starting zigbee2mqtt. 

### Zigbee2MQTT version

Most recent

### Adapter firmware version

Most recent from zigbee2mqtt supported site 

### Adapter

Sonoff 3.0 adapter plus

lcsjandrey, 2022-10-01:
I had the same issue. It seems to be related with Proxmox exposing the USB device, I spent several hours to try fix it :(
Unfortunately I ended up creating a Virtual Machine instead of a container and the problem was solved.

guarddog13, 2022-10-01:
> I had the same issue. It seems to be related with Proxmox exposing the USB device, I spent several hours to try fix it :( Unfortunately I ended up creating a Virtual Machine instead of a container and the problem was solved.

I had to edit the config file in my proxmox shell. The script I used was passing through ttyUSB0. My zwave stick already had that..changed it to ttyUSB1 and working well now.

KidA001, 2022-10-02:
I am running HA and Zigbee2MQTT on proxmox as well. Are you using ttek's scripts to install? Once you have your Sonoff 3.0 properly flashed, run this prom your proxmox shell:
 ```bash
bash -c "$(wget -qLO - https://github.com/tteck/Proxmox/raw/main/misc/usb-passthrough.sh)" -s 106` 
```
and replace `106`  with the LXC ID of your zigbee2mqtt instance. Restart your Z2Mqtt instance

Then from your Z2MQTT instance console run:
`ls -l /dev/serial/by-id`

Example Output: `lrwxrwxrwx 1 root root 13 Jun 19 17:30 usb-1a86_USB_Serial-if00-port0 -> ../../ttyUSB0`

Then edit your config and add that as the port
```yaml
serial:
  port: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
```

Read more here: https://tteck.github.io/Proxmox/