Koenkk/zigbee2mqtt #14248: Battery conversion for Visonic MCT 340 E is incorrect

rjbutler, 2022-09-29:
### What happened?

I moved a several of these devices from Hubitat to Home Assistant/Z2M and the battery levels all report significantly lower than they were on Hubitat.  They were mostly accurate on Hubitat.  For example, 50% on Hubitat became 8% on Zigbee2MQTT.  A fresh battery reports at 42% in Zigbee2MQTT but correctly showed 100% in Hubitat.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZDongle-E

rjbutler, 2022-09-29:
This code doesn't look correct to me.  A report of 29 results 42%.  It should be something closer to 90%

debug 2022-09-29 09:17:02: Received Zigbee message from 'Bedroom left window', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryVoltage":### **29**}' from endpoint 1 with groupID 0
info  2022-09-29 09:17:02: MQTT publish: topic 'zigbee2mqtt/Bedroom left window', payload '{"battery":42,"battery_low":false,"contact":false,"linkquality":236,"tamper":false,"temperature":21,"voltage":2900}'

    if (option === '3V_2100') {
        if (voltage < 2100) {
            percentage = 0;
        } else if (voltage < 2440) {
            percentage = 6 - ((2440 - voltage) * 6) / 340;
        } else if (voltage < 2740) {
            percentage = 18 - ((2740 - voltage) * 12) / 300;
        } else if (voltage < 2900) {
            percentage = 42 - ((2900 - voltage) * 24) / 160;
        } else if (voltage < 3000) {
            percentage = 100 - ((3000 - voltage) * 58) / 100;
        } else if (voltage >= 3000) {
            percentage = 100;
        }
        percentage = Math.round(percentage);
    }

Koenkk, 2022-09-29:
You can experiment with different voltage -> battery % calculations. e.g.:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_cymsnfvf'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_2xlvlnez'}],
    model: 'ZP-LZ-FR2U',
    vendor: 'Moes',
    description: 'Zigbee 3.0 dual USB wireless socket plug CUSTOM',
    fromZigbee: [fz.on_off, fz.tuya_switch_power_outage_memory],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'),
        exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage')],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await device.getEndpoint(1).read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(device.getEndpoint(1));
        await reporting.onOff(device.getEndpoint(2));
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, wait till new battery report comes in

rjbutler, 2022-09-29:
Thanks @Koenkk  I'll give it a try.   Do you think the logic in the existing converter is correct though?  As I said the maximum battery percentage it will convert to is 42% unless the battery is over 3v in which case it's hard coded to 100.  That doesn't seem right.

Koenkk, 2022-09-30:
None of the logics are correct, voltage to battery % conversion is quite hard. There are a lot of variables to take into account, temperature, discharge curve etc. The goal of battery % is to provide a rough estimation when you need to replace the battery (e.g. < 10%). So please experiment with different algorithms.

closed 2022-11-07 by github-actions[bot]