Koenkk/zigbee2mqtt #14249: [New device support]: Aqara E1 3 gang switch (with neutral)

fata1ityz, 2022-09-29:
### Link

https://homekitnews.com/2021/05/07/aqara-introduce-new-eu-style-e1-switches/

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x54ef441000570681","nwkAddr":28979,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.switch.acn040","epList":[1,2,3,41,42,43,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,2,3,4,5,6,18,64704],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.acn040","appVersion":27}},"aqaraOpple":{"attributes":{"223":{"type":"Buffer","data":[22,33,0,0,23,33,0,0,12,33,0,0,24,33,0,0,30,33,0,0,31,33,39,0,32,33,0,0,34,33,27,1,35,33,0,0,36,32,0,37,35,172,8,139,15,250,33,0,0,251,33,0,0,252,35,0,0,0,0,253,35,0,0,0,0,254,35,128,0,4,0]},"229":{"type":"Buffer","data":[1,0,1,0,0,4,45]},"247":{"type":"Buffer","data":[100,16,0,101,16,1,102,16,0,3,40,33,5,33,1,0,154,32,0,8,33,27,1,9,33,0,1,11,32,0,10,33,0,0,12,32,1]}}},"genOnOff":{"attributes":{"245":57750272,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6,18,64704],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"245":117440518,"onOff":1}},"aqaraOpple":{"attributes":{"512":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[0,3,4,5,6,18,64704],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"245":57750272,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"41":{"epId":41,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"42":{"epId":42,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":2}}},"binds":[],"configuredReportings":[],"meta":{}},"43":{"epId":43,"inClusterList":[],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":27,"stackVersion":2,"hwVersion":1,"dateCode":"07-22-2022","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2114939527},"lastSeen":1664467291871,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the new device support guide but got stuck at detecting action for different button. As this is a 3 gang wall switch, the action should trigger like: 'single_left', 'double_left', 'single_center', 'double_center', 'single_right', 'double_right', but regardless which button was press, it only trigger 'single' or 'double'. Have tried creating the converter base on Aqara E1 2 Gang switch but same result. Debug logs below:

`Debug 2022-09-30 00:17:07Received Zigbee message from 'kitchen_switch', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 41 with groupID 0
Info 2022-09-30 00:17:07MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"action":"single","device_temperature":33,"flip_indicator_light":null,"led_disabled_night":null,"linkquality":76,"operation_mode_center":"decoupled","operation_mode_left":null,"operation_mode_right":null,"power":null,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","state_center":"ON","state_left":"ON","state_right":"OFF","update":{"state":null},"update_available":null,"voltage":null}'
Info 2022-09-30 00:17:07MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"action":"","device_temperature":33,"flip_indicator_light":null,"led_disabled_night":null,"linkquality":76,"operation_mode_center":"decoupled","operation_mode_left":null,"operation_mode_right":null,"power":null,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","state_center":"ON","state_left":"ON","state_right":"OFF","update":{"state":null},"update_available":null,"voltage":null}'
Info 2022-09-30 00:17:07MQTT publish: topic 'zigbee2mqtt/kitchen_switch/action', payload 'single'
Debug 2022-09-30 00:17:25Saving state to file /config/zigbee2mqtt/state.json
Debug 2022-09-30 00:17:36Received Zigbee message from 'kitchen_switch', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":2}' from endpoint 42 with groupID 0
Info 2022-09-30 00:17:36MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"action":"double","device_temperature":33,"flip_indicator_light":null,"led_disabled_night":null,"linkquality":58,"operation_mode_center":"decoupled","operation_mode_left":null,"operation_mode_right":null,"power":null,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","state_center":"ON","state_left":"ON","state_right":"OFF","update":{"state":null},"update_available":null,"voltage":null}'
Info 2022-09-30 00:17:36MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"action":"","device_temperature":33,"flip_indicator_light":null,"led_disabled_night":null,"linkquality":58,"operation_mode_center":"decoupled","operation_mode_left":null,"operation_mode_right":null,"power":null,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","state_center":"ON","state_left":"ON","state_right":"OFF","update":{"state":null},"update_available":null,"voltage":null}'
Info 2022-09-30 00:17:36MQTT publish: topic 'zigbee2mqtt/kitchen_switch/action', payload 'double'
Debug 2022-09-30 00:17:57Received Zigbee message from 'kitchen_switch', type 'attributeReport', cluster 'genOnOff', data '{"245":57750272,"onOff":1}' from endpoint 3 with groupID 0
Info 2022-09-30 00:17:57MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"action":null,"device_temperature":33,"flip_indicator_light":null,"led_disabled_night":null,"linkquality":58,"operation_mode_center":"decoupled","operation_mode_left":null,"operation_mode_right":null,"power":null,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","state_center":"ON","state_left":"ON","state_right":"ON","update":{"state":null},"update_available":null,"voltage":null}'
Debug 2022-09-30 00:17:58Received Zigbee message from 'kitchen_switch', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 43 with groupID 0
Info 2022-09-30 00:17:58MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"action":"single","device_temperature":33,"flip_indicator_light":null,"led_disabled_night":null,"linkquality":61,"operation_mode_center":"decoupled","operation_mode_left":null,"operation_mode_right":null,"power":null,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","state_center":"ON","state_left":"ON","state_right":"ON","update":{"state":null},"update_available":null,"voltage":null}'
Info 2022-09-30 00:17:58MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"action":"","device_temperature":33,"flip_indicator_light":null,"led_disabled_night":null,"linkquality":61,"operation_mode_center":"decoupled","operation_mode_left":null,"operation_mode_right":null,"power":null,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","state_center":"ON","state_left":"ON","state_right":"ON","update":{"state":null},"update_available":null,"voltage":null}'
Info 2022-09-30 00:17:58MQTT publish: topic 'zigbee2mqtt/kitchen_switch/action', payload 'single'
Debug 2022-09-30 00:18:22Received Zigbee message from 'kitchen_switch', type 'attributeReport', cluster 'genOnOff', data '{"245":57750272,"onOff":0}' from endpoint 3 with groupID 0
Info 2022-09-30 00:18:22MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"action":null,"device_temperature":33,"flip_indicator_light":null,"led_disabled_night":null,"linkquality":69,"operation_mode_center":"decoupled","operation_mode_left":null,"operation_mode_right":null,"power":null,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","state_center":"ON","state_left":"ON","state_right":"OFF","update":{"state":null},"update_available":null,"voltage":null}'
Debug 2022-09-30 00:18:23Received Zigbee message from 'kitchen_switch', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 43 with groupID 0
Info 2022-09-30 00:18:23MQTT publish: topic 'zigbee2mqtt/kitchen_switch', payload '{"action":"single","device_temperature":33,"flip_indicator_light":null,"led_disabled_night":null,"linkquality":80,"operation_mode_center":"decoupled","operation_mode_left":null,"oper [... cut ...]
                'single_left_center', 'double_left_center', 'single_left_right', 'double_left_right',
                'single_center_right', 'double_center_right', 'single_all', 'double_all']),
            e.power_outage_memory(), e.device_temperature(), e.flip_indicator_light(),
        ],
        onEvent: preventReset,
        configure: async (device, coordinatorEndpoint, logger) => {
            await device.getEndpoint(1).write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
        },
```

wilsonkhlam, 2022-10-31:
Will the next version support this switch? I have bought this switch but it is not working.

fata1ityz, 2022-11-09:
@wilsonkhlam found out how to support for this switch but not sure how to commit the code here. Will need to add the switch model ZNQBKG31LM into [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters)/[converters](https://github.com/Koenkk/zigbee-herdsman-converters/tree/master/converters)/toZigbee.js 

[image]

wilsonkhlam, 2022-11-13:
@fata1ityz, can you post the change of `fromZigbee.js` and `toZigbee.js` here? I can help apply your change and create pull request to the `zigbee-herdsman-converters` repository.

closed 2022-12-22 by github-actions[bot]