Koenkk/zigbee2mqtt #14253: Aqara QBKG12LM and QBKG11LM return wrong power source (battery)

mikolka-lsh, 2022-09-29:
### What happened?

Two of many Aqara QBKG12LM and QBKG11LM in my network returns wrong power source - battery... as far as it is wall switches for 1 and 2 rocks...
You could see this on screenshot.
I try to pair it again - same situation exactly with this two switches.
[image]

### What did you expect to happen?

It must return "Mains (single phase)" as all other switches of this model...

### How to reproduce it (minimal and precise)

I don't really know as far as this issue only with dedicated switches...

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20210813

### Adapter

zStack3x0

Koenkk, 2022-09-29:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

mikolka-lsh, 2022-10-01:
Koenkk, last update 1.28.0 solves this situation but if I try to see a map - this two switches still counts not as a router but as end devices...
[image]

Koenkk, 2022-10-02:
The power source doesn't determine router/endevice. Can you provide the herdsman debug log of: force device through z2m frontend, restart z2m and pair it?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

mikolka-lsh, 2022-10-03:
And now this switch still as a end device on map of my network...

mikolka-lsh, 2022-10-04:
Can I grab it now if I'm switch off Hardeman after all this operations? Or I need turn it on again and reproduce all steps again?

mikolka-lsh, 2022-10-04:
Just grub it now with: docker logs CONTAINER_ID > /tmp/container.log 2>&1
P.S. zigbee_herdsman_debug: true turn off (delete this line from configuration) yesterday just after repairing and log grabbing...
[container.log](https://github.com/Koenkk/zigbee2mqtt/files/9708928/container.log)

Koenkk, 2022-10-04:
> P.S. zigbee_herdsman_debug: true turn off (delete this line from configuration)

I don't see the herdsman debug log in the log you  provided, please enable this.

mikolka-lsh, 2022-10-04:
Sorry it was my mistake - I'm redacting socat part of file (in UI) but whe I'm switched to YAML - all done better and I can see herdsman lines now in log...

Noow I done same steps:
1. Add zigbee_herdsman_debug: true to HA addon
2. Restart addon
3. Delete guest.room-switch from Z2M UI
4. Add this device again
5. Rename devise again to guest.room-switch
6. Open SSH and grub all logs from Z2M container
7. Switch off herdsman logs

[container.log](https://github.com/Koenkk/zigbee2mqtt/files/9709338/container.log)

Koenkk, 2022-10-04:
Thanks for the log this is what I need. I missed one step: first force remove the device through the z2m frontend, restart z2m and then capture the logs of pairing it.

mikolka-lsh, 2022-10-04:
I'm remove this device from UI but after restarting as I write before... This is not it?  You need just pairing in herdsman without removing?

mikolka-lsh, 2022-10-05:
> first force remove the device through the z2m frontend, restart z2m and then capture the logs of pairing it

Here it is: [container.log](https://github.com/Koenkk/zigbee2mqtt/files/9713159/container.log)

Koenkk, 2022-10-05:
Thanks, can you now provide the data/database.db entry part of this device? This switch indeed announces itself as a router.

mikolka-lsh, 2022-10-05:
How to grub part of database? Or just to pack it full?

Koenkk, 2022-10-05:
You can open it in a text editor.

mikolka-lsh, 2022-10-06:
Sorry for such long response.
Here it is:
{"id":19,"type":"Router","ieeeAddr":"0x00158d0002c20a0f","nwkAddr":44942,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.ctrl_ln1.aq1","epList":[1,2,3,5],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,3,6,16,5,10,1,2],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65281":{"3":28,"5":428,"8":4446,"9":2816,"100":0,"149":10.810918807983398,"152":0,"154":0},"65521":0,"modelId":"lumi.ctrl_ln1.aq1","appVersion":1,"manufacturerName":"LUMI","powerSource":1,"zclVersion":1,"stackVersion":2,"hwVersion":18,"dateCode":"12-23-2021"}},"genOnOff":{"attributes":{"61440":61836800,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":9,"inClusterList":[12],"outClusterList":[12,4],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":83,"inClusterList":[12],"outClusterList":[12],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":0,"inClusterList":[18,16],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":18,"dateCode":"12-23-2021","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1946197436},"lastSeen":1665042939483,"defaultSendRequestWhen":"immediate"}

If you need to compare entries between the same switches - here the same switch but looks like a router on the map:
{"id":2,"type":"Router","ieeeAddr":"0x00158d0002c239d0","nwkAddr":39599,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.ctrl_ln1.aq1","epList":[1,2,3,5],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,3,6,16,5,10,1,2],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65281":{"3":28,"5":487,"8":4446,"9":2816,"100":0,"149":0.9194470643997192,"152":0,"154":0},"65521":0,"modelId":"lumi.ctrl_ln1.aq1","appVersion":1,"manufacturerName":"LUMI","powerSource":1,"zclVersion":1,"stackVersion":2,"hwVersion":18,"dateCode":"12-23-2021"}},"genOnOff":{"attributes":{"61440":60468992,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":9,"inClusterList":[12],"outClusterList":[12,4],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":83,"inClusterList":[12],"outClusterList":[12],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":0,"inClusterList":[18,16],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":18,"dateCode":"12-23-2021","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1946197436},"lastSeen":1665043082929,"defaultSendRequestWhen":"immediate"}

mikolka-lsh, 2022-10-06:
How to create such log?

mikolka-lsh, 2022-10-06:
If I understand correctly - here it is: https://pastebin.com/qzCReSLz

mikolka-lsh, 2022-10-06:
Yes, "bathroom.1-switch-main" - it is second problematic device...
Before this we debug just "guest.room-switch"...

mikolka-lsh, 2022-10-07:
I'm still waiting if you need additional debug from second one problem switch - "bathroom.1-switch-main" still shows as end device on map and I'm even not try to re-pair it...

Koenkk, 2022-10-07:
Try the same as you did for the other device; force remove the device through the z2m frontend, restart z2m and pair the device

mikolka-lsh, 2022-10-09:
Thank you, @Koenkk!
This issue is fully adressed and resolved!

closed 2022-10-09 by mikolka-lsh