Koenkk/zigbee2mqtt #14257: Lonsonho zigbee dimmers (1 gang and 2 gang) are out of sync between physical switch and remote switching

JHeijting, 2022-09-30:
### What happened?

I use various Lonsonho [TS110E_1gang](https://www.zigbee2mqtt.io/devices/TS110E_1gang.html#lonsonho-ts110e_1gang) and one [TS110E_2gang](https://www.zigbee2mqtt.io/devices/TS110E_2gang.html#lonsonho-ts110e_2gang) in combination with mostly router devices and the ITead Sonoff Zigbee 3.0 USB Dongle Plus model ZBDongle-E coordinator with Z2M as addon (version 1.27.2-1) in home assistant. The Zigbee dimmers are connected physically to pulse switches. For the 2gang dimmer, they are exposed as 'momentary' switch (for the 1gang this setting is not available it seems).

Since using this coordinator (recently upgraded from Texas Instruments CC2531) the signal strenght on each device is perfect and the network is working very snappy.

I believe to have a wrong configuration of doing something wrong, or maybe have found an issue with these type or dimmers or a bug.

The issue appears to be caused by 2 issues:
- Switching remotely using Home Assistant makes the dimmer go out of sync between Home Assistant and the physical switch;
- After switching remotely using Home Assistant, for some reason the brightness goes to 0 and after turning the light 'on' with the physical switch it will be on, but on brightness 0.

### What did you expect to happen?

I expected that no matter how I switch the light using these Zigbee dimmers, the state will always be in sync between Home Assistant and the physical switch. I also expect this to be for setting the brightness.

### How to reproduce it (minimal and precise)

Reproduction - current state: light is off and home assistant entity is off:
- Turn on light using home assistant entity - light goes on and home assistant entity turns on (as expected);
- Press physical switch - nothing happens, light stays on and home assistant entity stays on (not as expected);
- Press physical switch (second time) - light goes off and home assistant entity goes off (not really as expected, but now it seems in sync again);
- press physical switch - light goes on and home assistant entity goes on (as expected);
- Turn off light using home assistant entity - light goes off and home assistant entity turns off (as expected);
- Press physical switch - nothing happens, light stays off and home assistant entity stays off (not as expected);
- Press physical switch (second time) - light stays off and home assistant entity turns on (not as expected). NOTE: in home assistant I can now see the brightness being 0. This can now be fixed in 2 ways. 
- Options 1: hold the physical switch (now first the light goes on the lowest brightness, now release the switch and hold it again to cycle to a higher brightness). 
- Option 2: manually set the brightness in Home assistant entity.

### Zigbee2MQTT version

1.27.2-1

### Adapter firmware version

EZSP v8 -  6.10.3.0 build 297

### Adapter

ITead Sonoff Zigbee 3.0 USB Dongle Plus model ZBDongle-E

AlexMKX, 2022-10-05:
I have same problem. Try the "optimistic" mode of the dimmer. It should work.

JHeijting, 2022-10-05:
Thank you for your reply! I have the dimmers set on "optimistic" mode. This was turned on by default and I haven't changed it. However, I _might_ have found a fix for now by force deleting the device and re-adding it: https://github.com/Koenkk/zigbee2mqtt/discussions/11091

closed 2022-11-12 by github-actions[bot]