Koenkk/zigbee2mqtt #14258: [New device support]: On-Air Combi CTW,303-0136

Keeper127, 2022-09-30:
### Link

http://www.hfh-solutions.eu/

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0x00124b00234169a0","nwkAddr":18991,"manufId":0,"manufName":"hfh Solution GmbH & CO.KG","powerSource":"Mains (single phase)","modelId":"On-Air Combi CTW,303-0136","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":544,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"hfh Solution GmbH & CO.KG","dateCode":"20131206","powerSource":1,"swBuildId":"1.0.8","modelId":"On-Air Combi CTW,303-0136"}},"lightingColorCtrl":{"attributes":{"currentHue":0,"currentSaturation":0,"remainingTime":0,"currentX":24939,"currentY":24701,"colorTemperature":500,"colorMode":2,"numPrimaries":3,"primary1X":45915,"primary1Y":19615,"primary1Intensity":159,"primary2X":9090,"primary2Y":53397,"primary2Intensity":240,"primary3X":9895,"primary3Y":1490,"primary3Intensity":254,"enhancedCurrentHue":0,"enhancedColorMode":0,"colorLoopActive":0,"colorLoopDirection":0,"colorLoopTime":25,"colorLoopStartEnhancedHue":8960,"colorLoopStoredEnhancedHue":0,"colorCapabilities":16,"colorTempPhysicalMin":155,"colorTempPhysicalMax":495}},"genIdentify":{"attributes":{"identifyTime":0}},"genGroups":{"attributes":{"nameSupport":0}},"genScenes":{"attributes":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"attributes":{"onOff":0,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genOnOffSwitchCfg":{"attributes":{}},"genLevelCtrl":{"attributes":{"currentLevel":170,"remainingTime":0}}},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":544,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"genLevelCtrl":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.8","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1664538422377,"defaultSendRequestWhen":"immediate"}

### Comments

pairing was ok, but no possibility to control it via iobroker. the raw controls via:
{
    "devId": "zigbee.0.00124b00234169ea",
    "ep": "11",
    "cid": "genOnOff",
    "cmd": "on",
    "cmdType": "functional",
    "zclData": {
        "onOff": {}
    },
    "cfg": null
}
works, but i cannot integrate it to iobroker.

I tried to add new file with the example of this page :https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
but after rebooting the zigbee controller, the zigbee network crases and could not be started correctly.

If you need some more information about the devices (i got 2 of these) please let me know, so i can give you support

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['On-Air Combi CTW,303-0136'],
    model: 'Light1',
    vendor: 'hfh Solution GmbH & CO.KG',
    description: 'LED-Controller',
    // Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    // - extend.light_onoff_brightness
    // - extend.light_onoff_brightness_colortemp
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    extend: extend.light_onoff_brightness_colortemp({colorTempRange: [155, 495]}),
};

module.exports = definition;
```

### Supported color modes

color temperature

### Color temperature range

"colorTempPhysicalMin":155,"colorTempPhysicalMax":495}

Koenkk, 2022-09-30:
Converter looks good so added it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-30 by Koenkk

Keeper127, 2022-10-01:
Hi Koenkk,
thanks for helping, but please keep in mind, that i just copied the external converter from the axample. I dont know if this configuration is set up correctly. I just know, what to to in the expert view of the iobroker zigbee adapter to switch on an of the LED-controller.

Is the database entry helpful? or do you know a way to figure out, which kind of converters will be needed?

Best regards

Koenkk, 2022-10-01:
There is a high probability that this will work so please check with the next z2m version.

Keeper127, 2022-10-01:
do you also know if this will be included in one of the next updates of the zigbee adapter of iobroker? I have seen that you're also working on this right?
Or is there a way to update this adapter with zhe z2m version? If so, would it be possible to give some instruction by your side, how to do this?

Sorry for my replies but im not familiar with this part of the software.

Best regards

Koenkk, 2022-10-02:
You should ask the iobroker team, I'm not involved in iobroker.