Koenkk/zigbee2mqtt #14277: After upgrade to 1.28.0 device MOSZB-140 unable to update status

Xaldimo, 2022-10-01:
### What happened?

After I upgraded to 1.28.0 all devices worked as expected except MODZB-140. None of the exposed fields would update. This includes:
- Temperature
- Occupancy
- Tampering
- Illuminance_lux
The only field updated is the link quality.

### What did you expect to happen?

Device to update fields as expected.

### How to reproduce it (minimal and precise)

Follow normal upgrade procedure (using update.sh script) from 1.27.2 to 1.28.0

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20220219

### Adapter

Electrolama zzh!

### Debug log

info  2022-10-01 22:17:34: Configuring '0x0015bc001a01e658'
info  2022-10-01 22:17:34: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e658', payload '{"battery":40,"battery_low":false,"illuminance_lux":null,"last_seen":"2022-10-01T22:17:34+01:00","led_control":null,"linkquality":10,"occupancy":true,"occupancy_timeout":null,"tamper":true,"temperature":null,"voltage":2700}'
info  2022-10-01 22:17:35: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e658', payload '{"battery":40,"battery_low":false,"illuminance_lux":null,"last_seen":"2022-10-01T22:17:35+01:00","led_control":null,"linkquality":10,"occupancy":true,"occupancy_timeout":null,"tamper":true,"temperature":null,"voltage":2700}'
info  2022-10-01 22:17:36: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e658', payload '{"battery":40,"battery_low":false,"illuminance_lux":null,"last_seen":"2022-10-01T22:17:36+01:00","led_control":null,"linkquality":18,"occupancy":true,"occupancy_timeout":null,"tamper":true,"temperature":null,"voltage":2700}'
info  2022-10-01 22:17:40: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e658', payload '{"battery":40,"battery_low":false,"illuminance_lux":null,"last_seen":"2022-10-01T22:17:40+01:00","led_control":null,"linkquality":14,"occupancy":true,"occupancy_timeout":null,"tamper":true,"temperature":null,"voltage":2700}'
info  2022-10-01 22:17:43: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e658', payload '{"battery":40,"battery_low":false,"illuminance_lux":null,"last_seen":"2022-10-01T22:17:43+01:00","led_control":null,"linkquality":3,"occupancy":true,"occupancy_timeout":null,"tamper":true,"temperature":null,"voltage":2700}'

Xaldimo, 2022-10-01:
Also when trying to remove and re-add the device shows the same behaviour.

Downgrading to 1.27.2 resolved the issue and all fields are updating as expected.

sjorge, 2022-10-02:
Can you post and actually debug log of when it’s trying to (re-)configure? I just removed and repaired mine a few times and it worked every time

DubhAd, 2022-10-03:
I just upgraded to 1.28.0 and my MOSZB-140 is reporting in fine. I did get a number of errors like this in the log
```
Failed to configure 'family_room_multi', attempt 3 (Error: Read 0x0015bc001a01e59c/35 genBasic(["develcoLedControl"],
```
however the device is working and all sensors are updating.

sjorge, 2022-10-03:
> I just upgraded to 1.28.0 and my MOSZB-140 is reporting in fine. I did get a number of errors like this in the log
> 
> ```
> Failed to configure 'family_room_multi', attempt 3 (Error: Read 0x0015bc001a01e59c/35 genBasic(["develcoLedControl"],
> ```

Was was the error reason? Timeout?

Could be the device is sleeping when z2m is restarted so it hits a timeout when trying to reconfigure. That's not unique to this device though, but something all battery devices sometimes have issues with.

DubhAd, 2022-10-03:
Timeout, yes. TBH I just ignored it. It was working fine before, it's working fine just now, and no other devices are misbehaving.

sjorge, 2022-10-03:
> Timeout, yes. TBH I just ignored it. It was working fine before, it's working fine just now, and no other devices are misbehaving.

z2m will probably try again at some point, what you'll be missing is:
- initial read for led_control, so it will be missing or `null` until you manually read it using a /get or until the configure succeeds
- initial read for occupancy_timeout (same as above)
- reporting for batteryVoltage, so battery might be null until that's fixed, but it was there on the previous release so you're not missing anything that was there before.

DubhAd, 2022-10-03:
Cheers. At some point I'll wake it and reconfigure then.

Xaldimo, 2022-10-03:
OK I have just upgraded back to 1.28.0 and the motion sensor worked for a while, but then it again stopped updating the status. It now shows the livingroom is occupied for last 1 hour, while we even went out for a walk so nobody was in.

Attached is the debug log as requested.
[2022-10-03.19-23-54.log](https://github.com/Koenkk/zigbee2mqtt/files/9700044/2022-10-03.19-23-54.log)

Xaldimo, 2022-10-03:
I can see a difference that when viewing the log running 1.28.0 there is an extra entity called "led_control":null,

sjorge, 2022-10-03:
Yes, `led_control` was recently added, It's null because the (re)configure never seems to be able to complete. It never got to reading that attribute, the reconfigure is hitting a timeout
```
error 2022-10-03 19:25:19: Failed to configure 'motion_livingroom', attempt 1 (Error: Read 0x0015bc001a01e658/35 genBasic(["develcoLedControl"], {"sendWhen":"active","timeout":10000,"disab
leResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4117,"transactionSequenceNumber":null,"writeUnd
iv":false}) failed (Timeout - 37193 - 35 - 7 - 0 - 1 after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:564:17)
    at processTimers (node:internal/timers:507:7))
```

I also see some '/get' requests to try and read it manually (triggered via frontend?) that also timeout.

I think the coordinator has a bunch of requests queued up for the device from the reconfigure that are all stuck and slowly timeing out.

Can you hit the little button inside to keep it awake while it tries to reconfigure? (Or repairing and keeping it awake while it does the configure should work too)

Xaldimo, 2022-10-03:
OK I have now done the following:
1. Remote entity "motion_livingroom"
2. Stop z2m
3. Upgrade to 1.28.0
4. Try to re-pair unit, this took about 10 min and several 'enable join' to be able to get it registered
5. Trying the internal tamper switch shows it is updating
6. After about 2 min trying to use the tamper switch to check updating and it stops. it keeps stating tamper=true while the switch it pushed down.

Also when viewing the data it keeps the illuminance_lux and temperature set to null, while on 1.27.2 they do show values from the beginning.

Attached again the debug log. The unit is now showing as 0x0015bc001a01e658 (not renamed yet)

[2022-10-03.21-06-44.log](https://github.com/Koenkk/zigbee2mqtt/files/9700691/2022-10-03.21-06-44.log)

Xaldimo, 2022-10-03:
When checking the 1.27.2 log I notice that it states " Successfully configured '0x0015bc001a01e658'" while this doesn't show in the 1.28.0 log.

Also it can be seen that it does populate the illuminance and temperature already.

info  2022-10-03 21:30:26: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e658', payload '{"battery":40,"battery_low":false,"illuminance_lux":null,"last_seen":"2022-10-03T21:30:26+01:00","linkquality":69,"occupancy":true,"tamper":false,"temperature":22.3,"voltage":2700}'
info  2022-10-03 21:30:26: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e658', payload '{"battery":40,"battery_low":false,"illuminance_lux":null,"last_seen":"2022-10-03T21:30:26+01:00","linkquality":65,"occupancy":true,"tamper":false,"temperature":22.3,"voltage":2700}'
info  2022-10-03 21:30:26: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e658', payload '{"battery":40,"battery_low":false,"illuminance_lux":null,"last_seen":"2022-10-03T21:30:26+01:00","linkquality":69,"occupancy":true,"tamper":false,"temperature":22.3,"voltage":2700}'
info  2022-10-03 21:30:27: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e658', payload '{"battery":40,"battery_low":false,"illuminance":13551,"illuminance_lux":23,"last_seen":"2022-10-03T21:30:27+01:00","linkquality":69,"occupancy":true,"tamper":false,"temperature":22.3,"voltage":2700}'
info  2022-10-03 21:30:28: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e658', payload '{"battery":40,"battery_low":false,"illuminance":13551,"illuminance_lux":23,"last_seen":"2022-10-03T21:30:28+01:00","linkquality":72,"occupancy":true,"tamper":false,"temperature":22.3,"voltage":2700}'
info  2022-10-03 21:30:28: Successfully configured '0x0015bc001a01e658'
info  2022-10-03 21:30:29: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e658', payload '{"battery":40,"battery_low":false,"illuminance":13518,"illuminance_lux":22,"last_seen":"2022-10-03T21:30:29+01:00","linkquality":69,"occupancy":true,"tamper":false,"temperature":22.3,"voltage":2700}'
info  2022-10-03 21:30:34: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e658', payload '{"battery":40,"battery_low":false,"illuminance":13518,"illuminance_lux":22,"last_seen":"2022-10-03T21:30:34+01:00","linkquality":65,"occupancy":true,"tamper":true,"temperature":22.3,"voltage":2700}'
info  2022-10-03 21:30:40: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a01e658', payload '{"battery":40,"battery_low":false,"illuminance":13393,"illuminance_lux":22,"last_seen":"2022-10-03T21:30:40+01:00","linkquality":65,"occupancy":true,"tamper":true,"temperature":22.3,"voltage":2700}'

sjorge, 2022-10-03:
It indeed did not complete the configuration again. Again there were timeouts.

Were you keeping it awaken with the metal hook (tamper switch) or the tiny brown button next to it?

@Koenkk i’m not really seeing a reason why it would always timeout here, also that should not affect the ssIasZone commands stopping after a while. Do you see something I’m missing?

Xaldimo, 2022-10-03:
Yes I kept using the tamper switch to keep it awake and also to see if it kept showing the entry be updated when pressing and releasing the switch.
As stated it kept doing it for about 2 minutes and then stopped updating the tamper entity (stating 'true' while switch was pressed)

Xaldimo, 2022-10-03:
Not sure if this helps but a difference I noticed also between 1.27.2 and 1.28.0 was that while using 1.27.2 I could easily re-pair the unit within the first join session, within 20 seconds. But using 1.28.0 I had to try it several times and go through at least 4 join sessions (so lasting each at least 4 minutes).

DubhAd, 2022-10-04:
I removed and re-paired the device, to see if I could get the occupancy timeout and LED control working.

Now ... now almost nothing works 😂 
```json
{
    "battery": null,
    "battery_low": false,
    "device": {
        "dateCode": "2021-05-25 20:14",
        "friendlyName": "family_room_multi",
        "ieeeAddr": "0x0015bc001a01e59c",
        "manufacturerID": 4117,
        "manufacturerName": "frient A/S",
        "model": "MOSZB-140",
        "networkAddress": 46444,
        "powerSource": "Battery",
        "type": "EndDevice",
        "zclVersion": 1
    },
    "illuminance_lux": null,
    "last_seen": "2022-10-04T10:00:44.452Z",
    "led_control": null,
    "linkquality": 58,
    "occupancy": true,
    "occupancy_timeout": 20,
    "tamper": true,
    "temperature": null
}
```

It had _appeared_ to pair fine, no timeouts:

```
info  2022-10-04 09:53:05: Device '0x0015bc001a01e59c' joined
...
info  2022-10-04 09:53:05: Starting interview of '0x0015bc001a01e59c'
info  2022-10-04 09:53:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a01e59c","ieee_address":"0x0015bc001a01e59c","status":"started"},"type":"device_interview"}'
info  2022-10-04 09:53:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a01e59c","ieee_address":"0x0015bc001a01e59c"},"type":"device_announce"}'
info  2022-10-04 09:53:44: Successfully interviewed '0x0015bc001a01e59c', device has successfully been paired
info  2022-10-04 09:53:44: Device '0x0015bc001a01e59c' is supported, identified as: Develco Motion sensor (MOSZB-140)
```

Before I'd re-paired it the reporting tab was configured for reporting the temperature and light level (and something else), now there's nothing listed there. All the options there were there are now blank. For instance:
[image]

sjorge, 2022-10-04:
Was there a line about configuring the device, I usually shows up after the joined message.That the step that sets up all the reporting. ~ sjorge

DubhAd, 2022-10-04:
I'm not seeing any mention of "configure" in the logs for it, not even failed. Whereas when I first upgraded I did get the errors.
```
error 2022-10-04 07:32:05: Failed to configure 'family_room_multi', attempt 4 (Error: ConfigureReporting 0x0015bc001a01e59c/35 genPowerCfg([{"attribute":"batteryVoltage","minimumReportInterval":3600,"maximumReportInterval":43200,"reportableChange":100}], {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 36637 - 35 - 80 - 1 - 7 after 10000ms)
```

DubhAd, 2022-10-04:
> Can you try to manual trigger the configure and keep the device awake while it tries (there is a button on the frontend for this).

II did that, it didn't seem to do anything... I'll try again.

> I did do a full reset before repairing (hold the button until the led blinks continuously (It goes 1 blink, 2 blinks, blinking continuously) it takes a bit before it gets there.

That's what I did, every time. And then (as before when I paired it to 1.27) tapped the button every few seconds until it said the interview was complete.

sjorge, 2022-10-04:
> That's what I did, every time. And then (as before when I paired it to 1.27) tapped the button every few seconds until it said the interview was complete.

Keep tapping it a bit longer until you see the start and (hopefully) finish of the configure.

DubhAd, 2022-10-04:
So, _reconfigure_ doesn't do anything. The only possible new thing in the log is
```
log.txt:info  2022-10-04 11:26:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"family_room_multi","ieee_address":"0x0015bc001a01e59c"},"type":"device_announce"}'
```

I'll remove it again and re-re-pair it and see if that changes anything

DubhAd, 2022-10-04:
Ok, there is no attempt to configure within the 2 minutes after interview completes

sjorge, 2022-10-04:
> Ok, there is no attempt to configure within the 2 minutes after interview completes

Very weird, there should be, otherwise there is no reporting setup.

@Koenkk any ideas why it would not trigger a configure step? There used to be a bug with some leftover state a long time ago but AFAIK that got fixed. (At least I've never run into that again)

DubhAd, 2022-10-04:
Yeah, I have a 1.21 install still where if I re-pair an old device it won't show in the UI until I restart.

You're wondering if a manual cleanup of `database.db` may be required to remove all trace of the old device, so that the full interview process starts from nothing?

DubhAd, 2022-10-04:
Ok, it looks like a problem with 1.28.

I did a fresh install of 1.28. Pairing and configuration logs of that are here: https://dpaste.org/XFTX9/raw

It never goes any further, I spent 6 minutes keeping it awake and nothing happened.

I dropped that instance back to 1.27 and hit _reconfigure_, the results were pretty much instant: https://dpaste.org/WY16a/raw

Xaldimo, 2022-10-04:
> I did do a full reset before repairing (hold the button until the led blinks continuously (It goes 1 blink, 2 blinks, blinking continuously) it takes a bit before it gets there.

In my test before I also done the same thing (press until it blinks continuously) and still wouldn't complete the registration.

DubhAd, 2022-10-04:
Debug log joining and (not) configuring with 1.28: https://dpaste.org/oBx1K
Debug log for configuring with 1.27: https://dpaste.org/QwTLP

Interestingly on 1.28 I can manually bind endpoint 35 (battery) and 38 (temperature) to match what 1.27 does, and then manually configure reporting for those and 39 (Illuminance). This gets me back to how it was when I upgraded to 1.28, so that's something.

sjorge, 2022-10-04:
You should also be able to add the 'batteryVoltage' reporting and read (via de frontend) the 'led_control' and 'occupancy_timeout', thats basically what happens in the configure step. So I'm not sure why it's failing.

I can't get mine in that state using zzh or a zzhp-lite, it always succeeds after a few seconds.

sjorge, 2022-10-04:
With the zzh it was close to the coordinator as that is my development network, with the zzhp-lite it was through a Ubisys S1 acting as a router.

DubhAd, 2022-10-04:
I paired it both to my Sonoff-P (main network, firmware 20220726) and a CC2531 (test network, firmware very old), directly for both, and in the case of the Sonoff-P also via a router (the test network has no other devices on it).

> You should also be able to add the 'batteryVoltage' reporting and read (via de frontend) the 'led_control' and 'occupancy_timeout', thats basically what happens in the configure step. So I'm not sure why it's failing.

Yeah, it's reporting the batteryVoltage now. I'll try waking it and reading those values manually later - possibly tomorrow when I won't have family underfoot.

Koenkk, 2022-10-04:
@DubhAd can you provide the herdsman debug log when triggering the configure? After the configure wait 120 seconds.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

DubhAd, 2022-10-04:
> @DubhAd can you provide the herdsman debug log when triggering the configure? After the configure wait 120 seconds.

Yup, I can do that tomorrow on my test setup 👍

X4LD1M0, 2022-10-07:
Any update on this? Is there anything I can provide?

sjorge, 2022-10-07:
Don't really have any ideas left, I cannot reproduce it on release or dev (which has a few more minor optimizations)

watergypsy, 2022-10-09:
I'm having exactly the same issue as DubhAd

Attached are the log files from my system if they are of any use identifying the issue.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9741464/log.txt)
[log-debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/9741465/log-debug.txt)

JimGermany, 2022-10-19:
FYI: Same problem with:

Home Assistant 2022.9.7
Supervisor 2022.10.0
Operating System 9.0
Frontend 20220907.2 - latest

Sonoff Zigbee 3.0 USB Dongle
DEVNAME: /dev/ttyUSB0
DEVPATH: >-
  /devices/pci0000:00/0000:00:1e.0/0000:01:1b.0/usb2/2-2/2-2:1.0/ttyUSB0/tty/ttyUSB0
ID_BUS: usb
ID_MODEL: CP2102N_USB_to_UART_Bridge_Controller
ID_MODEL_ENC: CP2102N\x20USB\x20to\x20UART\x20Bridge\x20Controller
ID_MODEL_ID: ea60
ID_PATH: pci-0000:01:1b.0-usb-0:2:1.0
ID_PATH_TAG: pci-0000_01_1b_0-usb-0_2_1_0
ID_REVISION: '0100'

Motion sensor (MOSZB-140) Develco

After the 1.28.0-1 update, the motion detector is no longer recognized. Version 1.27.2-1 is working.

sjorge, 2022-10-19:
> [log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9741464/log.txt) [log-debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/9741465/log-debug.txt)

Thanks for the logs, it looks like everyone so far that is having problems hitting timeouts while trying to lead de led_control attribute.

Can you try the current dev branch to see if it is better? I've wrapped the reading in a try-catch so it should at least continue after a timeout and setup the temperature and illuminance reporting.

It also now read the develco specific firmware version attribute, I'd be interested in what values it contains.

I wonder the led_control will only work on very new firmwares.

federicosantolini, 2022-10-20:
I was able to manually setup the illuminance configuring the endpoint, but the temperature is still missing. is there a way to install the previous version of z2m?
[image]

X4LD1M0, 2022-10-20:
You can use the following steps to downgrade if required. Please make sure to change the tag number to the version you want.

```
sudo systemctl stop zigbee2mqtt
cd /opt/zigbee2mqtt
cp -R data data-backup
git checkout HEAD -- npm-shrinkwrap.json
git clone https://github.com/Koenkk/zigbee2mqtt
git checkout tags/1.27.2
npm ci
cp -R data-backup/* data
rm -rf data-backup
sudo systemctl start zigbee2mqtt
```

watergypsy, 2022-10-20:
> Thanks for the logs, it looks like everyone so far that is having problems hitting timeouts while trying to lead de led_control attribute.
>
> Can you try the current dev branch to see if it is better? [...]

Hopefully I have done this correctly, these are the log files for the dev branch as I re-associate the sensor. Slightly strangely it was only partially successful on the first attempt, with occupancy and tamper working but no luminance or temp. Then after pushing the tamper switch it automatically reconfigured and this time luminance and temp seem to work too.

I have then reverted to the 1.28 (latest) pull and the configuration seems to keep working with  all the parameters being recorded.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9829164/log.txt)
[log-debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/9829165/log-debug.txt)

Thanks again for your work on this.

X4LD1M0, 2022-10-20:
@watergypsy are the sensors still working after a while? I had it that if I could get it to register, it stopped updating the fields (tamper, occupancy, etc)

watergypsy, 2022-10-20:
Yep, seems to be continuing to work having reverted to 1.28.0, all parameters updating. The occupancy timeout has reverted to 0 (was at 20 on the dev version) but it still works.

sjorge, 2022-10-20:
> Yep, seems to be continuing to work having reverted to 1.28.0, all parameters updating. The occupancy timeout has reverted to 0 (was at 20 on the dev version) but it still works.

Looks like it's just led_control that is causing some issues, I'll try and go over the logs you posted after work. Hopefully it captured the read of the develco firmware number attribute and I can make reading it conditional so the configure should no longer timeout.

watergypsy, 2022-10-20:
Still seems to be reading (1.28.0)

Firmware build date     2021-05-25 19:02
Firmware version    3.6.6

If that saves you trawling the logs!

X4LD1M0, 2022-10-21:
How can I view the firmware version? I checked my debug log but don't see any indication of that value.

X4LD1M0, 2022-10-21:
OK have installed the dev branch and can see mine is also running 3.6.6

sjorge, 2022-10-21:
> OK have installed the dev branch and can see mine is also running 3.6.6

Thanks for the extra data point, very helpful!

X4LD1M0, 2022-10-21:
I now downgraded to 1.28.0 and it seems it again stopped updating its fields. the occupancy remains true, even when there is no motion for several minutes.

```
info  2022-10-21 10:30:51: MQTT publish: topic 'zigbee2mqtt/motion_livingroom', payload '{"battery":40,"battery_low":false,"current_firmware":"3.6.6","illuminance":22474,"illuminance_lux":177,"last_seen":"2022-10-21T10:30:51+01:00","led_control":null,"linkquality":65,"occupancy":true,"occupancy_timeout":20,"tamper":false,"temperature":21.6,"voltage":2700}'
debug 2022-10-21 10:31:21: Received Zigbee message from 'motion_livingroom', type 'read', cluster 'genTime', data '["time"]' from endpoint 35 with groupID 0
info  2022-10-21 10:31:21: MQTT publish: topic 'zigbee2mqtt/motion_livingroom', payload '{"battery":40,"battery_low":false,"current_firmware":"3.6.6","illuminance":22474,"illuminance_lux":177,"last_seen":"2022-10-21T10:31:21+01:00","led_control":null,"linkquality":69,"occupancy":true,"occupancy_timeout":20,"tamper":false,"temperature":21.6,"voltage":2700}'
❯ tail -f 2022-10-21.10-28-37.log | grep motion_livingroom
info  2022-10-21 10:31:28: MQTT publish: topic 'zigbee2mqtt/motion_livingroom', payload '{"battery":40,"battery_low":false,"current_firmware":"3.6.6","illuminance":22474,"illuminance_lux":177,"last_seen":"2022-10-21T10:31:28+01:00","led_control":null,"linkquality":69,"occupancy":true,"occupancy_timeout":20,"tamper":false,"temperature":21.6,"voltage":2700}'

```
I also seen the following error in the log (not set to debug so this is the only line it gave me):

`error 2022-10-21 10:32:56: Failed to configure 'motion_livingroom', attempt 2 (Error: Bind 0x0015bc001a01e658/35 genPowerCfg from '0x00124b001ccea402/1' failed (AREQ - ZDO - bindRsp after 10000ms)`

After upgrading back again to dev branch I can see the occupancy change back to false after approx 20 seconds

```
info  2022-10-21 10:38:38: MQTT publish: topic 'zigbee2mqtt/motion_livingroom', payload '{"battery":40,"battery_low":false,"current_firmware":"3.6.6","illuminance":22474,"illuminance_lux":177,"last_seen":"2022-10-21T10:38:38+01:00","led_control":null,"linkquality":65,"occupancy":true,"occupancy_timeout":20,"tamper":false,"temperature":21.6,"voltage":2700}'
debug 2022-10-21 10:38:59: Received Zigbee message from 'motion_livingroom', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":48}' from endpoint 35 with groupID 0
info  2022-10-21 10:38:59: MQTT publish: topic 'zigbee2mqtt/motion_livingroom', payload '{"battery":40,"battery_low":false,"current_firmware":"3.6.6","illuminance":22474,"illuminance_lux":177,"last_seen":"2022-10-21T10:38:59+01:00","led_control":null,"linkquality":65,"occupancy":false,"occupancy_timeout":20,"tamper":false,"temperature":21.6,"voltage":2700}'

```

sjorge, 2022-10-21:
I'm glad at least the dev branch seems to be working, aside from led_control that is.

I think, since configure() never completes on 1.28.0 it will always auto trigger a re-configure when the device sends a message, which will then timeout again and repeat. On dev with the extra try-catch even with the timeout, the configure succeeds so it won't try again and the device messages get processes properly.

X4LD1M0, 2022-10-21:
Would be nice if there was a way to upgrade those units, but guess will have to stick with what is on it. I guess something in Z2M would need to check which f/w is on it to decide what fields it can retrieve?

sjorge, 2022-10-21:
> Would be nice if there was a way to upgrade those units, but guess will have to stick with what is on it. I guess something in Z2M would need to check which f/w is on it to decide what fields it can retrieve?

Correct, I'm going to try and read those develco firmware fields during interview so they are available in the meta object during the configure() call so I can avoid the attribute if required. Not sure on the finer details yet.

Ofcourse somehow getting an OTA image would be even better! But Frient/Develco have been rather non-responsive to my e-mail(s), there is more in the MOSZB-1xx technical doc that don't work on mine either so I'm guessing in the future newer firmware might show up on devices :(

sjorge, 2022-10-21:
@X4LD1M0  getting closer to a clean solution but I have a follow up question, did the occupancy_timeout read/write work on your older firmware when on the dev branch?

X4LD1M0, 2022-10-21:
I am still on the dev branch since that one works 😃 

I just increased it from 20 to 30, and it seems to work:

```
debug 2022-10-21 19:19:15: Received MQTT message on 'zigbee2mqtt/motion_livingroom/set' with data '{"occupancy_timeout":30}'
debug 2022-10-21 19:19:15: Publishing 'set' 'occupancy_timeout' to 'motion_livingroom'
debug 2022-10-21 19:19:24: Received MQTT message on 'zigbee2mqtt/motion_livingroom/get' with data '{"occupancy_timeout":""}'
debug 2022-10-21 19:19:24: Publishing get 'get' 'occupancy_timeout' to 'motion_livingroom'
debug 2022-10-21 19:20:09: Received Zigbee message from 'motion_livingroom', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":49}' from endpoint 35 with groupID 0
info  2022-10-21 19:20:09: MQTT publish: topic 'zigbee2mqtt/motion_livingroom', payload '{"battery":40,"battery_low":false,"current_firmware":"3.6.6","illuminance":7820,"illuminance_lux":6,"last_seen":"2022-10-21T19:20:09+01:00","led_control":null,"linkquality":69,"occupancy":true,"occupancy_timeout":20,"tamper":false,"temperature":20.9,"voltage":2700}'
info  2022-10-21 19:20:10: MQTT publish: topic 'zigbee2mqtt/motion_livingroom', payload '{"battery":40,"battery_low":false,"current_firmware":"3.6.6","illuminance":7820,"illuminance_lux":6,"last_seen":"2022-10-21T19:20:10+01:00","led_control":null,"linkquality":69,"occupancy":true,"occupancy_timeout":20,"tamper":false,"temperature":20.9,"voltage":2700}'
info  2022-10-21 19:20:10: MQTT publish: topic 'zigbee2mqtt/motion_livingroom', payload '{"battery":40,"battery_low":false,"current_firmware":"3.6.6","illuminance":7820,"illuminance_lux":6,"last_seen":"2022-10-21T19:20:10+01:00","led_control":null,"linkquality":69,"occupancy":true,"occupancy_timeout":30,"tamper":false,"temperature":20.9,"voltage":2700}'

```

sjorge, 2022-10-21:
@X4LD1M0 great! I'll change the minimal required version major firmware version for that attribute to 3 then :)

sjorge, 2022-10-21:
I've made a PR against zhc that should fix these issues 🤞

z2m should now read the develco specific sw version and skip certain features based on the major version.
It could be that occupany_timeout and/or led_control don't show up after this gets merged, but the (re-)configure should no longer fail now. If you leave/join the device again they should show up depending on the fw version.

X4LD1M0, 2022-10-22:
@sjorge I have just pulled the latest dev branch. I am again in the state where it doesn't seem to complete its interview. It showing some values, but for example the tamper is true, while I have closed the casing.

device is: 0x0015bc001a01e658

Below is the debug log:
[2022-10-22.21-45-20.log](https://github.com/Koenkk/zigbee2mqtt/files/9845251/2022-10-22.21-45-20.log)

sjorge, 2022-10-22:
Very interesting! It does not look like the read for the `develcoPrimarySwVersion` and `develcoPrimaryHwVersion` attributes was triggered at all!
That is why you're no seeing either led_control or occupancy_timeout :(

Based on the logs I did see the configure step correct configure illuminance, temperature, and battery reporting. (ssIasZone enrollment is during interview and not as part of configuration.

I also noticed that the device indeed reported the tamper bit set to true for a very long time until this entry:
```debug 2022-10-22 21:58:05: Received Zigbee message from '0x0015bc001a01e658', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":48}' from endpoint 35 with groupID 0```

So progress, but I'm puzzled why it's not triggeting the read of `develcoPrimarySwVersion` and `develcoPrimaryHwVersion` attributes.

cc @Koenkk any idea why it would not trigger the read in the onEvent function for this device after my PR from yesterday?
It triggers it for mine as you can see here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4812#issue-1418730541
```
Zigbee2MQTT:debug 2022-10-21 20:44:20: Received Zigbee message from '0x0015bc001a0238ec', type 'readResponse', cluster 'genBasic', data '{"develcoPrimaryHwVersion":{"data":[2,2,0],"type":"Buffer"},"develcoPrimarySwVersion":{"data":[4,0,3],"type":"Buffer"}}' from endpoint 35 with groupID 0
```

Admittedly there could theoretical be a race were configure happens before this is read, but I would still expect to see the read in the log somewhere but it's absent completely. I wonder if it might be better to read these during the device interview itself? But I did not see a way to do this  from onEvent.

Koenkk, 2022-10-23:
It could also be that the read fails, so you won't get the read response. For testing purposes it might be better to create an external converter such that people can test it without making changes to zigbee-herdsman-converters. You can also add `console.log()` statements in there.

sjorge, 2022-10-23:
Here is an external convertor to iterate more quickly:
```node
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const constants = require('zigbee-herdsman-converters/lib/constants');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

// develco specific cosntants
const manufacturerOptions = {manufacturerCode: 0x1015};

/* MOSZB-1xx - ledControl - bitmap8 - r/w
 * 0x00 Disable LED when movement is detected.
 * 0x01 Enables periodic fault flashes. These flashes are used to indicate e.g. low battery level.
 * 0x02 Enables green application defined LED. This is e.g. used to indicate motion detection.
 * Default value 0xFF ( seems to be fault + motion)
 */
const develcoLedControlMap = {
    0x00: 'off',
    0x01: 'fault_only',
    0x02: 'motion_only',
    0xFF: 'both',
};

const fwOnEvent = async (type, data, device, options, state) => {
    /**
     * Develco devices have a manufacturer specific field on genBasic
     *   for sw and hw versions.
     *
     * We read those during deviceInterview
     *   so that this information is usable during configure() calls
     *   to skip some features on older devices that might cause
     *   timeouts and other issues.
     */

    if (type === 'deviceInterview') { // WARN: also reading on deviceAnnounce hits a timeout!
        console.log(`MOSZB-140 onEvent: triggerd by ${type} ...`);
        try {
            const data = await device.endpoints[0].read('genBasic', ['develcoPrimarySwVersion', 'develcoPrimaryHwVersion'],
                manufacturerOptions);

            console.log("MOSZB-140 onEvent: read returned ...");
            if (data.hasOwnProperty('develcoPrimarySwVersion')) {
                device.softwareBuildID = data.develcoPrimarySwVersion.join('.');
            }

            if (data.hasOwnProperty('develcoPrimaryHwVersion')) {
                device.hardwareVersion = data.develcoPrimaryHwVersion.join('.');
            }
        } catch (error) {
            console.log("MOSZB-140 onEvent: we are in the catch ???");
        }
    } else {
        console.log(`MOSZB-140 onEvent: ignoring type ${type} ...`);
    }
};

// develco specific convertors
const develco = {
    fz: {
        led_control: {
            cluster: 'genBasic',
            type: ['attributeReport', 'readResponse'],
            options: [],
            convert: (model, msg, publish, options, meta) => {
[... 227 lines cut ...]
Zigbee2MQTT:info  2022-10-23 11:06:55: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a0238ec', payload '{"illuminance":18135,"illuminance_lux":65,"linkquality":196,"temperature":24.5}'
Zigbee2MQTT:debug 2022-10-23 11:06:56: Received Zigbee message from '0x0015bc001a0238ec', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryAlarmState":0,"batteryVoltage":29}' from endpoint 35 with groupID 0
MOSZB-140 onEvent: ignoring type message ...
Zigbee2MQTT:info  2022-10-23 11:06:56: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a0238ec', payload '{"battery":80,"battery_low":false,"illuminance":18135,"illuminance_lux":65,"linkquality":193,"temperature":24.5,"voltage":2900}'
Zigbee2MQTT:debug 2022-10-23 11:06:58: Received Zigbee message from '0x0015bc001a0238ec', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryAlarmState":0,"batteryVoltage":30}' from endpoint 35 with groupID 0
MOSZB-140 onEvent: ignoring type message ...
Zigbee2MQTT:info  2022-10-23 11:06:58: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a0238ec', payload '{"battery":100,"battery_low":false,"illuminance":18135,"illuminance_lux":65,"linkquality":196,"temperature":24.5,"voltage":3000}'
Zigbee2MQTT:debug 2022-10-23 11:06:58: Received Zigbee message from '0x0015bc001a0238ec', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 35 with groupID 0
MOSZB-140 onEvent: ignoring type message ...
Zigbee2MQTT:info  2022-10-23 11:06:58: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a0238ec', payload '{"battery":100,"battery_low":false,"illuminance":18135,"illuminance_lux":65,"linkquality":193,"temperature":24.5,"voltage":3000}'
MOSZB-140 configure: checking softwareBuildID (4.0.3) ...
MOSZB-140 configure: reading develcoAlarmOffDelay ...
Zigbee2MQTT:debug 2022-10-23 11:06:58: Received Zigbee message from '0x0015bc001a0238ec', type 'readResponse', cluster 'ssIasZone', data '{"develcoAlarmOffDelay":20}' from endpoint 35 with groupID 0
MOSZB-140 onEvent: ignoring type message ...
Zigbee2MQTT:info  2022-10-23 11:06:58: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a0238ec', payload '{"battery":100,"battery_low":false,"illuminance":18135,"illuminance_lux":65,"linkquality":193,"occupancy_timeout":20,"temperature":24.5,"voltage":3000}'
MOSZB-140 configure: reading develcoLedControl ...
Zigbee2MQTT:debug 2022-10-23 11:06:58: Received Zigbee message from '0x0015bc001a0238ec', type 'readResponse', cluster 'genBasic', data '{"develcoLedControl":255}' from endpoint 35 with groupID 0
MOSZB-140 onEvent: ignoring type message ...
[... cut ...] :20,"temperature":24.5,"voltage":3000}'
Zigbee2MQTT:info  2022-10-23 11:06:58: Successfully configured '0x0015bc001a0238ec'
```

@Koenkk I've seen two caseses while testing where the deviceInterview event type was triggered before z2m even read the devices model! How would this be possible? It was after the device left the network and I rejoined it without restarting z2m.

Is there some state not being claired? We had similar issue before. But I can't reproduce it on demand which is frustrating.

sjorge, 2022-10-23:
@Koenkk bad news, I can trigger the device info sticks around after leave more or less successfully now.

```
Zigbee2MQTT:warn  2022-10-23 15:23:00: Device '0x0015bc001a0238ec' left the network
Zigbee2MQTT:info  2022-10-23 15:23:00: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a0238ec","ieee_address":"0x0015bc001a0238ec"},"type":"device_leave"}'
Zigbee2MQTT:info  2022-10-23 15:23:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x0015bc001a0238ec"},"type":"device_removed"}'
Zigbee2MQTT:debug 2022-10-23 15:26:09: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"dlgtq-1","value":true}'
Zigbee2MQTT:info  2022-10-23 15:26:09: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-10-23 15:26:10: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"dlgtq-1"}'
Zigbee2MQTT:debug 2022-10-23 15:26:12: Saving state to file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-10-23 15:26:14: Device '0x0015bc001a0238ec' joined
Zigbee2MQTT:info  2022-10-23 15:26:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a0238ec","ieee_address":"0x0015bc001a0238ec"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-10-23 15:26:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0015bc001a0238ec"},"type":"device_connected"}'
Zigbee2MQTT:info  2022-10-23 15:26:14: Starting interview of '0x0015bc001a0238ec'
Zigbee2MQTT:info  2022-10-23 15:26:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a0238ec","ieee_address":"0x0015bc001a0238ec","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-10-23 15:26:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0015bc001a0238ec"},"type":"pairing"}'
Zigbee2MQTT:debug 2022-10-23 15:26:14: Device '0x0015bc001a0238ec' announced itself
Zigbee2MQTT:info  2022-10-23 15:26:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a0238ec","ieee_address":"0x0015bc001a0238ec"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-10-23 15:26:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x0015bc001a0238ec"},"type":"device_announced"}'
Zigbee2MQTT:debug 2022-10-23 15:26:16: Retrieving state of '0x0015bc001a0238ec' after reconnect
Zigbee2MQTT:error 2022-10-23 15:26:24: Failed to interview '0x0015bc001a0238ec', device has not successfully been paired
Zigbee2MQTT:info  2022-10-23 15:26:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc001a0238ec","ieee_address":"0x0015bc001a0238ec","status":"failed"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-10-23 15:26:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x0015bc001a0238ec"},"type":"pairing"}'
Zigbee2MQTT:debug 2022-10-23 15:26:25: Received Zigbee message from '0x0015bc001a0238ec', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":53}' from endpoint 35 with groupID 0
Zigbee2MQTT:info  2022-10-23 15:26:25: MQTT publish: topic 'zigbee2mqtt/0x0015bc001a0238ec', payload '{"battery_low":false,"linkquality":193,"occupancy":true,"tamper":true}'
Zigbee2MQTT:debug 2022-10-23 15:26:26: Received Zigbee message from '0x0015bc001a0238ec', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":262147,"imageType":386,"manufacturerCode":4117}' from endpoint 35 with groupID 0
Zigbee2MQTT:debug 2022-10-23 15:26:26: Device '0x0015bc001a0238ec' requested OTA
Zigbee2MQTT:debug 2022-10-23 15:26:36: Received Zigbee message from '0x0015bc001a0238ec', type 'read', cluster 'genTime', data '["time"]' from endpoint 35 with groupID 0
````

So far I've verified the following:
- database.json does not contain the device anymore
- frontend also does not know about the device anymore.
- configuration.yaml still contains the following entry:
```
devices:
  '0x0015bc001a0238ec':
    friendly_name: '0x0015bc001a0238ec'
```

The entry in configuration.yaml remains after stopping z2m. It does not seem to impact the join after starting z2m again though.

I'm 90% sure some state sticks around now. Although that is unrelated to the issues still remaining with the device, it does muddy the data a bit.

[36 more replies]

closed 2022-11-02 by X4LD1M0