Koenkk/zigbee2mqtt #14280: Aqara QBKG12LM and QBKG11LM after return wrong power source (battery) fix still end device on network map

mikolka-lsh, 2022-10-02:
### What happened?

After solving issue #14253 in last update 1.28.0 - in network map -I still see this two switches not as a router but as end devices...
You could see it here on screenshot in left side:
[image]

### What did you expect to happen?

Wall switches must be a routers in network

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20210813

### Adapter

zStack3x0

Koenkk, 2022-10-03:
Can you provide the herdsman debug log when pairing the device? I requested this in https://github.com/Koenkk/zigbee2mqtt/issues/14253#issuecomment-1264704617 but the herdsman debug log is not included.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

mikolka-lsh, 2022-10-03:
[log.zip](https://github.com/Koenkk/zigbee2mqtt/files/9700635/log.zip)
Here it is.
What I done:

1. Add zigbee_herdsman_debug: true to HA addon
2. Restart addon
3. Delete guest.room-switch from Z2M UI
4. Add this device again
5. Open SSH and grub all logs from Z2M container

And now this switch still as a end device on map of my network...

Koenkk, 2022-10-04:
Lets continue in https://github.com/Koenkk/zigbee2mqtt/issues/14253

closed 2022-10-04 by Koenkk