Koenkk/zigbee2mqtt #14284: BSD29 smart plug fails to bind

paddyobrien, 2022-10-02:
### What happened?

Attempting to bind a new smartplug fails. This plug reports as a Tuya Smartplug but is actually one of [these](https://www.aliexpress.com/item/1005003212619366.html)

[image]

It looks like this may be related/similar to https://github.com/Koenkk/zigbee2mqtt/issues/12760 which was fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/4437

### Zigbee2MQTT version

1.28.0 commit: 03ba647

### Adapter firmware version

6.10.0.0 build 169

### Adapter

Popp ZB-Stick

### Debug log

```
info  2022-10-02 15:09:09: Device '0x70b3d52b6000f337' is supported, identified as: TuYa Smart plug (with power monitoring) (TS011F_plug_1)
info  2022-10-02 15:09:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart plug (with power monitoring)","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Instantaneous measured electrical current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":7,"description":"Recover state after power outage","name":"power_outage_memory","property":"power_outage_memory","type":"enum","values":["on","off","restore"]},{"access":7,"description":"Plug LED indicator mode","name":"indicator_mode","property":"indicator_mode","type":"enum","values":["off","off/on","on/off","on"]},{"features":[{"access":3,"description":"Enables/disables physical input on the device","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"type":"lock"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS011F_plug_1","options":[{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Calibrates the current value (percentual offset), takes into effect on next report of device.","name":"current_calibration","property":"current_calibration","type":"numeric"},{"access":2,"description":"Calibrates the voltage value (percentual offset), takes into effect on next report of device.","name":"voltage_calibration","property":"voltage_calibration","type":"numeric"}],"supports_ota":true,"vendor":"TuYa"},"friendly_name":"0x70b3d52b6000f337","ieee_address":"0x70b3d52b6000f337","status":"successful","supported":true},"type":"device_interview"}'
info  2022-10-02 15:09:10: Configuring '0x70b3d52b6000f337'
info  2022-10-02 15:09:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Smart plug (with power monitoring)","friendly_name":"0x70b3d52b6000f337","model":"TS011F_plug_1","supported":true,"vendor":"TuYa"},"type":"pairing"}'
error 2022-10-02 15:09:10: Failed to configure '0x70b3d52b6000f337', attempt 3 (Error: Read 0x70b3d52b6000f337/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/tuya.js:2155:13)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
warn  2022-10-02 15:09:10: Device '0x70b3d52b6000f337' left the network
```

paddyobrien, 2022-10-02:
I tried adding the `try/catch` from the PR linked above, I no longer get the failed to configure error but the device still does not work correctly. Seems to join and then immediately leave the network now:

```
info  2022-10-02 16:16:48: Device '0x70b3d52b6000f337' is supported, identified as: TuYa Smart plug (with power monitoring by polling) (TS011F_plug_3)
info  2022-10-02 16:16:48: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart plug (with power monitoring by polling)","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Instantaneous measured electrical current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":7,"description":"Recover state after power outage","name":"power_outage_memory","property":"power_outage_memory","type":"enum","values":["on","off","restore"]},{"access":7,"description":"Plug LED indicator mode","name":"indicator_mode","property":"indicator_mode","type":"enum","values":["off","off/on","on/off","on"]},{"features":[{"access":3,"description":"Enables/disables physical input on the device","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"type":"lock"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS011F_plug_3","options":[{"access":2,"description":"This device does not support reporting electric measurements so it is polled instead. The default poll interval is 60 seconds, set to -1 to disable.","name":"measurement_poll_interval","property":"measurement_poll_interval","type":"numeric","value_min":-1},{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Calibrates the current value (percentual offset), takes into effect on next report of device.","name":"current_calibration","property":"current_calibration","type":"numeric"},{"access":2,"description":"Calibrates the voltage value (percentual offset), takes into effect on next report of device.","name":"voltage_calibration","property":"voltage_calibration","type":"numeric"}],"supports_ota":true,"vendor":"TuYa"},"friendly_name":"0x70b3d52b6000f337","ieee_address":"0x70b3d52b6000f337","status":"successful","supported":true},"type":"device_interview"}'
info  2022-10-02 16:16:48: Configuring '0x70b3d52b6000f337'
info  2022-10-02 16:16:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Smart plug (with power monitoring by polling)","friendly_name":"0x70b3d52b6000f337","model":"TS011F_plug_3","supported":true,"vendor":"TuYa"},"type":"pairing"}'
info  2022-10-02 16:16:49: Successfully configured '0x70b3d52b6000f337'
warn  2022-10-02 16:16:49: Device '0x70b3d52b6000f337' left the network
info  2022-10-02 16:16:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b6000f337","ieee_address":"0x70b3d52b6000f337"},"type":"device_leave"}'
info  2022-10-02 16:16:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x70b3d52b6000f337"},"type":"device_removed"}'
warn  2022-10-02 16:16:49: Device '0x70b3d52b6000f337' left the network
info  2022-10-02 16:16:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b6000f337","ieee_address":"0x70b3d52b6000f337"},"type":"device_leave"}'
info  2022-10-02 16:16:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x70b3d52b6000f337"},"type":"device_removed"}'
warn  2022-10-02 16:16:49: Device '0x70b3d52b6000f337' left the network
info  2022-10-02 16:16:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b6000f337","ieee_address":"0x70b3d52b6000f337"},"type":"device_leave"}'
info  2022-10-02 16:16:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x70b3d52b6000f337"},"type":"device_removed"}'
warn  2022-10-02 16:16:49: Device '0x70b3d52b6000f337' left the network
info  2022-10-02 16:16:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b6000f337","ieee_address":"0x70b3d52b6000f337"},"type":"device_leave"}'
info  2022-10-02 16:16:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x70b3d52b6000f337"},"type":"device_removed"}'
warn  2022-10-02 16:16:49: Device '0x70b3d52b6000f337' left the network
info  2022-10-02 16:16:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x70b3d52b6000f337","ieee_address":"0x70b3d52b6000f337"},"type":"device_leave"}'
info  2022-10-02 16:16:49: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x70b3d52b6000f337"},"type":"device_removed"}'
warn  2022-10-02 16:16:49: Device '0x70b3d52b6000f337' left the network
```

paddyobrien, 2022-10-02:
Weirdly it looks like it has changed from reporting as `TS011F_plug_1` to `TS011F_plug_3`

paddyobrien, 2022-10-02:
Herdsman logs here:

[herdsman.log](https://github.com/Koenkk/zigbee2mqtt/files/9692862/herdsman.log)

atv2016, 2023-05-07:
I still get this with the BSD29, would love for it to be fixed.