Koenkk/zigbee2mqtt #14285: Dresden ElectroniK FLS-PP (Mega23M12) unable to OTA update

HA-TB303, 2022-10-02:
### What happened?

When trying to OTA update the FLS-PP (Mega23M12) I get ""timeout device did not request any image blocks"

Brand: Dresden Electronik 
Type: FLS-PP (in zigbee2mqtt paired as 'Mega23M12' 
Current firmware version on the device: 020E.201000A0 (build date: 20151008)

Zigbee2mqtt version: [1.28.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.0)
Coordinator: Sonoff zigbee 3.0 usb dongle plus
Coordinator firmware: 20220928 (the previous release version did not work either)

### What did you expect to happen?

The OTA update to complete successfully

### How to reproduce it (minimal and precise)

Try to OTA update the FLS-PP from 020E.201000A0 to latest.

### Zigbee2MQTT version

1.28.0

### Adapter firmware version

20220928

### Adapter

Sonoff zigbee 3.0 usb dongle plus

### Debug log

Debug 2022-10-02 16:04:01Updating to latest '0x00212effff00a372' (FLS-PP3)
Debug 2022-10-02 16:04:01getNewImage for '0x00212effff00a372', meta {"fileVersion":537919733,"fileSize":202911,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/DresdenElektronik/1135-0000-201000F5-FLS-PP3_RGBW_16Mhz.zigbee","sha512":"5aa0a6156c8fe672b0fdfcc6e97bb21244cc5497ca86b75c2d371db85c0712334f88886a0903723f9230c21c6f3ee3a19a0e6ebdcdfc2ac5390fd4c8830be402"}
Debug 2022-10-02 16:04:01OTA update checksum validation succeeded for '0x00212effff00a372'
Debug 2022-10-02 16:04:01getNewImage for '0x00212effff00a372', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4405,"imageType":0,"fileVersion":537919733,"zigbeeStackVersion":2,"otaHeaderString":"�w}6@\u0000`>@\u0000\u0013p@\u0000\u0001\u0000\u0000\u0000�6@\u0000�\u0015@\u0000 �@\u0000��","totalImageSize":202911}
Debug 2022-10-02 16:04:01Got new image for '0x00212effff00a372'
Info 2022-10-02 16:04:05MQTT publish: topic 'zigbee2mqtt/Led controller huisbed tim', payload '{"brightness":203,"brightness_rgb":115,"brightness_white":229,"color":{"h":25,"hue":25,"s":95,"saturation":95,"x":0.5267,"y":0.4133},"color_mode":"color_temp","color_mode_rgb":"color_temp","color_rgb":{"hue":25,"saturation":95,"x":0.5267,"y":0.4133},"color_temp":500,"color_temp_rgb":500,"device":{"applicationVersion":2,"dateCode":"20151008\u0000","friendlyName":"Led controller huisbed tim","hardwareVersion":1,"ieeeAddr":"0x00212effff00a372","manufacturerID":4405,"manufacturerName":"dresden elektronik\u0000","model":"Mega23M12","networkAddress":5424,"powerSource":"Mains (single phase)","softwareBuildID":"020E.201000A0\u0000","stackVersion":3,"type":"Router","zclVersion":1},"linkquality":94,"state":"OFF","state_rgb":"OFF","state_white":"OFF","update":{"progress":0.03,"remaining":11171,"state":"updating"},"update_available":false}'
Info 2022-10-02 16:06:49MQTT publish: topic 'zigbee2mqtt/Led controller huisbed tim', payload '{"brightness":203,"brightness_rgb":115,"brightness_white":229,"color":{"h":25,"hue":25,"s":95,"saturation":95,"x":0.5267,"y":0.4133},"color_mode":"color_temp","color_mode_rgb":"color_temp","color_rgb":{"hue":25,"saturation":95,"x":0.5267,"y":0.4133},"color_temp":500,"color_temp_rgb":500,"device":{"applicationVersion":2,"dateCode":"20151008\u0000","friendlyName":"Led controller huisbed tim","hardwareVersion":1,"ieeeAddr":"0x00212effff00a372","manufacturerID":4405,"manufacturerName":"dresden elektronik\u0000","model":"Mega23M12","networkAddress":5424,"powerSource":"Mains (single phase)","softwareBuildID":"020E.201000A0\u0000","stackVersion":3,"type":"Router","zclVersion":1},"linkquality":94,"state":"OFF","state_rgb":"OFF","state_white":"OFF","update":{"state":"available"},"update_available":true}'
Debug 2022-10-02 16:06:49Update of 'Led controller huisbed tim' failed (Error: Timeout: device did not request any image blocks)
Info 2022-10-02 16:06:49MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Led controller huisbed tim' failed (Timeout: device did not request any image blocks)","meta":{"device":"Led controller huisbed tim","status":"update_failed"},"type":"ota_update"}'

HA-TB303, 2022-11-02:
This issue still persists.

djoeney, 2022-11-04:
I have the same issue here is some information.

Dresden Device:
-Firmware build date: 20151008
-Firmware version: 020E.201000A0

Zigbee2mqtt: 
-Version 1.28.1-dev commit: [9540b6c](https://github.com/Koenkk/zigbee2mqtt/commit/9540b6c)

Adapter
Sonoff zigbee 3.0 usb dongle plus
Coordinator type:
- zStack3x0
Coordinator revision
- 20211217

pschaumburg, 2022-12-03:
Same here.
With the exact same builds and versions.

CoolID2, 2022-12-08:
I have exact the same issue on my three Devices here is some information.
The smallest distance between the coordinator and a device is 3m

2x Dresden Device:  
- Firmware build date: 20151008
- Firmware version: 020C.201000A0

1x Dresden Device:  
- Firmware build date: 20180517
- Firmware version: 0214.201000EB

Zigbee2mqtt:
- Version 1.28.4 commit: [52e545f](https://github.com/Koenkk/zigbee2mqtt/commit/52e545f)

Adapter:

- Coordinator type:      ConBee2/RaspBee2

- Coordinator version: 0x26780700     I also tried 0x26650700

KidA001, 2023-01-07:
Bump on this issue. I have two of these devices:
Model: `Dresden Elektronik Mega23M12`
Current Firmware Version: `0214.201000EB`
Zigbee Model: `FLS-PP3`
[Link to device](https://www.zigbee2mqtt.io/devices/Mega23M12.html#dresden%2520elektronik-mega23m12)

Currently Running Z2M: [1.29.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.29.0) commit: [dddce57](https://github.com/Koenkk/zigbee2mqtt/commit/dddce57)
Coordinator type: `zStack3x0`
Coordinator revision: `20220219`

I get the same `timeout device did not request any image blocks` error when trying to update. Any help appreciated.

KidA001, 2023-01-19:
@Koenkk any chance this could be looked at?

johnzairis, 2023-02-20:
Exactly the same issue here. 

> Info 2023-02-20 10:11:42Updating 'Strip' to latest firmware
Info 2023-02-20 10:11:42MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'Strip' to latest firmware","meta":{"device":"Strip","status":"update_in_progress"},"type":"ota_update"}'
Info 2023-02-20 10:11:46Update of 'Strip' at 0.03%, ≈ 177 minutes remaining
Info 2023-02-20 10:14:31MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Strip' failed (Timeout: device did not request any image blocks)","meta":{"device":"Strip","status":"update_failed"},"type":"ota_update"}'
Error 2023-02-20 10:14:31Update of 'Strip' failed (Timeout: device did not request any image blocks)

KidA001, 2023-03-24:
Bump to keep-alive

closed 2023-05-27 by github-actions[bot]

ThxGiving, 2024-01-22:
Same here. Any news? Maybe the firmware is not the right one?

Mangar2005, 2024-02-04:
same here for 3 of these devices.

Snort007, 2024-02-28:
Same here. I have 7 Dresden Elektronik Mega23M12.
Firmware date: 20151008
Firmware version: 020E.201000A0
2024-02-28 08:00:09Update of '0x00211effef01c84b' failed (OTA: Timeout, device did not request any image blocks)
Zigbee2mqtt Homeassistant Addon 1.35.3-1
I also tried the from vendor recommended power cycle to speed up the OTA,
because the devices only requesting firmware updates in regular times!?
https://phoscon.de/de/support#update-of-DE-devices (devices FAQ)
But maybe the Zigbee2mqtt timeout makes trouble here!?

stormmurdoc, 2024-06-23:
Same here. Device Firmware: 

- 0214.201000EB
- 020E.201000A0

xagos, 2024-09-07:
I have five of theses devices. Same here - update is not possible due to given reasons:

Actual firmware: 0214.201000EB

ldti, 2024-12-23:
Same issues here..

onlynow2, 2024-12-27:
Same issue for me..

beachbatsch, 2024-12-31:
Same here:
- Firmware 0214.201000EB
- Coordinator: slzb-06m
- Plattform:  home assistant
- Zigbee2Mqtt

Solveta, 2025-05-17:
Same problem here. Did anybody asked the company about it?

wigledav, 2025-05-26:
I have the same problem. After some chatGPT research and a lots of trial and error I managed to persuade my FLS to slowly update OTA. The key seemed to be actually powering the device ON at the same time as starting the OTA process and also having the Open Join enabled in zigbee2mqtt.
 
I had also added the following to my zigbee2mqtt configuration.yaml file but it may be irrelevant:

ota:
  # local files are automatically picked up from /config/zigbee2mqtt/firmware/
  # here's how to add a *remote* image so it shows up in the UI dropdown:
  remote:
    - fileName: "1135-0000-201000F5-FLS-PP3_RGBW_16Mhz.zigbee"
      fileVersion: 537919733
      imageType: 0
      manufacturerCode: 4405
      url: "https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/images/DresdenElektronik/1135-0000-201000F5-FLS-PP3_RGBW_16Mhz.zigbee"
      sha512: "5aa0a6156c8fe672b0fdfcc6e97bb21244cc5497ca86b75c2d371db85c0712334f88886a0903723f9230c21c6f3ee3a19a0e6ebdcdfc2ac5390fd4c8830be402"

  # slow the transfer if your mesh is flaky (optional)
  image_block_response_delay: 500
  default_maximum_data_size: 30

My device is not 6% through its firmware upgrade and has about an hour to run.

This was the GPT comment that caught my eye: - "air-gapping"

Large images can take hours if they have to hop multiple routers. Users have found that bringing the device right next to the coordinator (or “**air-gapping**” it by powering it off then on immediately after starting OTA) can dramatically improve reliability—and doing one update at a time avoids saturating the network .

Now at 11%. Will let you know :)

JiroMusik, 2025-12-09:
It is still the same behaviour - is there any more tips on how to get the timing right? It shows starting update and 89 minutes and thats it - fails after 2 minutes