Koenkk/zigbee2mqtt #14287: [Feature request]: Add Power-on behavior for HEIMAN HS2WDS

olivr2s, 2022-10-02:
### Is your feature request related to a problem? Please describe

The HEIMAN HS2WDS bulb light support the `Power-on behavior`.
I tried with MQTT-explorer and it works perfectly.
Is it possible to add this parameter in the template.

### Describe the solution you'd like

Add `Power-on behavior`.

### Describe alternatives you've considered

-

### Additional context

-

olivr2s, 2022-11-02:
Do I have to provide other information to add the function?

Koenkk, 2022-11-02:
Added this now! (for all lights supporting it).

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-02 by Koenkk