Koenkk/zigbee2mqtt #14288: S40ZBTPB fails interview in master / no sign of life in latest-dev (1.27.2-dev commit: 3453585) with Z-DongleE

y8s, 2022-10-02:
### What happened?

On first pairing the S40ZBTPB seemed to work but after removing and re-pairing, interview fails and the device shows up as "unsupported".

Potentially similar to this issue, but the dev didn't fix it for me:

https://github.com/Koenkk/zigbee2mqtt/issues/14135

### What did you expect to happen?

Pair a second time as it did the first and show up in device list as the actual supported device.

Note that I have a second S40ZBTPB currently working and I'm not about to un-pair it.

### How to reproduce it (minimal and precise)

Pair the S40ZBTPB smart plug -- interview completed.

wait a while and re-start the pairing process--device leaves the network then attempts to rejoin -- interview fails.

load up the latest-dev
delete the device (force remove)
plug it in, start the pairing process -- nothing happens, no log entries, no green pop up message about interviewing.

### Zigbee2MQTT version

1.27.2 commit: https://github.com/Koenkk/zigbee2mqtt/commit/7dc48fb6d657f2560b42b93bc47c569620b3582f

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Z-DongleE (ezsp v8)

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9692981/log.txt)

kirovilya, 2022-10-02:
update to 1.28.0 and try again

y8s, 2022-10-02:
> update to 1.28.0 and try again

1.28.0 did the trick. I guess I hadn't pulled the latest docker image last night.

Thank you!

closed 2022-10-03 by Koenkk