Koenkk/zigbee2mqtt #14292: [New device support]: Tuya EU 1 Gang Zigbee Switch - Model no. SS800ZB-EU

BDV77, 2022-10-02:
### Link

https://nl.aliexpress.com/item/1005003400336642.html?spm=a2g0o.order_list.0.0.2f1879d210N7Nq&gatewayAdapt=glo2nld

### Database entry

Device '0xa4c13868c1e576c3' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_aim0ztek' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### Comments

Just a simple wall mounted switch, also bought a 2-gang and this gave no issues

Koenkk, 2022-10-03:
`_TZ3000_aim0ztek` is supported in z2m 1.28.0

BDV77, 2022-11-11:
Hi Koen, 

Thanks (again) for delivering support!!
After upgrading to 1.28.2 the device is recoginzed as a plug rather then a switch, on first glance it seems to function properly but it would be nice if it would be corrected to the correct device type.

Thanks in advance