Koenkk/zigbee2mqtt #14311: FireAngel W2-Module

r-win, 2022-10-03:
### What happened?

As earlier noted in issue #9270, this device is only reporting link quality. Battery percentage and carbon_monoxide properties are both Null. 

The current state of the device is as follows:

```json
{
    "battery": null,
    "battery_low": false,
    "last_seen": "2022-10-03T15:42:43.403Z",
    "linkquality": 83,
    "voltage": 3000,
    "carbon_monoxide": null
}
```

Using the dev console, I can request properties batteryVoltage, batteryRatedVoltage and batteryVoltMinThres, which results in:

{"batteryVoltage":30,"batteryRatedVoltage":30,"batteryVoltMinThres":19}

Earlier this afternoon the batteryRatedVoltage value was 31.

I think the existing converter (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/fireangel.js) is missing a convert function, where the voltage will be converted to a battery value.

I also think the carbon_monoxide value should be converted from NULL to OFF

### What did you expect to happen?

I expected to see a battery % and a correct state.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

slae.sh_cc2652rb

r-win, 2022-10-04:
I've just found out that the Smoke detector from Fireangel use the exact same Zigbee module (you have to plug them in). That also means that the smoke detector is now detected as a Carbon Monoxide device instead of a Smoke detector.

Is there another way besides zigbeeModel and model to uniquely identify a device? Right now, there seems no difference between both devices, and I can't define a custom device provider in devices.yaml.

r-win, 2022-10-04:
It seems that I get them reasonably okay to work with the following (custom) device script. It has a few strange things, that I know not enough about.

For instance: the initial value of zoneStatus stays empty, unless I explicitly invoke a read on that attribute. The methods W2_ias_carbon_monoxide_alarm_1 and W2_ia as_smoke_alarm_1 are exact copies from fromZigbee.js, with additional types so that the responses are read during configuration.

Also, since this is a device with generic module, a solution is needed for the image. Right now, it's not possible to update the image of this device, even though it's not a Carbon Monoxide detector, but a Smoke Detector. Or it could be any detector, depending on the value of zoneType.

I just finished this script tonight. When it's a bit more polished, and the strange quirks are either out or confirmed to be alright, I can create a pull request for this. Please advice on the images though.

```javascript
/*************************************************************************************************
 *
 * Fireangel Zigbee module
 *
 * Documentation: https://www.eps-vertrieb.de/downloads/file/id/fd9a9593f3a3f1720b4654ea895a470b
 *
 *************************************************************************************************/

const {repInterval} = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

const converters = {
    W2_ias_carbon_monoxide_alarm_1: {
        cluster: 'ssIasZone',
        type: ['commandStatusChangeNotification', 'attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
            return {
                carbon_monoxide: (zoneStatus & 1) > 0,
                tamper: (zoneStatus & 1<<2) > 0,
                battery_low: (zoneStatus & 1<<3) > 0,
            };
        },
    },
    W2_ias_smoke_alarm_1: {
        cluster: 'ssIasZone',
        type: ['commandStatusChangeNotification', 'attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
            return {
                smoke: (zoneStatus & 1) > 0,
                tamper: (zoneStatus & 1<<2) > 0,
                battery_low: (zoneStatus & 1<<3) > 0,
                supervision_reports: (zoneStatus & 1<<4) > 0,
                restore_reports: (zoneStatus & 1<<5) > 0,
                trouble: (zoneStatus & 1<<6) > 0,
                ac_status: (zoneStatus & 1<<7) > 0,
                test: (zoneStatus & 1<<8) > 0,
            };
        },
    },
    W2_module_device_type: {
        cluster: 'ssIasZone',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const payload = {};

            const data = msg.data;
            if (data && data.hasOwnProperty('zoneType')) {
                payload.device_type = data.zoneType;
            }
            return payload;
        }
    },
    W2_module_battery: {
[... 45 lines cut ...]
                var device_type = endpoint.clusters.ssIasZone.attributes.zoneType

                if (device_type === 65535) { // Carbon monoxide
                    retval.push(e.carbon_monoxide());
                } else if (device_type === 40) { // Smoke
                    retval.push(exposes.binary('smoke', ea.STATE, true, false).withDescription('Smoke alarm status'));
                }
            }
            return retval;
        },
        configure: async(device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            const bindClusters = ['genPowerCfg', 'ssIasZone'];
            await reporting.bind(endpoint, coordinatorEndpoint, bindClusters);
            await reporting.batteryVoltage(endpoint);
            await reporting.batteryAlarmState(endpoint);
            await endpoint.read('genPowerCfg', ['batteryVoltage', 'batteryVoltMinThres', 'batteryRatedVoltage', 'batteryAlarmState']);

            await endpoint.read('ssIasZone', ['zoneType', 'currentZoneSensitivityLevel', 'zoneStatus']);

            device.powerSource = 'Battery';
            device.save();
        }
    },
];

r-win, 2022-10-23:
So, this is not working, since the devices (smoke detectors) are not reconnecting anymore, so their status is never updated. The CO2 module still works, which baffles me. 

How can I force the devices to:
a) Stay connected, or
b) Let them update their state frequently

closed 2022-11-30 by github-actions[bot]

thijselblaso, 2022-12-05:
@r-win I have the same w2-module zigbee as well. I use it in the Carbon monoxide alarm. I have it paired to zigbee2mqtt. Battery and Carbon both 'null'. Is there a way to fix this?

erwin-faceit, 2023-01-06:
The state of my device includes this information:

```
{
    "ac_status": false,
    "battery": 82,
    "battery_low": false,
    "carbon_monoxide": false,
    "device_type": 43,
    "last_seen": "2023-01-06T17:42:33.714Z",
    "linkquality": 10,
    "restore_reports": false,
    "sensitivity": "low",
    "smoke": false,
    "supervision_reports": false,
    "tamper": false,
    "test": false,
    "trouble": false,
    "voltage": 3000
}
```

What does your carbon monoxide returns?

DJerike, 2024-03-07:
> The state of my device includes this information:
>
> ```
> {
>     "ac_status": false,
>     "battery": 82,
>     "battery_low": false,
>     "carbon_monoxide": false,
>     "device_type": 43,
>     "last_seen": [...]

I have a zbst-ae630-eur angeleye smoke detector, and trying to add it to zigbee2mqtt, without succes.
I saw in this post that tou get values from these devices.
Can you help me how i can get this to work?
I tried the script in this post but it is not working for me.

thanks.

prolife86, 2024-03-13:
> So, this is not working, since the devices (smoke detectors) are not reconnecting anymore, so their status is never updated. The CO2 module still works, which baffles me.
> 
> How can I force the devices to: a) Stay connected, or b) Let them update their state frequently

Were you able to resolve this issue?