Koenkk/zigbee2mqtt #14312: Sonoff ZBDongle-P as ROUTER: PING fail (timeout)

keness521, 2022-10-03:
### What happened?

I flashed a factory fresh Sonoff Dongle-P with the v20220908 ROUTER firmware, although my first pairing attempt failed (interview failed) the second went through fine.

I connected it directly to my coordinator (also a Sonoff ZBDongle-P, v20220219 firmware) which is quite distant, but it still shows 57/65 lqi to the coordinator in the Z2M map.

It worked fine once the interview completed, and immediate took on a million dashed lines in the Z2M map, However, after just a few minutes, the PING will fail (timeout) twice and it will go OFFLINE. I can press the reconfigure button, and it will come back, but within just a few minutes, it will do the same.

Is it too far from the coordinator, even though the lqi looks reasonable-if-not-stellar?

I also flashed a different Dongle-P with v20220125 ROUTER firmware, and joined it to a completely separate network in a different location. (Also a Sonoff ZBDongle-P as coordinator on this network, v20220219 firmware.)

In this case, it is only about 4 feet away from the coordinator of that network.

Exact same results... After a couple minutes, I get a failed to PING message, then shortly later, another, and after two failures, it goes "Disabled".

### What did you expect to happen?

I expected it to... not have PING timeouts? :)

### How to reproduce it (minimal and precise)

Happens with both devices within minutes. One is on a release version router firmware, the other I think a newer development router firmware, on two separate networks in separate locations. (Coordinator is the same model and version for both networks.)

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

COOR: v 20220219, ROU: v20220908 & v20220125

### Adapter

Sonoff ZBDongle-P

### Debug log

Will update this ASAP... I didn't copy the error messages and now I'm just waiting for it to timeout again. But it did two timeouts, and after the second is when it goes "Offline".

keness521, 2022-10-03:
I should have been more patient... It only took another minute to happen. :)

Warning 2022-10-03 13:40:49Failed to ping 'Office Router' (attempt 1/2, Read 0x00124b00257caa12/242 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 13889 - 242 - 71 - 0 - 1 after 10000ms))

Warning 2022-10-03 13:41:15Failed to ping 'Office Router' (attempt 2/2, Read 0x00124b00257caa12/242 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 13889 - 242 - 72 - 0 - 1 after 10000ms))

jerrm, 2022-10-03:
Please try restarting HA, if that doesn't work reboot the host.

Please report back results.

See my issue: https://github.com/Koenkk/zigbee2mqtt/issues/13758#issue-1354893710

ashokaiyar, 2022-10-03:
> I should have been more patient... It only took another minute to happen. :)

While configuring that device (ti.router), under reporting, decrease the maximum reporting interval. I use 3 minutes.

[image]

keness521, 2022-10-04:
> > I should have been more patient... It only took another minute to happen. :)
> 
> While configuring that device (ti.router), under reporting, decrease the maximum reporting interval. I use 3 minutes.

Thanks, I've made that update to both devices, and so far (about an hour) there have been no PING failures and the routers have stayed online.

I'll report back after a more significant amount of time has passed.

Just because I like to understand things, what did that change from 3600 to 180... do?

keness521, 2022-10-04:
> Please try restarting HA, if that doesn't work reboot the host.
> 
> Please report back results.
> 
> See my issue: [#13758 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/13758#issue-1354893710)

Thanks for the tip. I actually ended up seeing the tip[ right after yours first (about decreasing the maximum reporting interval( and I implemented that. I didn't restart yet because I didn't want to do both at the same time and then never know which one was the actual solution.

I have two more routers to set up, so if they do the same (I suspect they will) I'll restart the pi everything is running on and connected to, and see if that also fixes it.

Thanks!

jerrm, 2022-10-04:
Reporting interval will work. 3 minutes seems a little excessive.  Setting about 30 seconds less than whatever your mains availability setting is configured for should be sufficient. 

But...

Please try the z2m restart if the next stick does the same thing.  My personal experience has to be considered anecdotal,  but if it is consistent, a simple "restart" will be an easy fix for anyone else who comes along with the same issue.

ashokaiyar, 2022-10-04:
> Just because I like to understand things, what did that change from 3600 to 180... do?

Sets the reporting interval to be 3 minutes in place of one hour. On my mesh, routers are marked as offline if they haven't reported in 20 minutes., while end-devices are marked as offline if they haven't reported for 24 hours. Setting the reporting interval to 3 minutes, gives several opportunities for the device to contact the coordinator before being marked online. Communicating once in 3 minutes shouldn't make a lick of difference to the mesh (if it does, there are real issues with the mesh that to be fixed!).

> My personal experience has to be considered anecdotal,

This approach did not work for my setup.

jerrm, 2022-10-04:
> I'd be curious to see if it works for your setup, and honestly befuddled as to why a reboot should fix this issue of a router dropping off (or being marked as offline) because it doesn't report within a preset interval.

I was befuddled why the router wasn't responding to the 10 minute ping from z2m.  I would guess something in z2m is out of sync.  Could be the ping request is malformed before it is sent or not properly processed when received.  Whatever the case the reboot or restart seems to fix it.

The reboot/restart has been consistent here with my third and fourth router sticks.  At this point I have four sticks as routers,  None have a reporting interval set and all stay online consistently.

On the first stick I never noticed an issue, but suspect I had rebooted or restarted somewhere along the line before it was a problem.  The second stick was going offline, so I set the reporting interval, then much later (months) noticed it started responding to pings - I assume an intervening restart "fixed" it.  I removed the reporting interval and all remains well.   The third stick was the first time I tried a more methodical debug, the reboot step fixed it.  On the fourth stick, I restarted the z2m container instead of rebooting and all works.

keness521, 2022-10-05:
Some updates... The ZBDongle-P on which I decreased the maximum reporting interval has been operating flawlessly.

The other ZBDongle-P I had also made the same adjustment, but then I also decided to flash it with the v20220908 version router firmware since that didn't seem to be the source of the problem and it is newer. When I did this, I struggled getting it re-connected to the network, and ultimately I think I deleted it completely and then re-paired it. (I'll need to look into this later, I want to be able to smoothly upgrade the firmware.)

Once it was re-added, the maximum reporting interval was back to the default, and the problem started immediately again.

So I did the other potential solution offered, I simply restarted the Raspberry Pi that Z2M and coordinator are on, and it ALSO seems to have fixed the problem.

So yeah, it really does seem that simply restarting ALSO clears up the problem.

I still have two other ZBDongle-P's I will be setting up in the not-too-distant future. I'll see what they do as well.

(The frequency of reporting on the one I just restarted, and didn't change the maximum reporting interval, is less frequent, but still not very many minutes.)

jerrm, 2022-10-05:
> ultimately I think I deleted it completely and then re-paired it

Unfortunately, re-pairing is necessary after a router flash.  

For the last upgrade I did, I had initially re-paired without removing first and had an issue with the reported fw version not updating. You might double check the reported version to make sure yours is correct.   @Koenkk reccomended doing a force remove, restart, then re-pair.   That worked for me.

@Koenkk suggested OTA updates may be in the future, which would eliminate a re-pair. We'll have to wait and see.

> it really does seem that simply restarting ALSO clears up the problem.

Good to hear.  Please post your experience on the two other sticks when done.

closed 2022-11-12 by github-actions[bot]