Koenkk/zigbee2mqtt #14316: Z2M sends empty HomeAssistan autodiscover configuration messages

Nikolay-Ch, 2022-10-03:
### What happened?

When I turn on Autodiscover on my HA and tunr on HomeAssistant integration option in Z2M, nothing happends. Devices from Z2M not appears in HomeAssistant devices list.

When I examine logs on the HomeAssistant side, I found, that Z2M send one full configuration message with payload, and then send some more configuration messages (3-5 messages) to the same topic with empty payload. So, when Home Assistant receive this empty message, it delete this device from self database.

When I see logs on the Z2M side, I not found nothing interesting. Еhe message is sent with a filled payload.

Additionally, I watch the messages through the MQTT.fx client - it sees these empty configuration messages

### What did you expect to happen?

I expect, that my devices (AQARA temperature sensors) are discovered in Home Assistant

### How to reproduce it (minimal and precise)

Start Z2M, configure on it Home Assistant integration
Start HomeAssistant, turn on it Z2M plugin Autodiscovery option.

### Zigbee2MQTT version

1.27.0 commit: a9b8808

### Adapter firmware version

20211222

### Adapter

zStack30x

Koenkk, 2022-10-04:
Can you provide the debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Nikolay-Ch, 2022-10-07:
So... This logs from second Z2M instance...
```
debug 2022-10-07 23:58:59: Received MQTT message on 'homeassistant/sensor/0x8cf681aaaaaaaa42/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x8cf681aaaaaaaa42"],"manufacturer":"TuYa","model":"Door sensor (TS0203)","name":"0x8cf681aaaaaaaa42"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x8cf681aaaaaaaa42","name":"0x8cf681aaaaaaaa42 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x8cf681aaaaaaaa42","unique_id":"0x8cf681aaaaaaaa42_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
debug 2022-10-07 23:58:59: Clearing Home Assistant config 'homeassistant/sensor/0x8cf681aaaaaaaa42/linkquality/config'
info  2022-10-07 23:58:59: MQTT publish: topic 'homeassistant/sensor/0x8cf681aaaaaaaa42/linkquality/config', payload 'null'
```

first messager received from first mqtt-broker to second mqtt-broker (by bridge link).
first message created by first instance of Z2M in first Zigbee network.
looks like, that second instance clearing received message and send it back with zero-length payload.

Nikolay-Ch, 2022-10-07:
I think, that Z2M clears message in this code in homeassistant.ts (line 1330)
```
const entity = this.zigbee.resolveEntity(ID);
let clear = !entity || entity.isDevice() && !entity.definition;
```
This code set clear flag of received message because this instance not have any device with received ID...

Koenkk, 2022-10-08:
Do the two z2m instances have different `base_topic`? (https://www.zigbee2mqtt.io/guide/configuration/mqtt.html) This is mandatory and probably what causes the issue.

Nikolay-Ch, 2022-10-08:
Thanks! It is working now correctly! Thank you!

closed 2022-10-08 by Nikolay-Ch