Koenkk/zigbee2mqtt #14317: [Feature request]:

Mirarkitty, 2022-10-04:
### Is your feature request related to a problem? Please describe

Data is not always recent in the rest of the automation system, with caching and data retention. MQTT does not support timestamps on the data.

### Describe the solution you'd like

Add an **option** to add a timestamp together with other generated data like link quality, like

z2m/friendly {"humidity":42.7,"linkquality":171,**"timestamp":1664868114**}

### Describe alternatives you've considered

This can be solved by a MQTT subscriber that adds the timestamps before the rest of the system uses the data. It would just be practical and times closer to the source of the data if z2m adds it. 

### Additional context

-

Mirarkitty, 2022-10-04:
Added this and it works fine, so the patch is relatively simple (except the option reading part).

        // Add device linkquality.                                              
        if (entity.isDevice() && entity.zh.linkquality !== undefined) {
            message.linkquality = entity.zh.linkquality;
        }
        _**message.timestamp = Date.now()/1000;**_
        for (const extension of this.extensions) {
            (_a = extension.adjustMessageBeforePublish) === null || _a === void\
 0 ? void 0 : _a.call(extension, entity, message);
        }

Koenkk, 2022-10-04:
Please check the `last_seen` functionallity: https://www.zigbee2mqtt.io/guide/configuration/mqtt.html#server-connection

closed 2022-11-11 by github-actions[bot]