Koenkk/zigbee2mqtt #14319: Pairing failed / pairing loop

cranberryNL, 2022-10-04:
### What happened?

I've an issue with re-pairing Sonoff ZBMini switches. I'm able to succesfully pair a new ZBMini. But if an device has already paired (in my old/previous ConbeeII/Deconz setup and also in my new z2m/Sonoff zigbee 3.0 USB 3 Dongle-E), pairing starts, but is failing during the pairing process. Then a the process starts again and so it's in a loop of pairing - failing - pairing - failing and so on.
I tried to reset the device, remove it from a power source, but without success.

Setup:
Zigbee2MQTT version: 1.27.2 commmit 7dc48fb
Coordinator type: EZSP v8
Coordinator revision: 6.10.3.0 build 297
Frontend: 0.6.110
Sonoff Zigbee 3.0 USB Dongle Plus–ZBDongle-E
[image]

[puttyghs2.fritz.box 2022-10-ghs2.fritz.box 101004 - Copy.log](https://github.com/Koenkk/zigbee2mqtt/files/9704467/puttyghs2.fritz.box.2022-10-ghs2.fritz.box.101004.-.Copy.log)

### What did you expect to happen?

I hope the root cause can be found and most important a solution ;)

### How to reproduce it (minimal and precise)

- get a paired , well functioning Sonoff ZBMini
- press the reset button for ~ 10 sec so the device resets itself and tries to pair
- enable permit joining
- z2m sees the device and continues the pairing process

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus–ZBDongle-E

### Debug log

[puttyghs2.fritz.box 2022-10-ghs2.fritz.box 101004 - Copy.log](https://github.com/Koenkk/zigbee2mqtt/files/9704502/puttyghs2.fritz.box.2022-10-ghs2.fritz.box.101004.-.Copy.log)

kirovilya, 2022-10-04:
please update z2m to 1.28 and try again

cranberryNL, 2022-10-04:
Many thanks kirovilya, excuse me for upgrading first before opening a new problem report.
Upgrading solved the issue. I'm very, very happy. Many thanks!

closed 2022-10-04 by cranberryNL