Koenkk/zigbee2mqtt #14325: Startup hangs on Starting zigbee-herdsman

Lovasz-Akos, 2022-10-04:
### What happened?

Trying to start zigbee2mqtt 1.28 it hangs on Starting zigbee-herdsman (0.14.62), no error just stops there.

### What did you expect to happen?

Startup continues and zigbee2mqtt launches

### How to reproduce it (minimal and precise)

followed install instructions all the way even npm ci returned with expected message.
then went to npm start and it's hanging, with barely and CPU or RAM usage, so probably not even doing anything

### Zigbee2MQTT version

1.28

### Adapter firmware version

v8

### Adapter

ezsp

### Debug log

```bash
Zigbee2MQTT:info  2022-10-04 12:42:24: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
Zigbee2MQTT:info  2022-10-04 12:42:24: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:debug 2022-10-04 12:42:24: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/home/admin/zigbee2mqtt-1.28.0/data/coordinator_backup.json","databaseBackupPath":"/home/admin/zigbee2mqtt-1.28.0/data/database.db.backup","databasePath":"/home/admin/zigbee2mqtt-1.28.0/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/ttyUSB0"}}'
Zigbee2MQTT:debug 2022-10-04 12:47:24: Saving state to file /home/admin/zigbee2mqtt-1.28.0/data/state.json
```

Lovasz-Akos, 2022-10-04:
Additional info:
` ls -l /dev/serial/by-id` returns `usb-1a86_USB_Serial-if00-port0 -> ../../ttyUSB0`

`test -w /dev/ttyUSB0 && echo success || echo failure` returns `success`

my config is:
```yaml
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: false

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  port: /dev/ttyUSB0
  adapter: ezsp

frontend: true
advanced:
  log_level: debug

```

Koenkk, 2022-10-04:
@kirovilya this looks to be an EZSP specific issue. Can you provide the herdsman debug log of this? 

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Lovasz-Akos, 2022-10-04:
the debug log part of the issue is all I see even when using `advanced:
  log_level: debug`
and starting with `DEBUG=zigbee-herdsman* npm start`

kirovilya, 2022-10-04:
@Lovasz-Akos what stick are you using? maybe the port speed is different from 115200 or rtscts should be true?

Lovasz-Akos, 2022-10-04:
I'm using an EFR32MG21

kirovilya, 2022-10-04:
Check your chip with https://github.com/Elelabs/elelabs-zigbee-ezsp-utility 
maybe you have it in bootloader mode and you need to switch to normal mode

kirovilya, 2022-10-06:
@Lovasz-Akos yes, you need EZSP v8 or v9 :)

Lovasz-Akos, 2022-10-06:
okay It's up and running now, sorry about my carelessness, thank you for your time ❤

closed 2022-10-06 by Lovasz-Akos

perfettdamien, 2023-02-21:
HI, 
I struggle a bit with my dongle E plus V2, i followed https://github.com/Elelabs/elelabs-zigbee-ezsp-utility . 
And i can't restart my dongle in normal mode, it auto come back to bootloader mode...
`2023/02/21 13:48:18 Elelabs_EzspFwUtility:   EZSP adapter in bootloader mode detected:
2023/02/21 13:48:18 Elelabs_EzspFwUtility:   > appNwkJoinOrRejoinWhenPowerOn [291]:
2023/02/21 13:48:18 Elelabs_EzspFwUtility:   Launch in normal application mode
2023/02/21 13:48:29 Elelabs_EzspFwUtility:   EZSP adapter in bootloader mode detected:
2023/02/21 13:48:29 Elelabs_EzspFwUtility:   Usage:`
Thanks in advance for your help

atiburzi, 2023-06-24:
Hi,
you have to restart the dongle keep pressing the bootloader button when you power it.
Obviusly you have to dismount it from the shell.
Then you can re-flash it using whatever firmware.

I hope it can help you.