Koenkk/zigbee2mqtt #14326: [New device support]: SONOFF zigbee Model E - with router firmware

JacekKac, 2022-10-04:
### Link

https://sonoff.tech/product/diy-smart-switch/sonoff-zigbee-3-0-usb-dongle-plus-e/

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0xe0798dfffe719729","nwkAddr":53083,"manufId":4742,"manufName":"SONOFF","powerSource":"Unknown","modelId":"DONGLE-E_R","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"DONGLE-E_R","manufacturerName":"SONOFF","powerSource":0,"zclVersion":8,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20220515","swBuildId":"1.0.0"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20220515","swBuildId":"1.0.0","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1664890962123,"defaultSendRequestWhen":"immediate"}

### Comments

SONOFF E dongle flashed with https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/blob/master/Dongle-E/Router/Z3RouterUSBDonlge_EZNet6.10.3_V1.0.0.gbl

acting as a router

Koenkk, 2022-10-04:
`DONGLE-E_R` is supported in z2m 1.28.0

closed 2022-10-04 by Koenkk

JacekKac, 2022-10-05:
ok do you know how to set transmit power for this router? 

https://sonoff.tech/wp-content/uploads/2022/05/Configuring-ZBDongle-P-Transmit-Power-Tutorial.pdf
"
Set the transmit power through the serial port command SYS_SET_TX_POWER "

as i understand it's not possible from zigbe2mqqt config, it must be done by serial but don't know how ;(

Koenkk, 2022-10-05:
For routers you cannot, I'm working on this feature in https://github.com/Koenkk/Z-Stack-firmware/issues/341